espressif/idf-eclipse-plugin #2: ESP-IDF Directory selection dialog doesn't appear

RomanLutig, 2019-10-04:
I followed step by step the instruction.
Installed the plugin using URL method.
Then I go to:
Help > Espressif IDF Tools Manager > Install Tools
and after pushing install nothing happens.
Directory selection dialog doesn't appear.

I am using:
Eclipse - Version: 2019-09 R (4.13.0), Build id: 20190917-1200
Java - Java SE Development Kit 13

hwaldman, 2019-10-04:
I also see the same problem.
Eclipse version: Version: 2019-09 R (4.13.0)
java version "1.8.0_191"
Python 3.7.4

AwkBot, 2019-10-04:
Yes... I'm stopped in the same step.
The only workaround that I found in Windows, is by the command line, set ESP-IDF variables then run idf.py menuconfig... it is text mode however it the only alternative that I found

daniauri, 2019-10-05:
Yes, it's the same for me. @AwkBot Can you share what you did to make it working?

daniauri, 2019-10-05:
I bet it has something to do with the plugin not finding python 3.5. I have only the least version 3.7 installed from https://www.python.org/downloads/
[image]

Ah, it seems not. Does someone know how eclipse resolves the python path?
[image]

serenagrl, 2019-10-05:
I have also experienced the same issue but I had managed to workaround it. This is caused by missing registry entries when you have upgraded a version of python or uninstalled an older version before. You need to make sure you have exactly 1 version of python 2.7 and 3.7.x

The plugin searches for something like the following:
HKEY_CURRENT_USER\Software\Python\PythonCore\2.7\InstallPath
and/or
HKEY_CURRENT_USER\Software\Python\PythonCore\3.x\InstallPath

You can fix the issue by deleting any old python entries in your registry or if you want to play it safe, you can create the path that the plugin is searching for in the registry and point it to the python directory where you have installed it.

Hope this helps.

daniauri, 2019-10-05:
Can confirm the solution from @serenagrl. Deleted registry entry from 3.6 which I uninstalled a while ago. Now It's finding 3.7. Apperently the plugin brakes when the InstallPath Key is not found for any version (climbing to higher version numbers).

See https://www.python.org/dev/peps/pep-0514/.

Ede2016, 2019-10-06:
I solved this by uninstalling the newest Java and installing v8.
And then I edited the Java related line in the eclipse.ini to
-vm
C:\Program Files\Java\jdk1.8.0_221\bin

The eclipse.ini is here, at least on my PC: C:\Users\UserName\eclipse\cpp-2019-09\eclipse\eclipse.ini

Now the ESP-IDF directory selection dialog appears.
Before that I got rid of an old Python version and old Python registry entries. That did not change anything.

kolipakakondal, 2019-10-10:
Issue got fixed with the release **v1.0.0-beta** - https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.0.0-beta

There was an issue in identifying the python path in Windows OS with Java 11 and higher, and it got fixed now.

EduardShaid, 2019-10-13:
Installed the latest version v1.0.0-beta, but the result is the same - "Espressif IDF Tools Manager> Install Tools" does not start !!!

kolipakakondal, 2019-10-14:
> Installed the latest version v1.0.0-beta, but the result is the same - "Espressif IDF Tools Manager> Install Tools" does not start !!!

Could you please provide the console log error. If the IDF Tools console is not visible, you can access it using https://github.com/espressif/idf-eclipse-plugin#espressif-idf-tools-console

EduardShaid, 2019-10-14:
"Espressif IDF Plugins for Eclipse" is installed, but the "Espressif IDF Tools Console" is missing.

Eclipse IDE for C/C++ Developers
Version: 2018-12 (4.10.0)

kolipakakondal, 2019-10-14:
> "Espressif IDF Plugins for Eclipse" is installed, but the "Espressif IDF Tools Console" is missing.
> 
> Eclipse IDE for C/C++ Developers
> Version: 2018-12 (4.10.0)

That appears only when you invoke "Espressif IDF Tools Manager> Install Tools" but sometimes it won't switch to IDF Console View when you're running the install tools command.  You can toggle the console as mentioned here https://github.com/espressif/idf-eclipse-plugin#espressif-idf-tools-console

FabioGavinelli, 2019-10-23:
hi, I choosed the wrong directory when the first time the Choose Path window has appeared. Now I can't find how to update it and I can't install tools. 
How I am supposed to fix this problem?

kolipakakondal, 2019-10-25:
> hi, I choosed the wrong directory when the first time the Choose Path window has appeared. Now I can't find how to update it and I can't install tools.
> I tried to uninstall and reinstall the plugin but the directory selection dialog doesn't appear anymore
> How I am supposed to fix this problem?

Please follow the instructions from here to modify the IDF_PATH https://github.com/espressif/idf-eclipse-plugin#configureEnvironmentVariables

kolipakakondal, 2019-10-30:
Issue related to opening the console view while running the Install Tools from eclipse is fixed with the v1.0.0-beta.2 version. Please get the latest changes using the update site.

Check more here:
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.0.0-beta.2

closed 2019-10-30 by kolipakakondal

ralphs99, 2019-12-02:
Hello, I'm having the same problem (Directory selection dialog doesn't appear) with v1.0.0-beta.2

Eclipse:
Version: 2019-09 R (4.13.0)
Build id: 20190917-1200
Python: 3.80
Java SE Development Kit 13

Any thoughts about the cause of the trouble?
Thanks in advance, Ralph