espressif/idf-eclipse-plugin #4: Unable to open sdkconfig

serenagrl, 2019-10-05:
When double clicking on the sdkconfig, it will prompt a dialog stating "Starting JSON Configuration Server" and the it will failed with a message on a new tab:

"Error retrieving output from the json configserver, please check the error log"

The Error Log windows does not show any error details.

daniauri, 2019-10-05:
As @serenagrl said, same here. The Icon for sdconfig is not the same as shown in the description.
[image]
Maybe the wrong editor is linked? No output to Console either.
The Editor is called "SDK Configuration"...
[image]

AwkBot, 2019-10-05:
I have the same issues, I tried to manually select SDK Configuration then I face the same error.

jirawatkongkaen, 2019-10-07:
[image]
I've same issue with eclipse 2019-09 on windows 64-bits.

jirawatkongkaen, 2019-10-07:
I'v been solved this issue by install java 11

kolipakakondal, 2019-10-10:
The issue got fixed with **v1.0.0-beta** - https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.0.0-beta

Please use the update site to get the latest.

FilipNevezi, 2019-10-12:
> The issue got fixed with **v1.0.0-beta** - https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.0.0-beta
> 
> Please use the update site to get the latest.

How do you update? I tried updating and even uninstalling the plugin and then installing it again. However, when I then click "Help/Espressif IDF Tools Manager/Install Tools" it pretty much skips everything as it's already installed.

aleism2007, 2019-10-13:
Hi, I have the same problem

zhao791204, 2019-10-14:
I have meeting the same problem too, i use The windows 10.

zhao791204, 2019-10-14:
[image]
[image]

kolipakakondal, 2019-10-14:
> 
> How do you update? I tried updating and even uninstalling the plugin and then installing it again. However, when I then click "Help/Espressif IDF Tools Manager/Install Tools" it pretty much skips everything as it's already installed.

To update IDF Eclipse Plugins:
* Help > Check for Updates
* If updates are found, select `Espressif IDF Plugins for Eclipse` and deselect all other items and proceed with the installation.

@FilipNevezi  - Are you having issues with the Tools installation or sdkconfig editor launch?
If you're having issues with the tools installation, please provide the console message error which is appearing in the 'IDF Tools Console'. You can access the IDF Tools console if it's not active https://github.com/espressif/idf-eclipse-plugin#espressif-idf-tools-console

If you're having editor launch issue, please provide the error log https://github.com/espressif/idf-eclipse-plugin#error-log

AwkBot, 2019-10-14:
Hi All,

I just tested the updated version,  looks like there are some minor error:

`
!ENTRY com.espressif.idf.core 4 0 2019-10-14 08:38:48.574
!MESSAGE Illegal char <
> at index 49: C:\esp-idf\.espressif\python_env\idf4.1_py3.7_env
/Scripts\python
!STACK 0
java.nio.file.InvalidPathException: Illegal char <
> at index 49: C:\esp-idf\.espressif\python_env\idf4.1_py3.7_env
/Scripts\python
	at sun.nio.fs.WindowsPathParser.normalize(Unknown Source)
	at sun.nio.fs.WindowsPathParser.parse(Unknown Source)
	at sun.nio.fs.WindowsPathParser.parse(Unknown Source)
	at sun.nio.fs.WindowsPath.parse(Unknown Source)
	at sun.nio.fs.WindowsFileSystem.getPath(Unknown Source)
	at java.nio.file.Paths.get(Unknown Source)
	at com.espressif.idf.core.util.IDFUtil.findCommand(IDFUtil.java:154)
	at com.espressif.idf.core.util.IDFUtil.getIDFPythonEnvPath(IDFUtil.java:106)
	at com.espressif.idf.sdk.config.core.server.JsonConfigServer.start(JsonConfigServer.java:89)
	at com.espressif.idf.sdk.config.core.server.ConfigServerManager.getServer(ConfigServerManager.java:42)
	at com.espressif.idf.sdk.config.ui.SDKConfigurationEditor.initConfigServer(SDKConfigurationEditor.java:338)
	at com.espressif.idf.sdk.config.ui.SDKConfigurationEditor.lambda$0(SDKConfigurationEditor.java:150)
	at org.eclipse.jface.operation.ModalContext$ModalContextThread.run(ModalContext.java:122)

!ENTRY com.espressif.idf.core 1 0 2019-10-14 08:38:48.578
!MESSAGE [C:\esp-idf\.espressif\python_env\idf4.1_py3.7_env\Scripts\python.exe, D:\esp-idf\esp-idf\tools\idf.py, confserver]
`

kolipakakondal, 2019-10-14:
> !MESSAGE Illegal char <
> 
> > at index 49: C:\esp-idf.espressif\python_env\idf4.1_py3.7_env
> > /Scripts\python
> > !STACK 0
> > java.nio.file.InvalidPathException: Illegal char <
> > at index 49: C:\esp-idf.espressif\python_env\idf4.1_py3.7_env
> > /Scripts\python

It seems to be Illegal char **<** is inserted in the PATH. Can you verify PATH from the build preferences? https://github.com/espressif/idf-eclipse-plugin#configuring-environment-variables

aleism2007, 2019-10-14:
hi, I think the problem is with this dash "/" before the "Scripts", but i don't now how to change it.

kolipakakondal, 2019-10-15:
> hi, I think the problem is with this dash "/" before the "Scripts", but i don't now how to change it.

Understood the problem. **\r** character in front of **/Scripts\python** is causing the issue. We will fix it soon.

**As a workaround:**

1. Go to preferences https://github.com/espressif/idf-eclipse-plugin#configureEnvironmentVariables
2. Copy the value of **IDF_PYTHON_ENV_PATH** to the notepad (To avoid \r character)
3. Add again
4. Apply and close

serenagrl, 2019-10-15:
Hi,

I have tried using a fresh installed version of eclipse (2019.09) to test with the updated plugins and the problem still exist. I do not have Java 11 installed but instead uses the recommended Java 1.8.

zhao791204, 2019-10-15:
[image]
in CMD, json configration server run successfully. in eclipse,plugin still has error

rdkluijv, 2019-10-18:
The illegal character < message is from an incorrect path. %PATH% is not expended in the eclipse environment PATH. Removing it and concatenate the system path manually. This will remove the message but editing the sdkconfig still doesn't work.
My workaround is to open the file, change the settings and save it in a txt-editor.

robcazzaro, 2019-10-18:
I have the same error, in the SDK editor plug in, using the latest beta files and installing directly from the URL

The log says

!ENTRY com.espressif.idf.sdk.config.ui 4 0 2019-10-18 15:26:30.323
!MESSAGE Invalid thread access
!STACK 0
org.eclipse.swt.SWTException: Invalid thread access
	at org.eclipse.swt.SWT.error(SWT.java:4711)
	at org.eclipse.swt.SWT.error(SWT.java:4626)
	at org.eclipse.swt.SWT.error(SWT.java:4597)
	at org.eclipse.swt.widgets.Widget.error(Widget.java:452)
	at org.eclipse.swt.widgets.Widget.checkWidget(Widget.java:370)
	at org.eclipse.swt.widgets.Widget.checkParent(Widget.java:290)
	at org.eclipse.swt.widgets.Widget.<init>(Widget.java:155)
	at org.eclipse.swt.widgets.Control.<init>(Control.java:113)
	at org.eclipse.swt.widgets.Scrollable.<init>(Scrollable.java:85)
	at org.eclipse.swt.widgets.Composite.<init>(Composite.java:99)
	at com.espressif.idf.sdk.config.ui.SDKConfigurationEditor.createErrorPage(SDKConfigurationEditor.java:303)
	at com.espressif.idf.sdk.config.ui.SDKConfigurationEditor.lambda$0(SDKConfigurationEditor.java:142)
	at org.eclipse.jface.operation.ModalContext$ModalContextThread.run(ModalContext.java:122)

Mezgrman, 2019-10-27:
I too have this issue, running Eclipse 2019-03 (4.11.0) on WIn10 64 bit. Installed from scratch using the latest revision, but I get the "retrieving output" error. Nothing to be seen in the error log though, so I don't know where to start...

mishafarms, 2019-11-02:
I am experiencing the same issue. I am running Eclipse 2019-09R on linux. I am running in a docker (modified from the idf V4.0 Dockerfile).

Here is the log info.

!ENTRY com.espressif.idf.core 1 0 2019-11-02 15:32:50.551
!MESSAGE [null, /opt/esp/idf/tools/idf.py, confserver]

!ENTRY com.espressif.idf.sdk.config.core 4 0 2019-11-02 15:32:50.552
!MESSAGE 
!STACK 0
java.lang.NullPointerException
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1090)
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1071)
	at com.aptana.core.util.ProcessRunner.startProcess(ProcessRunner.java:166)
	at com.aptana.core.util.ProcessRunner.doRun(ProcessRunner.java:147)
	at com.aptana.core.util.ProcessRunner.run(ProcessRunner.java:86)
	at com.espressif.idf.sdk.config.core.server.JsonConfigServer.start(JsonConfigServer.java:97)
	at com.espressif.idf.sdk.config.core.server.ConfigServerManager.getServer(ConfigServerManager.java:42)
	at com.espressif.idf.sdk.config.ui.SDKConfigurationEditor.initConfigServer(SDKConfigurationEditor.java:338)
	at com.espressif.idf.sdk.config.ui.SDKConfigurationEditor.lambda$0(SDKConfigurationEditor.java:150)
	at org.eclipse.jface.operation.ModalContext$ModalContextThread.run(ModalContext.java:122)

baicaihuan, 2019-11-05:
remove the \r character，and the problem still exists:

!ENTRY com.espressif.idf.core 1 0 2019-11-05 10:22:31.083
!MESSAGE [C:\Users\uername\.espressif\python_env\idf4.1_py3.7_env\Scripts\python.exe, C:\Users\uername\esp\esp-idf\tools\idf.py, confserver]

kolipakakondal, 2019-11-05:
Hi All, I need help in verifying this issue with my local build zip file.

**How to verify?**

1. Download https://github.com/kolipakakondal/idf-eclipse-plugin-dev/blob/master/com.espressif.idf.update-1.0.0-SNAPSHOT.zip
2. In Eclipse, choose Help -> Install New Software
3. Click Add… button
4. Select Archive from Add repository dialog and select the file com.espressif.idf.update-1.0.0-SNAPSHOT.zip
5. Click Add
6. Select Espressif IDF from the list and proceed with the installation
7. Restart the Eclipse

Now, verify if you're able to launch sdkconfig editor for your project. If see any issues, check the "JSON Configuration Server Console" which is added newly with this fix.

Based on the feedback, I'll update the plugins update site URL.

rdkluijv, 2019-11-05:
Hi All,

Checked but it didn't work. I also could not find the JSON console. 
Window - Show View - Other ... and entering J
The update made a remark that it was an update but I assume that's correct. Did I do something wrong?

Bye,

Rob.

kolipakakondal, 2019-11-05:
> Checked but it didn't work. I also could not find the JSON console.
> Window - Show View - Other ... and entering J
> The update made a remark that it was an update but I assume that's correct. Did I do something wrong?

Have you restarted the eclipse after the plugin update? 
Can you verify the plugin version? It should be 1.0.0.201911050434

> Help > About Eclipse IDE> Installation Details > Installed Software > Search for "Espressif"

JSON Configuration Server console is part of the console view, it automatically opens up while launching the sdkconfig editor

[image]

rdkluijv, 2019-11-05:
Hi,

Found the console after the comment. The content is in the file. The date and time of the update is correct. I also noticed that Perl is found. That happend never before! 
Should I try it with a new project?

Bye,

Bor.

[json.txt](https://github.com/espressif/idf-eclipse-plugin/files/3808173/json.txt)

kolipakakondal, 2019-11-05:
> Found the console after the comment. The content is in the file. The date and time of the update is correct. I also noticed that Perl is found. That happend never before!
> Should I try it with a new project?

Okay, great. JSON response looks good. So what's the problem now - editor still showing an error? Yes, try with a new project and see what happens.

Any error reported in the error log?

rdkluijv, 2019-11-05:
Same error message as before.
Currently running the build of a template project.
Will send the error log when build done. 
Have to go in half an hour.

rdkluijv, 2019-11-05:
No problems building.
Some error when starting the editor.
Error log latest message: Server running, waiting for requests on stdin...
See nothing wrong.

rdkluijv, 2019-11-05:
Some more info:
I did exactly the same thing on my computer at work. That's the one where the ulp projects are also working. On this computer the patch is working! It didn't work before I checked.
Any suggestions what to check to fix it at home?

baicaihuan, 2019-11-05:
it works at first time, but fails at second time

[image]

[image]

rdkluijv, 2019-11-05:
Second time worked with me here.
Both the toolchain and eclipse are installed the default way with no problems.
OS: Win10 enterprise 2016 LTSB (64)
CPU: i7-4770 3.4 GHz
RAM: 8 GB
I see a difference in the error log at work and at home.
At work all the latest messages are repeated.

[image]

robcazzaro, 2019-11-05:
@kolipakakondal following your instructions it seems to be working now for me. Good job!

kolipakakondal, 2019-11-06:
> it works at first time, but fails at second time
> 
Was it taking too long to get the json output on the console? The editor will be shown with an error after 15sec timeout

rdkluijv, 2019-11-06:
I think that could be the cause. The computer at home has limited cpu power. Is it possible to adjust the timeout?

kolipakakondal, 2019-11-06:
Here is one more local build https://github.com/kolipakakondal/idf-eclipse-plugin-dev/raw/master/com.espressif.idf.update-1.0.0.201911060700.zip

Hopefully, this solves all the editor issues.

**To verify:**

1. Download above build
2. In Eclipse, choose Help -> Install New Software
3. Click Add… button
4. Select Archive from Add repository dialog and select the file com.espressif.idf.update-1.0.0.201911060700.zip
5. Click Add
6. Select Espressif IDF from the list and proceed with the installation
7. Restart the Eclipse

@rdkluijv Please try this

rdkluijv, 2019-11-06:
Installed it and checked if the updated software showed. 
Some timing.
After 30 seconds the error shows.
After 50 seconds the stdin line shows in the console.

kolipakakondal, 2019-11-06:
Okay, let's do this.

Here is the new build which increases the timeout to 120sec
https://github.com/kolipakakondal/idf-eclipse-plugin-dev/raw/master/com.espressif.idf.update-1.0.0.201911060844.zip

If you still see any issue, set this "-Dsdkconfig.timeout" vm property in the eclipse.ini file with a value

-`Dsdkconfig.timeout=200`

https://wiki.eclipse.org/Eclipse.ini#-vm_value

rdkluijv, 2019-11-06:
OK. It's working. I looked at the output in the console. A possible cause for the delay is that git is trying to find a repository. This computer is also slow, but silent, so I replace it. Probably fix more issues!
Thanks for the work.

kolipakakondal, 2019-11-06:
Okay, great. Yes, please open a new issue with all the details.

Mezgrman, 2019-11-07:
@kolipakakondal your build fixed it for me as well, thanks!

winkste, 2019-11-10:
Hi. 
I still get that error while trying to open the sdkconfig file. Please see attached my error log. As I'm facing other errors as well, they might be connected.
[sdkconfig.log](https://github.com/espressif/idf-eclipse-plugin/files/3828968/sdkconfig.log)

kolipakakondal, 2019-11-11:
> Hi.
> I still get that error while trying to open the sdkconfig file. Please see attached my error log. As I'm facing other errors as well, they might be connected.
> [sdkconfig.log](https://github.com/espressif/idf-eclipse-plugin/files/3828968/sdkconfig.log)

It looks like, it has a lot of other plugin errors - not sure if the workspace is corrupted. I've still not updated the url - have you tried with the local build which is provided above? and provide the json server console log if you're having any issue after that.

kolipakakondal, 2019-11-11:
sdkconfig editor launch issue got fixed with the **idf-eclipse-plugin-v1.0.0-beta.3** release.

Please get the latest changes using the update site url.

closed 2019-11-11 by kolipakakondal

winkste, 2019-11-13:
Hi.

BETA3 still doesn't solve my issue opening the sdkconfig dialog. Please see my log below. Although installation of BETA 3 manually worked, the question is if I messed up my eclipse installation. I saw some weird entries in the log
[error.log](https://github.com/espressif/idf-eclipse-plugin/files/3842813/error.log)

SinglWolf, 2019-11-17:
HI!
When double clicking on the sdkconfig, it will prompt a dialog stating "Starting JSON Configuration Server" and the it will failed with a message on a new tab:

"Error retrieving output from the json configserver, please check the error log"

Error Log:
`eclipse.buildId=4.14.0.I20191101-0300
java.version=11.0.5
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=ru_RU
Framework arguments:  -product org.eclipse.epp.package.cpp.product -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -product org.eclipse.epp.package.cpp.product -data file:/C:/Users/Alexander/eclipse-workspace/ -product org.eclipse.epp.package.cpp.product

com.espressif.idf.sdk.config.ui
Error
Sun Nov 17 21:56:20 YEKT 2019
Invalid thread access

org.eclipse.swt.SWTException: Invalid thread access
	at org.eclipse.swt.SWT.error(SWT.java:4720)
	at org.eclipse.swt.SWT.error(SWT.java:4635)
	at org.eclipse.swt.SWT.error(SWT.java:4606)
	at org.eclipse.swt.widgets.Widget.error(Widget.java:452)
	at org.eclipse.swt.widgets.Widget.checkWidget(Widget.java:370)
	at org.eclipse.swt.widgets.Widget.checkParent(Widget.java:290)
	at org.eclipse.swt.widgets.Widget.<init>(Widget.java:155)
	at org.eclipse.swt.widgets.Control.<init>(Control.java:112)
	at org.eclipse.swt.widgets.Scrollable.<init>(Scrollable.java:85)
	at org.eclipse.swt.widgets.Composite.<init>(Composite.java:99)
	at com.espressif.idf.sdk.config.ui.SDKConfigurationEditor.createErrorPage(SDKConfigurationEditor.java:320)
	at com.espressif.idf.sdk.config.ui.SDKConfigurationEditor.lambda$0(SDKConfigurationEditor.java:153)
	at org.eclipse.jface.operation.ModalContext$ModalContextThread.run(ModalContext.java:122)
`
Please, help!