espressif/idf-eclipse-plugin #5: Install tools (Traceback (most recent call last))

kipergrof, 2019-10-07:
Hello!

I cant install tools. I get this error: 
[image]
[image]

Does anyone have an idea for a solution?
Use : 
-java 8
-eclipse 2019-09
-esp-idf v4 and v4.1

kolipakakondal, 2019-10-11:
Can you run the below command by replacing it with your esp-idf directory? Does it output the version v4.x version number?

> git --work-tree=/Users/kondal/esp/esp-idf describe --tags

danak6jq, 2019-10-18:
I'm getting an install-tools traceback as well:

Installing tools...
This can take a while. Please be patient.
Executing C:\Users\Dana\AppData\Local\Programs\Python\Python37\python.exe C:\Users\Dana\Desktop\esp-idf\tools\idf_tools.py install all
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
Skipping xtensa-esp32-elf@esp32-2019r1-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190708 (already installed)
Skipping mconf@v4.6.0.0-idf-20190628 (already installed)
Skipping ninja@1.9.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)

Executing C:\Users\Dana\AppData\Local\Programs\Python\Python37\python.exe C:\Users\Dana\Desktop\esp-idf\tools\idf_tools.py install-python-env
Traceback (most recent call last):
  File "C:\Users\Dana\Desktop\esp-idf\tools\idf_tools.py", line 1344, in <module>
    main(sys.argv[1:])
  File "C:\Users\Dana\Desktop\esp-idf\tools\idf_tools.py", line 1340, in main
    action_func(args)
  File "C:\Users\Dana\Desktop\esp-idf\tools\idf_tools.py", line 1139, in action_install_python_env
    idf_python_env_path, _, virtualenv_python = get_python_env_path()
  File "C:\Users\Dana\Desktop\esp-idf\tools\idf_tools.py", line 840, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '--work-tree=' + global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
  File "C:\Users\Dana\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "C:\Users\Dana\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 472, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\Dana\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "C:\Users\Dana\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 1178, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified

Executing C:\Users\Dana\AppData\Local\Programs\Python\Python37\python.exe C:\Users\Dana\Desktop\esp-idf\tools\idf_tools.py export --format=key-value
Traceback (most recent call last):
  File "C:\Users\Dana\Desktop\esp-idf\tools\idf_tools.py", line 1344, in <module>
    main(sys.argv[1:])
  File "C:\Users\Dana\Desktop\esp-idf\tools\idf_tools.py", line 1340, in main
    action_func(args)
  File "C:\Users\Dana\Desktop\esp-idf\tools\idf_tools.py", line 980, in action_export
    idf_python_env_path, idf_python_export_path, virtualenv_python = get_python_env_path()
  File "C:\Users\Dana\Desktop\esp-idf\tools\idf_tools.py", line 840, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '--work-tree=' + global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
  File "C:\Users\Dana\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "C:\Users\Dana\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 472, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\Dana\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "C:\Users\Dana\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 1178, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified

igrr, 2019-10-18:
Can you please run `git --work-tree C:\Users\Dana\Desktop\esp-idf describe --tags` from cmd.exe to verify that `git` works?

danak6jq, 2019-10-18:
Ah, from cmd.exe:
Microsoft Windows [Version 10.0.18362.10024]
(c) 2019 Microsoft Corporation. All rights reserved.

C:\Users\Dana>git --work-tree C:\Users\Dana\Desktop\esp-idf describe --tags
'git' is not recognized as an internal or external command,
operable program or batch file.

From ESP-IDF command prompt:

C:\Users\Dana\Desktop\esp-idf>git --work-tree=C:\Users\Dana\Desktop\esp-idf describe --tags
v4.1-dev-474-g2e6398aff

but it occurs to me that Eclipse is not running in the same environment, perhaps? Do I need a reboot?

danak6jq, 2019-10-19:
> but it occurs to me that Eclipse is not running in the same environment, perhaps? Do I need a reboot?

For what it is worth, reboot did not help, seeing the same results.

kolipakakondal, 2019-10-30:
v1.0.0-beta.2 is available now! please get the latest changes using the update site.

Issue related to PATH got fixed, Check the release notes here:
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.0.0-beta.2

closed 2019-10-30 by kolipakakondal

danak6jq, 2019-10-30:
On 10/30/2019 6:47 AM, Kondal Kolipaka wrote:
>
> Closed #5 <https://github.com/espressif/idf-eclipse-plugin/issues/5>.
>
>

I still can not install the tools and get a working Eclipse IDE using the
documented directions. I only end up with IDF_PATH set in my
C/C++ Build Environment.

What log files do you need?

vcmorini, 2020-01-07:
Same here, stuck in:
Python requirements from C:\Users\vcmorini\AppData\Local\esp-idf\requirements.txt are satisfied.
Checking Python dependencies...
Running ninja in directory C:\Users\vcmorini\esp\hello_world\build
Executing "ninja all"...
Traceback (most recent call last):
  File "C:/Users/vcmorini/AppData/Local/esp-idf/tools/idf.py", line 642, in <module>
    main()
  File "C:/Users/vcmorini/AppData/Local/esp-idf/tools/idf.py", line 621, in main
    execute_action(actions[0], actions[1:])
  File "C:/Users/vcmorini/AppData/Local/esp-idf/tools/idf.py", line 605, in execute_action
    execute_action(dep, remaining_actions)
  File "C:/Users/vcmorini/AppData/Local/esp-idf/tools/idf.py", line 615, in execute_action
    function(action, args)
  File "C:/Users/vcmorini/AppData/Local/esp-idf/tools/idf.py", line 274, in build_target
    _run_tool(generator_cmd[0], generator_cmd + [target_name], args.build_dir)
  File "C:/Users/vcmorini/AppData/Local/esp-idf/tools/idf.py", line 86, in _run_tool
    subprocess.check_call(args, env=os.environ, cwd=cwd)
  File "C:\msys32\mingw32\lib\python2.7/subprocess.py", line 186, in check_call
    retcode = call(*popenargs, **kwargs)
  File "C:\msys32\mingw32\lib\python2.7/subprocess.py", line 173, in call
    return Popen(*popenargs, **kwargs).wait()
  File "C:\msys32\mingw32\lib\python2.7/subprocess.py", line 325, in __init__
    errread, errwrite)
  File "C:\msys32\mingw32\lib\python2.7/subprocess.py", line 575, in _execute_child
    startupinfo)
WindowsError: [Error 2] The system cannot find the file specified

And my selected python is 3.7.

I've strictly followed the doc: https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md#espressif-idf-eclipse-plugins

kfine100, 2020-05-02:
Is there any progress on this question? I am having the same issue installing ESP32 tools in Eclipse. I have Python 3.8 installed. Error is same as for others above:

Executing C:\Users\Kevin\AppData\Local\Programs\Python\Python38-32\python.exe D:\ESP-IDF\tools\idf_tools.py export --format=key-value
Traceback (most recent call last):
  File "D:\ESP-IDF\tools\idf_tools.py", line 1372, in <module>
    main(sys.argv[1:])
  File "D:\ESP-IDF\tools\idf_tools.py", line 1368, in main
    action_func(args)
  File "D:\ESP-IDF\tools\idf_tools.py", line 1008, in action_export
    idf_python_env_path, idf_python_export_path, virtualenv_python = get_python_env_path()
  File "D:\ESP-IDF\tools\idf_tools.py", line 846, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', 'describe', '--tags'],
  File "C:\Users\Kevin\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\Kevin\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\Kevin\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\Kevin\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified

kolipakakondal, 2020-05-04:
Hi @kfine100 

> idf_version_str = subprocess.check_output(['git', 'describe', '--tags'],

Looks like git is not their in the PATH, could you please verify the Eclipse CDT build environment PATH variable value.

`Preferences > C/C++ Build > Environment`

kfine100, 2020-05-04:
Thanks for your reply. I took a look at the environment varibles, but could not find which one specifically refers to the git executable. I know that I correctly entered that location during the Eclipse Tools installation dialog.

I am not too motivated to work further because I was able to get around this problem by simply creating a version.txt file with "v4.0" and adding it to the ESP-IDF directory.

But I appreciate your help!

mirzafahad, 2020-09-29:
I had the same problem as @kfine100. Running the eclipse as "Administrator" solved it.