espressif/idf-eclipse-plugin #6: Infinite loop on building with eclipse

Boldie, 2019-10-10:
I got an infinite loop while trying to build with eclipse. On console everything works as expected.

```Building in: /home/sven/workspace/Water Me/build
cmake --build . -- -v
CMake Error: The current CMakeCache.txt directory /home/sven/workspace/Water Me/build/CMakeCache.txt is different than the directory /home/sven/workspace/esp/water-me/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
[0/1] /usr/bin/cmake -H/home/sven/workspace/esp/water-me -B/home/sven/workspace/esp/water-me/build
-- Component directory /home/sven/workspace/esp/esp-idf/components/aws_iot does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/sven/workspace/esp/esp-idf/components/micro-ecc does not contain a CMakeLists.txt file. No component will be added
/usr/lib/git-core/git-submodule: 7: /usr/lib/git-core/git-submodule: basename: not found
/usr/lib/git-core/git-submodule: 7: /usr/lib/git-core/git-submodule: sed: not found
/usr/lib/git-core/git-submodule: 86: /usr/lib/git-core/git-sh-setup: basename: not found
/usr/lib/git-core/git-submodule: 86: /usr/lib/git-core/git-sh-setup: sed: not found
/usr/lib/git-core/git-submodule: 332: /usr/lib/git-core/git-sh-setup: uname: not found
-- Project version: 684fab7-dirty
-- Building ESP-IDF components for target esp32
-- Adding linker script /home/sven/workspace/esp/water-me/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/sven/workspace/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/sven/workspace/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/sven/workspace/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/sven/workspace/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/sven/workspace/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/sven/workspace/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/sven/workspace/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt coap console cxx driver efuse esp-tls esp32 esp32-wifi-manager esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mkspiffs mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc soc spi_flash spidriver spiffs spiffs_image tcp_transport tcpip_adapter tft ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/sven/workspace/esp/esp-idf/components/app_trace /home/sven/workspace/esp/esp-idf/components/app_update /home/sven/workspace/esp/esp-idf/components/asio /home/sven/workspace/esp/esp-idf/components/bootloader /home/sven/workspace/esp/esp-idf/components/bootloader_support /home/sven/workspace/esp/esp-idf/components/bt /home/sven/workspace/esp/esp-idf/components/coap /home/sven/workspace/esp/esp-idf/components/console /home/sven/workspace/esp/esp-idf/components/cxx /home/sven/workspace/esp/esp-idf/components/driver /home/sven/workspace/esp/esp-idf/components/efuse /home/sven/workspace/esp/esp-idf/components/esp-tls /home/sven/workspace/esp/esp-idf/components/esp32 /home/sven/workspace/esp/water-me/components/esp32-wifi-manager /home/sven/workspace/esp/esp-idf/components/esp_adc_cal /home/sven/workspace/esp/esp-idf/components/esp_common /home/sven/workspace/esp/esp-idf/components/esp_eth /home/sven/workspace/esp/esp-idf/components/esp_event /home/sven/workspace/esp/esp-idf/components/esp_gdbstub /home/sven/workspace/esp/esp-idf/components/esp_http_client /home/sven/workspace/esp/esp-idf/components/esp_http_server /home/sven/workspace/esp/esp-idf/components/esp_https_ota /home/sven/workspace/esp/esp-idf/components/esp_https_server /home/sven/workspace/esp/esp-idf/components/esp_local_ctrl /home/sven/workspace/esp/esp-idf/components/esp_ringbuf /home/sven/workspace/esp/esp-idf/components/esp_rom /home/sven/workspace/esp/esp-idf/components/esp_websocket_client /home/sven/workspace/esp/esp-idf/components/esp_wifi /home/sven/workspace/esp/esp-idf/components/espcoredump /home/sven/workspace/esp/esp-idf/components/esptool_py /home/sven/workspace/esp/esp-idf/components/expat /home/sven/workspace/esp/esp-idf/components/fatfs /home/sven/workspace/esp/esp-idf/components/freemodbus /home/sven/workspace/esp/esp-idf/components/freertos /home/sven/workspace/esp/esp-idf/components/heap /home/sven/workspace/esp/esp-idf/components/idf_test /home/sven/workspace/esp/esp-idf/components/jsmn /home/sven/workspace/esp/esp-idf/components/json /home/sven/workspace/esp/esp-idf/components/libsodium /home/sven/workspace/esp/esp-idf/components/log /home/sven/workspace/esp/esp-idf/components/lwip /home/sven/workspace/esp/water-me/main /home/sven/workspace/esp/esp-idf/components/mbedtls /home/sven/workspace/esp/esp-idf/components/mdns /home/sven/workspace/esp/water-me/components/indirectBuild/ESP32_TFT_library/components/mkspiffs /home/sven/workspace/esp/esp-idf/components/mqtt /home/sven/workspace/esp/esp-idf/components/newlib /home/sven/workspace/esp/esp-idf/components/nghttp /home/sven/workspace/esp/esp-idf/components/nvs_flash /home/sven/workspace/esp/esp-idf/components/openssl /home/sven/workspace/esp/esp-idf/components/partition_table /home/sven/workspace/esp/esp-idf/components/protobuf-c /home/sven/workspace/esp/esp-idf/components/protocomm /home/sven/workspace/esp/esp-idf/components/pthread /home/sven/workspace/esp/esp-idf/components/sdmmc /home/sven/workspace/esp/esp-idf/components/soc /home/sven/workspace/esp/esp-idf/components/spi_flash /home/sven/workspace/esp/water-me/components/indirectBuild/ESP32_TFT_library/components/spidriver /home/sven/workspace/esp/water-me/components/indirectBuild/ESP32_TFT_library/components/spiffs /home/sven/workspace/esp/water-me/components/indirectBuild/ESP32_TFT_library/components/spiffs_image /home/sven/workspace/esp/esp-idf/components/tcp_transport /home/sven/workspace/esp/esp-idf/components/tcpip_adapter /home/sven/workspace/esp/water-me/components/indirectBuild/ESP32_TFT_library/components/tft /home/sven/workspace/esp/esp-idf/components/ulp /home/sven/workspace/esp/esp-idf/components/unity /home/sven/workspace/esp/esp-idf/components/vfs /home/sven/workspace/esp/esp-idf/components/wear_levelling /home/sven/workspace/esp/esp-idf/components/wifi_provisioning /home/sven/workspace/esp/esp-idf/components/wpa_supplicant /home/sven/workspace/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sven/workspace/esp/water-me/build
[0/1] /usr/bin/cmake -H/home/sven/workspace/esp/water-me -B/home/sven/workspace/esp/water-me/build
-- Component directory /home/sven/workspace/esp/esp-idf/components/aws_iot does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/sven/workspace/esp/esp-idf/components/micro-ecc does not contain a CMakeLists.txt file. No component will be added
/usr/lib/git-core/git-submodule: 7: /usr/lib/git-core/git-submodule: basename: not found
/usr/lib/git-core/git-submodule: 7: /usr/lib/git-core/git-submodule: sed: not found
/usr/lib/git-core/git-submodule: 86: /usr/lib/git-core/git-sh-setup: basename: not found
/usr/lib/git-core/git-submodule: 86: /usr/lib/git-core/git-sh-setup: sed: not found
/usr/lib/git-core/git-submodule: 332: /usr/lib/git-core/git-sh-setup: uname: not found
-- Project version: 684fab7-dirty
-- Building ESP-IDF components for target esp32
-- Adding linker script /home/sven/workspace/esp/water-me/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/sven/workspace/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/sven/workspace/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/sven/workspace/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/sven/workspace/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/sven/workspace/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
[... cut ...] tft /home/sven/workspace/esp/esp-idf/components/ulp /home/sven/workspace/esp/esp-idf/components/unity /home/sven/workspace/esp/esp-idf/components/vfs /home/sven/workspace/esp/esp-idf/components/wear_levelling /home/sven/workspace/esp/esp-idf/components/wifi_provisioning /home/sven/workspace/esp/esp-idf/components/wpa_supplicant /home/sven/workspace/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sven/workspace/esp/water-me/build
```

Boldie, 2019-10-10:
I also fixed the PATH variable in Preferences (why the heck is this being modified, it also breaks my normal C++ projects), but now it is still an infinite loop.

Boldie, 2019-10-10:
Maybe this problem is related to the following problem: 
CMake Error: The current CMakeCache.txt directory /home/sven/workspace/Water Me/build/

It seems, the path is not correctly set here, my files are here: /home/sven/workspace/esp/water-me/build

"Water Me" is the the project name which shall never be used to create paths.

kolipakakondal, 2019-10-14:
Thanks for reporting. We will check this.

BadrBouaddi, 2019-10-15:
any new inputs about this problem, i produce the same log

rdkluijv, 2019-10-28:
Got this one as well. Killed Eclipse with the task manager. (Windows) Removed the content of the build directory in the workspace. Now it builds.
Problem may be caused because it was a template build before in the CMD environment.

winkste, 2019-11-13:
Hi.

Tested with BETA3 and still have the same issue. I didn't understand the last post. Which files from the build directory need to be deleted?

Screenshot of infinite build:
[image]

Project file list:
[image]

winkste, 2019-11-14:
So, after fixing some issues I was able to get the project build and also the sdkconfig opens. Let me try to recap it, have been some manual steps:
- trying to open the sdkconfig generated an error message and that the project was still build for the template, I had to execute /Volumes/esp/esp-idf/tools/idf.py fullclean (shown in console)
- after this the project no longer loops forever for the build but I got the message that ninja is not available
- checking the path variable in eclipse -> environment shows this path really was missing, I added it manually
- now I got errors regarding my python package installation: missing components. installing with the install.sh script as mentioned in the console of eclipse didn't solve the problem. running: "/usr/bin/python -m pip install --user -r /Volumes/esp/esp-idf/requirements.txt" fixed it and now I'm able to build the HelloWorld template and the sdkconfig dialog is showing up.

Hope that helps others as well...

dolhastelian, 2020-03-25:
same for me
manual build from console works fine
build from eclipse triggers a 100 x build loop then it stops
deleting the build folder and rebuild from eclipse works fine also for me

kolipakakondal, 2020-03-26:
This seems to happen when you import an existing project(IDF Make project) which was already built on a console. Please delete the build folder before importing it into the eclipse workspace.

kolipakakondal, 2020-06-03:
Hi All, We have fixed this issue with the v1.1.0 release. Please get the latest version using the update site.

Check this here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.1.0

closed 2020-06-03 by kolipakakondal