espressif/idf-eclipse-plugin #7: Import of existing project will destroy some files.

Boldie, 2019-10-10:
After importing an existing project, some files were truncated (content lost, 0 byte file size). These are:
partitions.csv, sdkconfig, CMakeList.txt, ...

Boldie, 2019-10-10:
After resetting it via git, the project can be built, but there are no index being built, which results in unsresolved includes and such things.

My conclusion: 
- workspace handling seems to be problematic, the plugin does not handle workspaces and directrories correctly (if one has a project not in the workspace root directory will break)
- Project name is used for path generation, ???? why not using the project path.
- C++ index not working nor I can correct it.
- Path variable globally modified which breaks other C++ projects.
- Plugin modifies too many files, also .gitignore will be modified etc. Do not add so much magic, because there may be more trouble than improvement.
-> Will deinstall it, unuseable for the moment.

kolipakakondal, 2020-03-13:
Hello, With the recent ranges, by default project will not be copied into the eclipse workspace unless you select "copy into workspace" option. With that, you don't see any changes in the existing project except adding a few additional resources like .project and .settings, etc. which are eclipse artifacts. Otherwise, it's just linking the project to the workspace.

Please update to the latest build and let us know if you see any difficulties.

kolipakakondal, 2020-06-03:
Hi, We have released v1.1.0, please get the latest using the update site.

Check this here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.1.0

Closing the issue as there is no update for quite some time. Feel free to reopen/create a new issue if you see any other problems. Thanks

closed 2020-06-03 by kolipakakondal