espressif/idf-eclipse-plugin #8: Windows 10 Ecplipse problem

aleism2007, 2019-10-11:
Hi, my name is Alejandro,

I have done the tutorial "Espressif IDF Eclipse Plugins". but when I get to the step "Installing ESP-IDF Tools", I encounter the following error in the Eclipse console:

---------------------------------------------------------------------------------------------------

Installing tools...
This can take a while. Please be patient.
Executing D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\python.exe C:\Users\Alejandro\Desktop\esp-idf\tools\idf_tools.py install all
D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\python.exe: can't open file 'C:\Users\Alejandro\Desktop\esp-idf\tools\idf_tools.py': [Errno 2] No such file or directory

Executing D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\python.exe C:\Users\Alejandro\Desktop\esp-idf\tools\idf_tools.py install-python-env
D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\python.exe: can't open file 'C:\Users\Alejandro\Desktop\esp-idf\tools\idf_tools.py': [Errno 2] No such file or directory

Executing D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\python.exe C:\Users\Alejandro\Desktop\esp-idf\tools\idf_tools.py export --format=key-value
D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\python.exe: can't open file 'C:\Users\Alejandro\Desktop\esp-idf\tools\idf_tools.py': [Errno 2] No such file or directory

------------------------------------------------------------------------------------------------

could someone help me solve the problem?

Thanks.

kolipakakondal, 2019-10-11:
To troubleshoot, can you please run the below command from the terminal and see what happens? Do you see the same issue?  If yes, can you please provide the complete console message.

> D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\python.exe C:\Users\Alejandro\Desktop\esp-idf\tools\idf_tools.py install all

`D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\python.exe`  - Does this path exist?  Seems to be something wrong with the folder name of `Python37-32`

kolipakakondal, 2019-10-11:
While we check, can you try with a 64-bit Python version?

aleism2007, 2019-10-11:
this is the result:

C:\Users\Alejandro>D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\python.exe C:\Users\Alejandro\Desktop\esp-idf\tools\idf_tools.py install all
D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\python.exe: can't open file 'C:\Users\Alejandro\Desktop\esp-idf\tools\idf_tools.py': [Errno 2] No such file or directory

aleism2007, 2019-10-11:
hi, the same with Python 64-bit

C:\Users\Alejandro>C:\Users\Alejandro\AppData\Local\Programs\Python\Python37\python.exe C:\Users\Alejandro\Desktop\esp-idf\tools\idf_tools.py install all
C:\Users\Alejandro\AppData\Local\Programs\Python\Python37\python.exe: can't open file 'C:\Users\Alejandro\Desktop\esp-idf\tools\idf_tools.py': [Errno 2] No such file or directory

aleism2007, 2019-10-11:
sorry but idf_tools.py is not in the tools folder, but i don't no why

kolipakakondal, 2019-10-11:
Make sure you're using esp-idf 4.x - https://github.com/espressif/esp-idf/releases/tag/v4.0-beta1

aleism2007, 2019-10-11:
Yes, you are right, that was the problem, it is already solved.
I take the opportunity to ask you why I can only "build" only once, if I close the project and open it again it appears in the console: "Build not configured correctly"

kolipakakondal, 2019-10-14:
I'm able to reproduce the issue. Until we fix this, as a workaround - you can restart the eclipse after opening the project. This should work.

aleism2007, 2019-10-14:
Thanks a lot,.

Have a nice day.

closed 2019-10-15 by aleism2007