espressif/idf-eclipse-plugin #10: Prerequisites - ESP-IDF 4.0 and above

versamodule, 2019-10-14:
From the install page
https://github.com/espressif/idf-eclipse-plugin

Under Prerequisites, it says:
ESP-IDF 4.0 and above : Follow the download instructions from [here](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#step-2-get-esp-idf)

That page tells you to download the " ESP-IDF Tools Installer ".  Yet when you run that there is no 4.0 option.

aleism2007, 2019-10-14:
this is the link: https://github.com/espressif/esp-idf/releases/tag/v4.0-beta1

versamodule, 2019-10-14:
yes, but the " ESP-IDF Tools Installer " does not pick that up. So I was just mentioning it as it might confuse people to say 4.0 and above.

kolipakakondal, 2019-10-15:
> yes, but the " ESP-IDF Tools Installer " does not pick that up. So I was just mentioning it as it might confuse people to say 4.0 and above.

"ESP-IDF Tools Installer" won't be used if you're using Eclipse.

Make sure you set up IDF v4.x  https://github.com/espressif/esp-idf/releases/tag/v4.0-beta1
and proceed with the rest of the installation process from here https://github.com/espressif/idf-eclipse-plugin#espressif-idf-eclipse-plugins

kolipakakondal, 2019-10-15:
> Under Prerequisites, it says:
> ESP-IDF 4.0 and above : Follow the download instructions from [here](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#step-2-get-esp-idf) [...]

I'll update the documentation to point to the esp-idf github link directly. https://github.com/espressif/esp-idf/releases

Thanks for reporting

igrr, 2019-10-15:
We've also enabled the `release/v4.0` branch in the installer. The actual `v4.0` option will also appear when this version is released.

kolipakakondal, 2019-10-30:
v1.0.0-beta.2 is available now! please get the latest changes using the update site.

Check more here:
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.0.0-beta.2

closed 2019-10-30 by kolipakakondal