espressif/idf-eclipse-plugin #11: idf-eclipse-plugin-v1.0.0-beta,  403 Forbidden

versamodule, 2019-10-14:
From this page here:
https://github.com/espressif/idf-eclipse-plugin/releases

The following on that page takes you to a 403 Forbidden page

> Installing IDF Plugins using update site URL:
> https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

kolipakakondal, 2019-10-15:
> The following on that page takes you to a 403 Forbidden page
> 
> > Installing IDF Plugins using update site URL:
> > https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

That's expected. The update site URL still works if you use it from the eclipse. Just that update site url doesn't have any welcome page to display.

Thanks for reporting.

versamodule, 2019-10-15:
> That's expected. The update site URL still works if you use it from the eclipse.

But it doesn't.  Eclipse is having a issue with that.

kolipakakondal, 2019-10-15:
Could you please post the error?

versamodule, 2019-10-15:
I should have before I wiped everything and restored my computer.
All i can say is after you follow the directions for "Installing IDF 
Plugins using update site url"
The Cmake will report and error while it does grab the other one.

Sincerely,

Jimp@Versamodule.com

igrr, 2019-10-15:
What happens if you select the "Show the original error..." option? Is there any specific error which explains why the plugin will not be installed?

versamodule, 2019-10-15:
[image]

kolipakakondal, 2019-10-16:
What is the Eclipse version? It should be Eclipse CDT Package 2018-09 and above. com.google.gson 2.8.2 is not available in your eclipse, my guess is that your Eclipse version is either older or not CDT package.

versamodule, 2019-10-16:
> What is the Eclipse version? It should be Eclipse CDT Package 2018-09 and above. com.google.gson 2.8.2 is not available in your eclipse, my guess is that your Eclipse version is either older or not CDT package.

The one i have been using here after my computer restore is an older one (Eclipse Oxygen).  I used it here to show you about the error rather than re-downloading the latest one. But when i started all of this i downloaded and tryed the latest one. It too had the same issue of not downloading everything.  Though I never looked at the error, so I don't know if that is the same.  Ether way, I was just letting you know there are issues. I don't plan on using eclipse for ESP.  Visual Studio is far better in my opinion.

closed 2019-10-16 by versamodule