espressif/idf-eclipse-plugin #9: Eclipse not flashing project

Bertoleti, 2019-10-14:
I've installed the plug in into Eclipse and I'm building the project successfully.
But when I try to flash/monitor the program, I have a issue.

Someone can help me to understand this problem, please?

The logs are bellow:

Traceback (most recent call last):
  File "/Users/macmini/esp/esp-idf/tools/idf.py", line 1556, in <module>
Adding "flash"'s dependency "all" to list of commands with default set of options.
Executing action: all (aliases: build)
Running ninja in directory /Users/macmini/eclipse-workspace/Wifi_Example/build
Executing "ninja all"...
    main()
  File "/Users/macmini/esp/esp-idf/tools/idf.py", line 1471, in main
    cli(prog_name=PROG)
  File "/Users/macmini/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/Users/macmini/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/Users/macmini/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages/click/core.py", line 1164, in invoke
    return _process_result(rv)
  File "/Users/macmini/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages/click/core.py", line 1102, in _process_result
    **ctx.params)
  File "/Users/macmini/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Users/macmini/esp/esp-idf/tools/idf.py", line 1096, in execute_tasks
    task.run(ctx, global_args, task.action_args)
  File "/Users/macmini/esp/esp-idf/tools/idf.py", line 687, in run
    self.callback(self.name, context, global_args, **action_args)
  File "/Users/macmini/esp/esp-idf/tools/idf.py", line 328, in build_target
    _run_tool(generator_cmd[0], generator_cmd + [target_name], args.build_dir)
  File "/Users/macmini/esp/esp-idf/tools/idf.py", line 104, in _run_tool
    subprocess.check_call(args, env=os.environ, cwd=cwd)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 535, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 522, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

kolipakakondal, 2019-10-14:
Have you selected the correct serial port for your particular board? Can you post your  PATH value from the eclipse build environment preference page

Bertoleti, 2019-10-14:
Yes.
When I flash programm by terminal commands, I have success.

I noticed that python path used by eclipse it's version 2.7, but I have installed version 3.7 (as required to installation). But when i run the command "./install.sh" the folder that appears in ".espressif/pyhton_env/" is idf4.0_py2.7_env and not idf4.0_py3.7_env .

I think that's may be a problem as well, but I can't use python 3.7 in that path.

kolipakakondal, 2019-10-15:
I believe **idf4.0_py2.7_env** shouldn't a problem. Eclipse uses the python path which is installed as part of the "Install Tools" process - which is ".espressif/pyhton_env"

To troubleshoot this issue further, I need to have the complete "Console View" log https://github.com/espressif/idf-eclipse-plugin#console-view-log

Bertoleti, 2019-10-25:
Hi kolipakakondal.

Sorry for the delay.

I've installed MacOS again and I did the process.

Now its working!!!

I don't know what I did wrong in the first time, but I appreciate your fast response to help me.

Thanks.

kolipakakondal, 2019-10-30:
Closing the issue hence it's fixed.

v1.0.0-beta.2 is available now! please get the latest changes using the update site. 

Check more here:
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.0.0-beta.2

closed 2019-10-30 by kolipakakondal