espressif/idf-eclipse-plugin #12: Building tools not working with latest python/pip/git

rdkluijv, 2019-10-16:
Hi,

On a windows 10 system with the latest python/git/pip I can't build the tools in eclipse 19-09. I can build them with the install and export batch files with the 'default' tools and old versions. I noticed in the console that the path to python seems to be hardcoded. Python is installed for all users in Roaming not Local. The environmental PATH is correctly set. The path to the python file is correct. I have another windows 10 system working so for me it's a minor issue. 
Thanks for the great work.

Regards,

Rob.

kolipakakondal, 2019-10-17:
Hi Rob, What is the python is version? Python path is picked from the Windows registry - can you verify the registry?

```
HKEY_CURRENT_USER\SOFTWARE\Python\PythonCore\
HKEY_LOCAL_MACHINE\SOFTWARE\Python\PythonCore\
```

kolipakakondal, 2019-10-17:
Check if the issue is similar to https://github.com/espressif/idf-eclipse-plugin/issues/14

rdkluijv, 2019-10-17:
Not similar but very related. It's also related to another issue. I'll explain.
On the working win10 PC I run the ESP IDF tool with all the default settings. No python and git where installed. Followed the instructions and got a working Hello World. Then I installed Eclipse but instead of  Hello World I used the Blink template. I order to get that working I had to change the pin number to 2 in the configuration. That didn't work. Didn't check the issues then so I didn't know where the problem was. I edited the configuration file manually and then Blink worked and the configuration file was sorted. I figured that me first installing the ESP IDF tool destroyed the setup of the plugin. So I took another system and followed the instructions in the readme for installation. Got the latest Python, Git, PIP and ESP-IDF. Then I got issue #14 but I knew that it worked with the previous version and at that time I know that the configuration was an issue. So my solution was uninstalling the latest software and use the defaulf ESP IDF installer. My guess is that uninstalling the latest Python is not working as I expect. I had to manually delete the directory after the install. Most probably the registry is also not cleaned with uninstall. I'll check and adjust the registry entries.

igrr, 2019-10-17:
The last error seems to be related to missing `git.exe`. Can you please check if you can run `git.exe` from the Windows  command prompt (cmd.exe)? If no, can you please go through the Git for Windows installation again, selecting "add Git to PATH" option this time?

rdkluijv, 2019-10-17:
Git runs in cmd. It's in the PATH and in the registry. See file.

[git.txt](https://github.com/espressif/idf-eclipse-plugin/files/3740987/git.txt)

This is my path where the esp-idf is cloned: C:\Users\rdkluijv\esp

Have to change eclipse paths. It's a mesh! Can I uninstall the plugin, remove the paths and install again?

IDF_PATH = C:\Users\rdkluijv\esp\
IDF_PYTHON_ENV_PATH = C:\Users\rdkluijv\.espressif\python_env\idf4.1_py3.7_env
OPENOCD_SCRIPTS = C:\Users\rdkluijv\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190708\openocd-esp32\share\openocd\scripts
PATH = C:\Users\rdkluijv\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin;C:\Users\rdkluijv\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin;C:\Users\rdkluijv\.espressif\tools\cmake\3.13.4\bin;C:\Users\rdkluijv\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190708\openocd-esp32\bin;C:\Users\rdkluijv\.espressif\tools\mconf\v4.6.0.0-idf-20190628\;C:\Users\rdkluijv\.espressif\tools\ninja\1.9.0\;C:\Users\rdkluijv\.espressif\tools\idf-exe\1.0.1\;C:\Users\rdkluijv\.espressif\tools\ccache\3.7\;C:\Users\rdkluijv\.espressif\python_env\idf4.1_py3.7_env\Scripts;C:\Users\rdkluijv\esp\tools;%PATH%;C:\Users\rdkluijv\esp\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin;C:\Users\rdkluijv\esp\esp-idf\tools;C:\Users\rdkluijv\.espressif\python_env\idf4.1_py3.8_env\Scripts

kolipakakondal, 2019-10-18:
From your PATH I could see %PATH% variable is not expanded - That could be causing an issue. This issue is already identified and fixed - we still need to update it in the github.

In the meantime, can you replace **%PATH%** with the PATH value from the terminal? and see it helps.
> $ echo %PATH%

rdkluijv, 2019-10-18:
Actions taken:
Deleted all paths in eclipse related to the esp-idf
Run install tools to create fresh paths
Deleted the not expanded %PATH%
Concatenated the system PATH manually
Result:
A building blink program
Not able to change the configuration

Thanks for the help

rdkluijv, 2019-10-20:
Should I close the issue?

kolipakakondal, 2019-10-22:
Please keep it open, will close this issue once we have a fix for the "PATH% variable is not expanded" issue in the GitHub and update site

kolipakakondal, 2019-10-30:
This is fixed with the latest v1.0.0-beta.2, please get the latest changes using the update site. Since this issue is related to PATH, make sure you delete all the build environment variables and re-run the "install tools" from the eclipse to configure the PATH again.

closed 2019-10-30 by kolipakakondal