espressif/idf-eclipse-plugin #13: Unresolved inclusion - mac os

simap, 2019-10-16:
using open jdk 13 on mac 10.14.6
fresh checkout of esp idf v4.0-beta1
fresh install of eclipse 2019-09
fresh plugin install
installed tools using tools manager

I created a new project from hello_world example. I'm getting a bunch of "unresolved inclusion" and red highlights all over the editor. Of note, "#include <stdio.h>" is found, and when I open this file it's pulling it from this path: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/stdio.h

Building fails, and is missing PATH for ninja. The other tools are there. 

Fixing PATH lets me build, but editor is still buggered.

I've tried going through the manual steps mentioned, but nothing seems to fix Eclipse's editor. I'm pretty sure its not using xtensa gcc when its generating these editor errors, and I would expect it to jump to the stdio.h from xtensa's gcc, not the system SDK.

[image]

Any help would be appreciated!

AwkBot, 2019-10-16:
It is a common issue for me as well, just compile the code and Indexer will resolve all symbols.

simap, 2019-10-16:
@AwkBot  I have compiled it (you can see it in the screen shot) but it doesn't resolve the editor issues. Restarting, rebuilding indexes, cleaning, rebuilding, refreshing, etc., don't clear it up.
Rebuilding the c++ index shows this:
```
Indexed 'asdf' (9 sources, 34 headers) in 0.925 sec: 219 declarations; 2,527 references; 4 unresolved inclusions; 15 syntax errors; 13 unresolved names (0.47%)
```

For whatever reason, it's using the mac os toolchain to parse it. After poking around more, I can generate a parser log (attached). Of note, there's a ton of mac and iphone defines, and not any mention of xtensa anywhere. 

[parser.log](https://github.com/espressif/idf-eclipse-plugin/files/3736262/parser.log)

simap, 2019-10-16:
AH!
So the trick is that even though builds work, the ["launch target" stuff](https://github.com/espressif/idf-eclipse-plugin#configuring-launch-target) is absolutely necessary. Once that is make, selected, and a rebuild happens, the editor clears up.

closed 2019-10-16 by simap