espressif/idf-eclipse-plugin #14: Error pip dependencies installation Windows 10

jmpmscorp, 2019-10-16:
Hello.

Trying to work with new eclipse plugin, I can't get it. Following readme.md guide, I have everything installed and tools installation console output doesn't show any error. However, when I try to compile any example, console shows that 'requiments.txt' dependencies can't be found.

Trying with 'install.bat' or 'C:\\Users\\JM\\.espressif\\python_env\\idf4.0_py3.8_env\\Scripts\\python.exe -m pip install -r D:\\esp-idf\\requirements.txt' on windows cmd, cryptography dependency shows error.

First error is related with Microsoft C++ Build Tools 14.0. I resolved it adding C++ tools into my Visual Studio 2019 installation. But now, I have another error I can't resolve:

```
C:\Windows\system32>C:\\Users\\JM\\.espressif\\python_env\\idf4.0_py3.8_env\\Scripts\\python.exe -m pip install -r D:\\esp-idf\\requirements.txt
Requirement already satisfied: setuptools in c:\users\jm\.espressif\python_env\idf4.0_py3.8_env\lib\site-packages (from -r D:\\esp-idf\\requirements.txt (line 4)) (41.4.0)
Requirement already satisfied: click>=5.0 in c:\users\jm\.espressif\python_env\idf4.0_py3.8_env\lib\site-packages (from -r D:\\esp-idf\\requirements.txt (line 8)) (7.0)
Requirement already satisfied: pyserial>=3.0 in c:\users\jm\.espressif\python_env\idf4.0_py3.8_env\lib\site-packages (from -r D:\\esp-idf\\requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in c:\users\jm\.espressif\python_env\idf4.0_py3.8_env\lib\site-packages (from -r D:\\esp-idf\\requirements.txt (line 10)) (0.18.0)
Collecting cryptography>=2.1.4
  Using cached https://files.pythonhosted.org/packages/c2/95/f43d02315f4ec074219c6e3124a87eba1d2d12196c2767fadfdc07a83884/cryptography-2.7.tar.gz
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in c:\users\jm\.espressif\python_env\idf4.0_py3.8_env\lib\site-packages (from -r D:\\esp-idf\\requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in c:\users\jm\.espressif\python_env\idf4.0_py3.8_env\lib\site-packages (from -r D:\\esp-idf\\requirements.txt (line 13)) (0.25)
Collecting six>=1.4.1
  Using cached https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Collecting asn1crypto>=0.21.0
  Using cached https://files.pythonhosted.org/packages/6b/8c/ff300484eca90b397d919408619fae479965bdd8a1df3d6d08d58e491da5/asn1crypto-1.2.0-py2.py3-none-any.whl
Collecting cffi!=1.11.3,>=1.8
  Using cached https://files.pythonhosted.org/packages/c9/a5/7f5b85daf1992172d7b84e2a17dd66b5a9577018fc6063d7bacfc0b97e14/cffi-1.13.0-cp38-cp38-win_amd64.whl
Processing c:\users\jm\appdata\local\pip\cache\wheels\f2\9a\90\de94f8556265ddc9d9c8b271b0f63e57b26fb1d67a45564511\pycparser-2.19-py2.py3-none-any.whl
Building wheels for collected packages: cryptography
  Building wheel for cryptography (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: 'C:\Users\JM\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe' 'C:\Users\JM\.espressif\python_env\idf4.0_py3.8_env\lib\site-packages\pip\_vendor\pep517\_in_process.py' build_wheel 'C:\Users\JM\AppData\Local\Temp\tmpojr1z10m'
       cwd: C:\Users\JM\AppData\Local\Temp\pip-install-dnuej4f1\cryptography
  Complete output (136 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.8
  creating build\lib.win-amd64-3.8\cryptography
  copying src\cryptography\exceptions.py -> build\lib.win-amd64-3.8\cryptography
  copying src\cryptography\fernet.py -> build\lib.win-amd64-3.8\cryptography
  copying src\cryptography\utils.py -> build\lib.win-amd64-3.8\cryptography
  copying src\cryptography\__about__.py -> build\lib.win-amd64-3.8\cryptography
  copying src\cryptography\__init__.py -> build\lib.win-amd64-3.8\cryptography
  creating build\lib.win-amd64-3.8\cryptography\hazmat
  copying src\cryptography\hazmat\_oid.py -> build\lib.win-amd64-3.8\cryptography\hazmat
  copying src\cryptography\hazmat\__init__.py -> build\lib.win-amd64-3.8\cryptography\hazmat
  creating build\lib.win-amd64-3.8\cryptography\x509
  copying src\cryptography\x509\base.py -> build\lib.win-amd64-3.8\cryptography\x509
  copying src\cryptography\x509\certificate_transparency.py -> build\lib.win-amd64-3.8\cryptography\x509
  copying src\cryptography\x509\extensions.py -> build\lib.win-amd64-3.8\cryptography\x509
  copying src\cryptography\x509\general_name.py -> build\lib.win-amd64-3.8\cryptography\x509
  copying src\cryptography\x509\name.py -> build\lib.win-amd64-3.8\cryptography\x509
  copying src\cryptography\x509\ocsp.py -> build\lib.win-amd64-3.8\cryptography\x509
  copying src\cryptography\x509\oid.py -> build\lib.win-amd64-3.8\cryptography\x509
  copying src\cryptography\x509\__init__.py -> build\lib.win-amd64-3.8\cryptography\x509
  creating build\lib.win-amd64-3.8\cryptography\hazmat\backends
  copying src\cryptography\hazmat\backends\interfaces.py -> build\lib.win-amd64-3.8\cryptography\hazmat\backends
  copying src\cryptography\hazmat\backends\__init__.py -> build\lib.win-amd64-3.8\cryptography\hazmat\backends
  creating build\lib.win-amd64-3.8\cryptography\hazmat\bindings
  copying src\cryptography\hazmat\bindings\__init__.py -> build\lib.win-amd64-3.8\cryptography\hazmat\bindings
  creating build\lib.win-amd64-3.8\cryptography\hazmat\primitives
  copying src\cryptography\hazmat\primitives\cmac.py -> build\lib.win-amd64-3.8\cryptography\hazmat\primitives
  copying src\cryptography\hazmat\primitives\constant_time.py -> build\lib.win-amd64-3.8\cryptography\hazmat\primitives
  copying src\cryptography\hazmat\primitives\hashes.py -> build\lib.win-amd64-3.8\cryptography\hazmat\primitives
  copying src\cryptography\hazmat\primitives\hmac.py -> build\lib.win-amd64-3.8\cryptography\hazmat\primitives
  copying src\cryptography\hazmat\primitives\keywrap.py -> build\lib.win-amd64-3.8\cryptography\hazmat\primitives
  copying src\cryptography\hazmat\primitives\padding.py -> build\lib.win-amd64-3.8\cryptography\hazmat\primitives
  copying src\cryptography\hazmat\primitives\poly1305.py -> build\lib.win-amd64-3.8\cryptography\hazmat\primitives
[... 84 lines cut ...]
  warning: no previously-included files found matching 'tox.ini'
  writing manifest file 'src\cryptography.egg-info\SOURCES.txt'
  running build_ext
  generating cffi module 'build\\temp.win-amd64-3.8\\Release\\_padding.c'
  creating build\temp.win-amd64-3.8
  creating build\temp.win-amd64-3.8\Release
  generating cffi module 'build\\temp.win-amd64-3.8\\Release\\_constant_time.c'
  generating cffi module 'build\\temp.win-amd64-3.8\\Release\\_openssl.c'
  building '_openssl' extension
  creating build\temp.win-amd64-3.8\Release\build
  creating build\temp.win-amd64-3.8\Release\build\temp.win-amd64-3.8
  creating build\temp.win-amd64-3.8\Release\build\temp.win-amd64-3.8\Release
  C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.23.28105\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Users\JM\AppData\Local\Programs\Python\Python38\include -IC:\Users\JM\AppData\Local\Programs\Python\Python38\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.23.28105\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /Tcbuild\temp.win-amd64-3.8\Release\_openssl.c /Fobuild\temp.win-amd64-3.8\Release\build\temp.win-amd64-3.8\Release\_openssl.obj
  _openssl.c
  build\temp.win-amd64-3.8\Release\_openssl.c(498): fatal error C1083: No se puede abrir el archivo incluir: 'openssl/opensslv.h': No such file or directory
  error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.23.28105\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2
  ----------------------------------------
  ERROR: Failed building wheel for cryptography
  Running setup.py clean for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
```
Can you point me to obtain a solution??

Thanks in advance.

jmpmscorp, 2019-10-17:
Auto answer...

It seems to be a problem with python 3.8. I downgraded to 3.7.5 and PIP requirements are satisfied. But I have now an error with bootloader :(. Console output:

```
Building in: E:\Code\esp-idf\wifi scan\build
cmake --build . -- -v
[1/281] C:\Users\JM\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe   -Iconfig -ID:/esp-idf/components/wear_levelling/include -ID:/esp-idf/components/wear_levelling/private_include -ID:/esp-idf/components/newlib/platform_include -ID:/esp-idf/components/freertos/include -ID:/esp-idf/components/heap/include -ID:/esp-idf/components/log/include -ID:/esp-idf/components/soc/esp32/include -ID:/esp-idf/components/soc/include -ID:/esp-idf/components/esp_rom/include -ID:/esp-idf/components/esp_common/include -ID:/esp-idf/components/xtensa/include -ID:/esp-idf/components/xtensa/esp32/include -ID:/esp-idf/components/esp32/include -ID:/esp-idf/components/driver/include -ID:/esp-idf/components/esp_ringbuf/include -ID:/esp-idf/components/esp_event/include -ID:/esp-idf/components/tcpip_adapter/include -ID:/esp-idf/components/lwip/include/apps -ID:/esp-idf/components/lwip/include/apps/sntp -ID:/esp-idf/components/lwip/lwip/src/include -ID:/esp-idf/components/lwip/port/esp32/include -ID:/esp-idf/components/lwip/port/esp32/include/arch -ID:/esp-idf/components/vfs/include -ID:/esp-idf/components/esp_wifi/include -ID:/esp-idf/components/esp_wifi/esp32/include -ID:/esp-idf/components/esp_eth/include -ID:/esp-idf/components/efuse/include -ID:/esp-idf/components/efuse/esp32/include -ID:/esp-idf/components/app_trace/include -ID:/esp-idf/components/spi_flash/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu++11 -fno-rtti -fno-exceptions -D_GNU_SOURCE -DIDF_VER=\"GIT-NOTFOUND\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/wear_levelling/CMakeFiles/__idf_wear_levelling.dir/crc32.cpp.obj -MF esp-idf\wear_levelling\CMakeFiles\__idf_wear_levelling.dir\crc32.cpp.obj.d -o esp-idf/wear_levelling/CMakeFiles/__idf_wear_levelling.dir/crc32.cpp.obj -c D:/esp-idf/components/wear_levelling/crc32.cpp
[2/281] cmd.exe /C "cd /D "E:\Code\esp-idf\wifi scan\build\bootloader" && C:\Users\JM\.espressif\tools\cmake\3.13.4\bin\cmake.exe --build ."
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-build bootloader/bootloader.elf bootloader/bootloader.bin bootloader/bootloader.map 
cmd.exe /C "cd /D "E:\Code\esp-idf\wifi scan\build\bootloader" && C:\Users\JM\.espressif\tools\cmake\3.13.4\bin\cmake.exe --build ."
[1/2] Linking C executable bootloader.elf
FAILED: bootloader.elf 
cmd.exe /C "cd . && C:\Users\JM\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/bootloader.elf.dir/project_elf_src.c.obj  -o bootloader.elf  esp-idf/soc/libsoc.a esp-idf/xtensa/libxtensa.a esp-idf/log/liblog.a esp-idf/micro-ecc/libmicro-ecc.a esp-idf/spi_flash/libspi_flash.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/main/libmain.a -Wl,--cref -Wl,--Map=E:/Code/esp-idf/wifi scan/build/bootloader/bootloader.map esp-idf/spi_flash/libspi_flash.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/spi_flash/libspi_flash.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/micro-ecc/libmicro-ecc.a esp-idf/soc/libsoc.a esp-idf/xtensa/libxtensa.a esp-idf/log/liblog.a esp-idf/soc/libsoc.a esp-idf/xtensa/libxtensa.a esp-idf/log/liblog.a -L D:/esp-idf/components/esp32/ld -T esp32.peripherals.ld D:/esp-idf/components/xtensa/esp32/libhal.a -Wl,--gc-sections -L D:/esp-idf/components/esp_rom/esp32/ld -T esp32.rom.ld -T esp32.rom.newlib-funcs.ld -T esp32.rom.libgcc.ld -L D:/esp-idf/components/bootloader/subproject/main -T esp32.bootloader.ld -T esp32.bootloader.rom.ld && cd ."
xtensa-esp32-elf-gcc.exe: error: scan/build/bootloader/bootloader.map: No such file or directory
ninja: build stopped: subcommand failed.
[3/281] cmd.exe /C "cd . && C:\Users\JM\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E remove esp-idf\efuse\libefuse.a && C:\Users\JM\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe qc esp-idf\efuse\libefuse.a  esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32/esp_efuse_table.c.obj esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/esp_efuse_api.c.obj esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/esp_efuse_fields.c.obj esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/esp_efuse_utility.c.obj && C:\Users\JM\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ranlib.exe esp-idf\efuse\libefuse.a && cd ."
[4/281] C:\Users\JM\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -ID:/esp-idf/components/esp_http_client/include -ID:/esp-idf/components/esp_http_client/lib/include -ID:/esp-idf/components/newlib/platform_include -ID:/esp-idf/components/freertos/include -ID:/esp-idf/components/heap/include -ID:/esp-idf/components/log/include -ID:/esp-idf/components/soc/esp32/include -ID:/esp-idf/components/soc/include -ID:/esp-id [... cut ...] -DIDF_VER=\"GIT-NOTFOUND\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/esp_http_client/CMakeFiles/__idf_esp_http_client.dir/esp_http_client.c.obj -MF esp-idf\esp_http_client\CMakeFiles\__idf_esp_http_client.dir\esp_http_client.c.obj.d -o esp-idf/esp_http_client/CMakeFiles/__idf_esp_http_client.dir/esp_http_client.c.obj   -c D:/esp-idf/components/esp_http_client/esp_http_client.c
ninja: build stopped: subcommand failed.
Build complete (0 errors, 0 warnings): E:\Code\esp-idf\wifi scan\build
```

kolipakakondal, 2019-10-17:
I have confirmed with the team, It seems that cryptography package has not uploaded py3.8 wheels yet: https://pypi.org/project/cryptography/#files

So for now have to use 3.7.x