espressif/idf-eclipse-plugin #15: Cannot install Eclipse plug ins with release 3.3 LTS

robcazzaro, 2019-10-18:
I'm new to esp-idf development (only ESP32 Arduino, even if I used Eclipse for STM32 and others)

I'm planning to use esp-idf 3.3 LTS, as it seems the best choice for long term, so I downloaded the ZIP file with all the submodules and expanded into my work directory

I followed the instructions here https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md. I'm on Windows 10, and with Python 3.7.4

I installed the plugins per instructions, following "Installing IDF Plugins using update site url". When I tried to follow "Installing ESP-IDF Tools", I got an error because the 3.3 release is missing idf_tools.py (and so does 3.1.6, btw). I copied a few missing files from the latest github version, under tools. When re-executing the "install tools", I got an error due to a missing Python library. I installed it with "pip install subprocess.run", relaunched again.

This time it downloaded the tools properly, but then the next 2 steps failed again with an error...

At this point I'm lost... I can see that the tools have been installed (using the "List installed tools" option), but I don't know if the environment has been set properly or not

What am I  missing?

Executing C:\Program Files\Python37\python.exe C:\ESP32\esp-idf-v3.3\tools\idf_tools.py install-python-env
Traceback (most recent call last):
  File "C:\ESP32\esp-idf-v3.3\tools\idf_tools.py", line 1317, in <module>
    main(sys.argv[1:])
  File "C:\ESP32\esp-idf-v3.3\tools\idf_tools.py", line 1313, in main
    action_func(args)
  File "C:\ESP32\esp-idf-v3.3\tools\idf_tools.py", line 1112, in action_install_python_env
    idf_python_env_path, _, virtualenv_python = get_python_env_path()
  File "C:\ESP32\esp-idf-v3.3\tools\idf_tools.py", line 813, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '-C', global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
  File "C:\Program Files\Python37\lib\subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "C:\Program Files\Python37\lib\site-packages\run\__init__.py", line 145, in __new__
    process = cls.create_process(command, stdin, cwd=cwd, env=env, shell=shell)
  File "C:\Program Files\Python37\lib\site-packages\run\__init__.py", line 121, in create_process
    shlex.split(command),
  File "C:\Program Files\Python37\lib\shlex.py", line 305, in split
    return list(lex)
  File "C:\Program Files\Python37\lib\shlex.py", line 295, in __next__
    token = self.get_token()
  File "C:\Program Files\Python37\lib\shlex.py", line 105, in get_token
    raw = self.read_token()
  File "C:\Program Files\Python37\lib\shlex.py", line 136, in read_token
    nextchar = self.instream.read(1)
AttributeError: 'list' object has no attribute 'read'

Executing C:\Program Files\Python37\python.exe C:\ESP32\esp-idf-v3.3\tools\idf_tools.py export --format=key-value
Traceback (most recent call last):
  File "C:\ESP32\esp-idf-v3.3\tools\idf_tools.py", line 1317, in <module>
    main(sys.argv[1:])
  File "C:\ESP32\esp-idf-v3.3\tools\idf_tools.py", line 1313, in main
    action_func(args)
  File "C:\ESP32\esp-idf-v3.3\tools\idf_tools.py", line 953, in action_export
    idf_python_env_path, idf_python_export_path, virtualenv_python = get_python_env_path()
  File "C:\ESP32\esp-idf-v3.3\tools\idf_tools.py", line 813, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '-C', global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
  File "C:\Program Files\Python37\lib\subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "C:\Program Files\Python37\lib\site-packages\run\__init__.py", line 145, in __new__
    process = cls.create_process(command, stdin, cwd=cwd, env=env, shell=shell)
  File "C:\Program Files\Python37\lib\site-packages\run\__init__.py", line 121, in create_process
    shlex.split(command),
  File "C:\Program Files\Python37\lib\shlex.py", line 305, in split
    return list(lex)
  File "C:\Program Files\Python37\lib\shlex.py", line 295, in __next__
    token = self.get_token()
  File "C:\Program Files\Python37\lib\shlex.py", line 105, in get_token
    raw = self.read_token()
  File "C:\Program Files\Python37\lib\shlex.py", line 136, in read_token
    nextchar = self.instream.read(1)
AttributeError: 'list' object has no attribute 'read'

igrr, 2019-10-18:
Sorry, at the moment the plugin supports IDF 4.x. We will update IDF 3.x to include features required by the plugin in the future.

https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md#prerequisites

robcazzaro, 2019-10-18:
As I said, I copied all the relevant files from 4.0, but still had errors.

Anyway, I downloaded 4.0-beta1, the officially suggested 4.0 build, rerun on a clean slate. Same missing module in Python (which is something many users without that module will face; the module is not installed by default in Python Windows)

And same error at the end (I removed all the download % from the console)

Installing tools...
This can take a while. Please be patient.
Executing C:\Program Files\Python37\python.exe C:\ESP32\esp-idf-v4.0-beta1\tools\idf_tools.py install all
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
Installing xtensa-esp32-elf@esp32-2019r1-8.2.0
Downloading xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-win32.zip to C:\Users\Roberto\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-win32.zip.tmp
0%
100%
Done
Extracting C:\Users\Roberto\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-win32.zip to C:\Users\Roberto\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0
Installing esp32ulp-elf@2.28.51.20170517
Downloading binutils-esp32ulp-win32-2.28.51-esp32ulp-20180809.zip to C:\Users\Roberto\.espressif\dist\binutils-esp32ulp-win32-2.28.51-esp32ulp-20180809.zip.tmp
0%
100%
Done
Extracting C:\Users\Roberto\.espressif\dist\binutils-esp32ulp-win32-2.28.51-esp32ulp-20180809.zip to C:\Users\Roberto\.espressif\tools\esp32ulp-elf\2.28.51.20170517
Installing cmake@3.13.4
Downloading cmake-3.13.4-win64-x64.zip to C:\Users\Roberto\.espressif\dist\cmake-3.13.4-win64-x64.zip.tmp
0%
100%
Done
Extracting C:\Users\Roberto\.espressif\dist\cmake-3.13.4-win64-x64.zip to C:\Users\Roberto\.espressif\tools\cmake\3.13.4
Installing openocd-esp32@v0.10.0-esp32-20190313
Downloading openocd-esp32-win32-0.10.0-esp32-20190313.zip to C:\Users\Roberto\.espressif\dist\openocd-esp32-win32-0.10.0-esp32-20190313.zip.tmp
0%
100%
Done
Extracting C:\Users\Roberto\.espressif\dist\openocd-esp32-win32-0.10.0-esp32-20190313.zip to C:\Users\Roberto\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313
Installing mconf@v4.6.0.0-idf-20190628
Downloading mconf-v4.6.0.0-idf-20190628-win32.zip to C:\Users\Roberto\.espressif\dist\mconf-v4.6.0.0-idf-20190628-win32.zip.tmp
0%
100%
Done
Extracting C:\Users\Roberto\.espressif\dist\mconf-v4.6.0.0-idf-20190628-win32.zip to C:\Users\Roberto\.espressif\tools\mconf\v4.6.0.0-idf-20190628
Installing ninja@1.9.0
Downloading ninja-1.9.0-win64.zip to C:\Users\Roberto\.espressif\dist\ninja-1.9.0-win64.zip.tmp
0%
100%
Done
Extracting C:\Users\Roberto\.espressif\dist\ninja-1.9.0-win64.zip to C:\Users\Roberto\.espressif\tools\ninja\1.9.0
Installing idf-exe@1.0.1
Downloading idf-exe-v1.0.1.zip to C:\Users\Roberto\.espressif\dist\idf-exe-v1.0.1.zip.tmp
0%
100%
Done
Extracting C:\Users\Roberto\.espressif\dist\idf-exe-v1.0.1.zip to C:\Users\Roberto\.espressif\tools\idf-exe\1.0.1
Installing ccache@3.7
Downloading ccache-3.7-w64.zip to C:\Users\Roberto\.espressif\dist\ccache-3.7-w64.zip.tmp
0%
97%
100%
Done
Extracting C:\Users\Roberto\.espressif\dist\ccache-3.7-w64.zip to C:\Users\Roberto\.espressif\tools\ccache\3.7

Executing C:\Program Files\Python37\python.exe C:\ESP32\esp-idf-v4.0-beta1\tools\idf_tools.py install-python-env
Traceback (most recent call last):
  File "C:\ESP32\esp-idf-v4.0-beta1\tools\idf_tools.py", line 1317, in <module>
    main(sys.argv[1:])
  File "C:\ESP32\esp-idf-v4.0-beta1\tools\idf_tools.py", line 1313, in main
    action_func(args)
  File "C:\ESP32\esp-idf-v4.0-beta1\tools\idf_tools.py", line 1112, in action_install_python_env
    idf_python_env_path, _, virtualenv_python = get_python_env_path()
  File "C:\ESP32\esp-idf-v4.0-beta1\tools\idf_tools.py", line 813, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '-C', global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
[... 16 lines cut ...]
Executing C:\Program Files\Python37\python.exe C:\ESP32\esp-idf-v4.0-beta1\tools\idf_tools.py export --format=key-value
Traceback (most recent call last):
  File "C:\ESP32\esp-idf-v4.0-beta1\tools\idf_tools.py", line 1317, in <module>
    main(sys.argv[1:])
  File "C:\ESP32\esp-idf-v4.0-beta1\tools\idf_tools.py", line 1313, in main
    action_func(args)
  File "C:\ESP32\esp-idf-v4.0-beta1\tools\idf_tools.py", line 953, in action_export
    idf_python_env_path, idf_python_export_path, virtualenv_python = get_python_env_path()
  File "C:\ESP32\esp-idf-v4.0-beta1\tools\idf_tools.py", line 813, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '-C', global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
  File "C:\Program Files\Python37\lib\subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "C:\Program Files\Python37\lib\site-packages\run\__init__.py", line 145, in __new__
    process = cls.create_process(command, stdin, cwd=cwd, env=env, shell=shell)
  File "C:\Program Files\Python37\lib\site-packages\run\__init__.py", line 121, in create_process
    shlex.split(command),
  File "C:\Program Files\Python37\lib\shlex.py", line 305, in split
    return list(lex)
  File "C:\Program Files\Python37\lib\shlex.py", line 295, in __next__
    token = self.get_token()
  File "C:\Program Files\Python37\lib\shlex.py", line 105, in get_token
    raw = self.read_token()
  File "C:\Program Files\Python37\lib\shlex.py", line 136, in read_token
    nextchar = self.instream.read(1)
AttributeError: 'list' object has no attribute 'read'

closed 2019-10-18 by robcazzaro