espressif/idf-eclipse-plugin #16: ULP template project not making ulp_main.h

rdkluijv, 2019-10-20:
Using the ulp template project I get this:

[ulp.txt](https://github.com/espressif/idf-eclipse-plugin/files/3747688/ulp.txt)

It looks like the ULP toolchain is installed but I'm not sure.
Question: Is it possible to use Eclipse for a ulp project?

kolipakakondal, 2019-10-22:
@rdkluijv Make sure you select launch target before you compile https://github.com/espressif/idf-eclipse-plugin#configuring-launch-target

I could able to compile ULP template from (System > ulp)
[global-build.log](https://github.com/espressif/idf-eclipse-plugin/files/3753838/global-build.log)

@igrr Could you think of any reason for this?

rdkluijv, 2019-10-22:
Couldn't build it from the command line either so it's not a problem due to the Eclipse plugin.
This line in the ULP coprocessor programming makes me believe the tools are installed:
"If you have already set up ESP-IDF with CMake build system according to the Getting Started Guide, then the ULP toolchain is already installed."
I'll install it manually to be sure.
Launch target is the same for all the examples I use. The others more or less work so I think that's not the problem.

closed 2026-01-22 by AndriiFilippov