espressif/idf-eclipse-plugin #17: Unable to resolve #include files

AwkBot, 2019-10-23:
Hi,
I did a fresh installation, ESP_IDF, eclipse and so on, the installation is now running fine however no #includes has been resolved and a build failed due missing bootloader.map file.

AwkBot, 2019-10-23:
Sorry,

Just to add additional information,  in fact the bootloader.elf was not generated.

`Building in: C:\Users\carlo\iCloudDrive\Documents\ESP32\Wrover-Protocol Analyzer\build
cmake --build . -- -v
[1/32] cmd.exe /C "cd . && C:\esp-idf\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E remove esp-idf\asio\libasio.a && C:\esp-idf\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe qc esp-idf\asio\libasio.a  esp-idf/asio/CMakeFiles/__idf_asio.dir/asio/asio/src/asio.cpp.obj && C:\esp-idf\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ranlib.exe esp-idf\asio\libasio.a && cd ."
[2/32] cmd.exe /C "cd . && C:\esp-idf\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E remove esp-idf\cbor\libcbor.a && C:\esp-idf\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe qc esp-idf\cbor\libcbor.a  esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cborencoder_close_container_checked.c.obj esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cborencoder.c.obj esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cborerrorstrings.c.obj esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cborparser_dup_string.c.obj esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cborparser.c.obj esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cborpretty_stdio.c.obj esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cborpretty.c.obj esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cbortojson.c.obj esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cborvalidation.c.obj esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/open_memstream.c.obj && C:\esp-idf\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ranlib.exe esp-idf\cbor\libcbor.a && cd ."
[3/32] cmd.exe /C "cd . && C:\esp-idf\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E remove esp-idf\coap\libcoap.a && C:\esp-idf\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe qc esp-idf\coap\libcoap.a  esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/address.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/async.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/block.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/coap_event.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/coap_hashkey.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/coap_session.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/coap_time.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/port/coap_debug.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/encode.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/mem.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/net.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/option.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/pdu.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/resource.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/str.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/subscribe.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/uri.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/coap_io.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/port/coap_notls.c.obj esp-idf/coap/CMakeFiles/__idf_coap.dir/port/coap_mbedtls.c.obj && C:\esp-idf\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ranlib.exe esp-idf\coap\libcoap.a && cd ."
[4/32] cmd.exe /C "cd . && C:\esp-idf\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E remove esp-idf\console\libconsole.a && C:\esp-idf\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe qc esp-idf\console\libconsole.a  esp-idf/console/CMakeFiles/__idf_console.dir/commands.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/split_argv.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/argtable3.c.obj esp-idf/console/CMakeFiles/__idf_console.dir/linenoise/linenoise.c.obj && C:\esp-idf\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ranlib.exe esp-idf\console\libconsole.a && cd ."
[5/32] cmd.exe /C "cd . && C:\esp-idf\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E remove esp-idf\esp_adc_cal\libesp_adc_cal.a && C:\esp-idf\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe qc esp-idf\esp_adc_cal\libesp_adc_cal.a  esp-idf/esp_adc_cal/CMakeFiles/__idf_esp_adc_cal.dir/esp_adc_cal.c.obj && C:\esp-idf\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ranlib.exe esp-idf\esp_adc_cal\libesp_adc_cal.a && cd ."
[6/32] cmd.exe /C "cd . && C:\esp-idf\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E remove esp-idf\esp_gdbstub\libesp_gdbstub.a && C:\esp-idf\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe qc esp-idf\esp_gdbstub\libesp_gdbstub.a  esp-idf/esp_gdbstub/CMakeFiles/__idf_esp_gdbstub.dir/src/gdbstub.c.obj esp-idf/esp_gdbstub/CMakeFiles/__idf_esp_gdbstub.dir/src/packet.c.obj esp-idf/esp_gdbstub/CMakeFiles/__idf_esp_gdbstub.dir/esp32/gdbstub_esp32.c.obj esp-idf/esp_gdbstub/CMakeFiles/__idf_esp_gdbstub.dir/xtensa/gdbstub_xtensa.c.obj && C:\esp-idf\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ranlib.exe esp-idf\esp_gdbstub\libesp_gdbstub.a && cd ."
[... cut ...] -idf\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ranlib.exe esp-idf\spiffs\libspiffs.a && cd ."
[20/32] cmd.exe /C "cd /D "C:\Users\carlo\iCloudDrive\Documents\ESP32\Wrover-Protocol Analyzer\build\bootloader" && C:\esp-idf\.espressif\tools\cmake\3.13.4\bin\cmake.exe --build ."
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-build bootloader/bootloader.elf bootloader/bootloader.bin bootloader/bootloader.map `

It is a new amd empty project.

kolipakakondal, 2019-11-26:
Sorry for the delay in getting back to you. Includes will be resolved only when the build succeeds. 

Check the below issues, they were reported on the same line.
https://github.com/espressif/esp-idf/issues/2936
https://github.com/espressif/esp-idf/issues/3981

Make sure you're on esp-idf 4.x latest and all the esp-idf-plugin prerequisites are met.

kolipakakondal, 2020-02-12:
Hi All, IDF Eclipse Plugin v1.0.0-beta5 (Build: 1.0.0.202002120536) is published to the update site. Please get the latest https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

This will resolve "unresolved includes" issue, hence closing the issue

closed 2020-02-12 by kolipakakondal