espressif/idf-eclipse-plugin #18: Build Issue

saahilshroff, 2019-10-25:
When I am trying to build the code for blinking the onboard LED, the following message is displayed:

"Building in: C:\Users\20148034\eclipse-workspace\blink\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=C:\Users\20148034\.espressif\releases\esp-idf-v3.3.zip -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DIDF_TARGET=esp32 C:\Users\20148034\eclipse-workspace\blink
fatal: not a git repository (or any of the parent directories): .git
CMake Warning at C:/Users/20148034/Desktop/IDF/esp-idf-master/tools/cmake/kconfig.cmake:39 (message):
  Failed to detect version of mconf-idf.  Return code was 1.
Call Stack (most recent call first):
  C:/Users/20148034/Desktop/IDF/esp-idf-master/tools/cmake/build.cmake:161 (__kconfig_init)
  C:/Users/20148034/Desktop/IDF/esp-idf-master/tools/cmake/idf.cmake:48 (__build_init)
  C:/Users/20148034/Desktop/IDF/esp-idf-master/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
CMake Error at C:/Users/20148034/Desktop/IDF/esp-idf-master/tools/cmake/build.cmake:271 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  C:/Users/20148034/Desktop/IDF/esp-idf-master/tools/cmake/build.cmake:397 (__build_check_python)
  C:/Users/20148034/Desktop/IDF/esp-idf-master/tools/cmake/project.cmake:341 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/20148034/eclipse-workspace/blink/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/20148034/eclipse-workspace/blink/build/CMakeFiles/CMakeError.log".
cmake --build . -- -v
ninja: error: loading 'build.ninja': The system cannot find the file specified.

Build complete (0 errors, 0 warnings): C:\Users\20148034\eclipse-workspace\blink\build_"

Kindly help me out!

closed 2019-10-30 by saahilshroff

SpiritSeal, 2020-03-28:
Hello @saahilshroff I am having a similar issue, how did you resolve yours?

phy455, 2020-07-08:
I am facing the same issue, can anyone help?