espressif/idf-eclipse-plugin #20: Install tools fails on Windows 10

danak6jq, 2019-10-31:
Even with the latest code, I'm getting an install tools failure that leaves only the IDF_PATH configured in eclipse (using 2019-12M1 but I do not believe that matters, I was using 2019-09 before).

What I realized is that 'git' is not found on the Windows PATH, so get_version() and such fail with file not found.

I added the git dir ("C:\Program Files\Git\cmd\" on my system) to my Windows PATH variable and now install tools succeeds - and now I get to use an IDE, I sure missed that.

So either this needs to be documented in the release notes/FAQ (I looked but perhaps missed it) or a solution needs to be found in the plugin.

Next thing, soon enough, will be getting OpenOCD going, but that's far too complicated. You know how happy I'll be when that works turnkey...

kolipakakondal, 2019-10-31:
Thanks for reporting this. You're right - we've not documented "git" in the prerequisite. But it's a must for installing tools. I'll update the documentation.

Report us if you see any issues with the OpenOCD debugging. I've added launch configuration files here https://github.com/espressif/idf-eclipse-plugin/tree/master/resources/Debugging%20Launch%20Configurations

You can import them using "Import Launch Configurations" Wizard from eclipse
`File > Import > Run/Debug > Launch Configurations `

https://help.eclipse.org/kepler/index.jsp?topic=%2Forg.eclipse.jdt.doc.user%2Freference%2Fwizards%2Fref-import_launch_configs.htm

After importing, make sure to modify the project-specific settings and GDB Paths on the configuration page.

For GDB Hardware debugging:
Refer this: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/#setup-of-openocd

We can use this thread for updating the documentation, If you see any issues with the OpenOCD debugging - please create a new issue.

danak6jq, 2019-10-31:
@kolipakakondal what is interesting is that I *do* have git installed on my system; I've been using ESP-IDF + tools installer before attempting to use eclipse + ESP-IDF plugin.

The issue is that the eclipse plugin needs to find git in the Windows PATH when eclipse is launched. In fact, I am pretty sure that the plug-in had just confirmed that git was already installed - but then invokes git without a full path, thus depending on the Windows PATH setting to include it.

danak6jq, 2019-10-31:
@igrr thank you for the follow-up. I appreciate the progress with the ESP-IDF tools!

kolipakakondal, 2019-12-24:
Fix is part of the idf-eclipse-plugin-beta.4 update. Please get the latest changes using the update site url.

closed 2019-12-24 by kolipakakondal