espressif/idf-eclipse-plugin #22: No Toolchain found for Target esp32

saahilshroff, 2019-11-06:
Hi,

I am following this documentation to get started on ESP32 using Eclipse, trying to replicate the steps on Windows 10.
I have successfully configured the target, but while compiling the project, the build console shows "No Toolchain found for Target esp32", unlike has been documented.
Kindly help! 
[image]

kolipakakondal, 2019-11-06:
"Install Tools" action from eclipse ideally configures the toolchain based on the IDF_PATH found.
But looking at the error - toolchain is not configured correctly due to some reason in your case.

**To troubleshoot:**

1. Run Install Tools once again https://github.com/espressif/idf-eclipse-plugin#installing-esp-idf-tools
2. Verify if all the environment variables configured correctly https://github.com/espressif/idf-eclipse-plugin#configuring-environment-variables
3. Verify toolchains - https://github.com/espressif/idf-eclipse-plugin#configuring-core-build-toolchains

If you see toolchain is not configured correctly - go and configure toolchain as mentioned.

saahilshroff, 2019-11-06:
Hi @kolipakakondal 

According to https://github.com/espressif/idf-eclipse-plugin, I am supposed to include "python_env/idf4.0_py3.7_env/bin" when I configure the "PATH" variable. 
However, these are the only folders in my python_env/idf4.1_py3.8_env directory: 

[image]

Should I just include these folders while configuring my "PATH" variable?

kolipakakondal, 2019-11-06:
That should be fine since plugin appends /bin or /Scripts based on the operating system.

However, I would suggest you to troubleshoot why these environment variables are not auto-configured. Tools console/error log might provide some info on this.

saahilshroff, 2019-11-06:
> "Install Tools" action from eclipse ideally configures the toolchain based on the IDF_PATH found.
> But looking at the error - toolchain is not configured correctly due to some reason in your case. [...]

Completed the steps as suggested. 
The error: 
[image]

saahilshroff, 2019-11-07:
While re-installing the tools (HELP->Espressif IDF Tools Manager->Install tools), the following message showed up on the console window. 
What should be done in such a case?

**CONSOLE OUTPUT:**
Installing tools...
This can take a while. Please be patient.
Executing C:\Users\40007382\AppData\Local\Programs\Python\Python38-32\python.exe D:\esp-idf-4.0-beta2\tools\idf_tools.py install all
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
Skipping xtensa-esp32-elf@esp32-2019r1-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190313 (already installed)
Skipping mconf@v4.6.0.0-idf-20190628 (already installed)
Skipping ninja@1.9.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)

Executing C:\Users\40007382\AppData\Local\Programs\Python\Python38-32\python.exe D:\esp-idf-4.0-beta2\tools\idf_tools.py install-python-env
fatal: not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
  File "D:\esp-idf-4.0-beta2\tools\idf_tools.py", line 1317, in <module>
    main(sys.argv[1:])
  File "D:\esp-idf-4.0-beta2\tools\idf_tools.py", line 1313, in main
    action_func(args)
  File "D:\esp-idf-4.0-beta2\tools\idf_tools.py", line 1112, in action_install_python_env
    idf_python_env_path, _, virtualenv_python = get_python_env_path()
  File "D:\esp-idf-4.0-beta2\tools\idf_tools.py", line 813, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '-C', global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
  File "C:\Users\40007382\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\40007382\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['git', '-C', 'D:\\esp-idf-4.0-beta2', 'describe', '--tags']' returned non-zero exit status 128.

Executing C:\Users\40007382\AppData\Local\Programs\Python\Python38-32\python.exe D:\esp-idf-4.0-beta2\tools\idf_tools.py export --format=key-value
fatal: not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
  File "D:\esp-idf-4.0-beta2\tools\idf_tools.py", line 1317, in <module>
    main(sys.argv[1:])
  File "D:\esp-idf-4.0-beta2\tools\idf_tools.py", line 1313, in main
    action_func(args)
  File "D:\esp-idf-4.0-beta2\tools\idf_tools.py", line 953, in action_export
    idf_python_env_path, idf_python_export_path, virtualenv_python = get_python_env_path()
  File "D:\esp-idf-4.0-beta2\tools\idf_tools.py", line 813, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '-C', global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
  File "C:\Users\40007382\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\40007382\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['git', '-C', 'D:\\esp-idf-4.0-beta2', 'describe', '--tags']' returned non-zero exit status 128.

kolipakakondal, 2019-11-07:
Make sure you set the **git** to the windows system environment **Path**

saahilshroff, 2019-11-07:
Hi @kolipakakondal 

I have set the **git** to the windows system environment **PATH**. The error persists!
[image]

kolipakakondal, 2019-11-07:
Can you please run **git -C D:\esp-idf-4.0-beta2 describe --tags** from the command line to verify that git works?

saahilshroff, 2019-11-07:
Hi @kolipakakondal 
The git was not working for **esp-idf-4.0-beta2**.  Cloned another esp-idf from the git.
Also, used python3.7 (64-bit version).

Everything works fine now!
Kudos to you for helping out. 
Thanks! :)

closed 2019-11-07 by saahilshroff

moneshrathod, 2023-09-24:
I tried Run>Run As>  Local C/C++ Application and it works for me