espressif/idf-eclipse-plugin #24: windows 10 - Command 'com.espressif.idf.ui.command.install' failed

oryxgod, 2019-11-08:
The following is displayed to me if I choose Help -> Espressif IDF Tools Manager -> Install Tools

I used the newest java version, I also tryed it with java8u221 - same result.

I used url to install the idf-eclipse-plugin from 
https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/
but i also tryed local installation with
idf-eclipse-plugin-v1.0.0-beta.2
also the same result.

Could you help me please?

Exception Stack Trace:
org.eclipse.core.commands.ExecutionException: No Python installations found in the system.
	at com.espressif.idf.ui.update.AbstractToolsHandler.execute(AbstractToolsHandler.java:71)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:283)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:95)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:318)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:252)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:173)
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:156)
	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:488)
	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:487)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:213)
	at org.eclipse.e4.ui.workbench.renderers.swt.HandledContributionItem.executeItem(HandledContributionItem.java:438)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.handleWidgetSelection(AbstractContributionItem.java:449)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.lambda$2(AbstractContributionItem.java:475)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4175)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1057)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:3988)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3587)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1160)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1049)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:660)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:559)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:150)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:150)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:660)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:597)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1468)

kolipakakondal, 2019-11-08:
>Exception Stack Trace:
org.eclipse.core.commands.ExecutionException: No Python installations found in the system.

It seems to be plugin is unable to identify python installations entries in the Windows registry.
Can you verify these locations?

```
HKEY_CURRENT_USER\SOFTWARE\Python\PythonCore\
HKEY_LOCAL_MACHINE\SOFTWARE\Python\PythonCore\
```

oryxgod, 2019-11-11:
Hello,

Thank your for fast response.

I can't find the locations

HKEY_CURRENT_USER\SOFTWARE\Python\

and

HKEY_LOCAL_MACHINE\SOFTWARE\Python\

in my regedit.

I tryed to reinstall python,

its installed under

C:\Program Files (x86)\Python38-32

but there is again no regedit entry.

Can i manually set this entry?

Do you possibly have an other solution?

Have a nice day!

Kindly regards
Kevin Blümel

kolipakakondal, 2019-11-11:
See if you can find a reason why python is not added to the Windows registry and fix the problem. That will further avoid the issues.

Anyway, you can also set a registry entry as specified below and it should work.
https://stackoverflow.com/questions/17872234/how-to-add-python-to-windows-registry

devslo, 2019-11-19:
I get the same error. I've checked registry and the key HKEY_LOCAL_MACHINE\SOFTWARE\Python\ seems to be fine (see image).

Version of Eclipse:
Version: 2019-09 R (4.13.0)
Build id: 20190917-1200

Java:  1.8.0_231 (64 bit)

[image]

kolipakakondal, 2019-11-20:
This is what is required. **InstallPath** is missing in your case!

[image]

devslo, 2019-11-20:
True. Thank you.

closed 2019-11-21 by kolipakakondal