espressif/idf-eclipse-plugin #25: Incorrect automatic plugin configuration and Build Error

Maurizio1, 2019-11-10:
My configuration is:
Ubuntu 19.10 in VMware Workstation 15 VM guest on Windows 10 Host
Eclipse 2019-09 R (4.13.0)
ESP-IDF v4.0-beta2

I have installed the plugin through the automatic procedure.
While I can successfully build and flash a program by command line, I get the following error in Eclipse:
 
Errors occurred during the build.
Errors running builder 'CDT Core Builder' on project 'hello_world'.
Building hello_world
Cannot run program "/home/maurizio/Development/esp/esp-idf/tools/cmake" (in directory "/home/maurizio/Development/eclipse-workspace/hello_world/build"): error=13, Permission denied
Building hello_world
Cannot run program "/home/maurizio/Development/esp/esp-idf/tools/cmake" (in directory "/home/maurizio/Development/eclipse-workspace/hello_world/build"): error=13, Permission denied

I believe to have carefully followed instructions while setting up the environment and the plugin.
It looks like the project is not recognized as an Espressif IDF Project, same thing If I manually configure parameters.
Environment variables (both in system and in Eclipse) are OK as stated on instructions.
Some parameter fields in Eclipsed are not automatically configured by the plugin.
Indexer doesn't recognize include files (even when, if manually configured, Eclipse successfully compiles the project with no errors)

kolipakakondal, 2019-11-11:
Either PATH or launch configuration is not configured correctly https://github.com/espressif/idf-eclipse-plugin#configuring-launch-target

Does the PATH auto-configured in the eclipse? This happens when you run "Install Tools" from eclipse

Maurizio1, 2019-11-11:
Dear Kondal,

thank you for your quick answer.

Launch configuration is correct (IDF Target = esp32 on ttyUSB0)
PATH in Eclipse is (I split the elements for readability):

/home/maurizio/Development/esp/esp-idf/components/esptool_py/esptool:
/home/maurizio/Development/esp/esp-idf/components/espcoredump:
/home/maurizio/Development/esp/esp-idf/components/partition_table/:
/home/maurizio/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin:
/home/maurizio/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:
/home/maurizio/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin:
/home/maurizio/.espressif/python_env/idf4.0_py2.7_env/bin:
/home/maurizio/Development/esp/esp-idf/tools:
/home/maurizio/.local/bin:
/usr/local/sbin:
/usr/local/bin:
/usr/sbin:
/usr/bin:
/sbin:
/bin:
/usr/games:
/usr/local/games:
/usr/local/ssl/bin:
/snap/bin

I have checked each element in PATH and they are correct.

The build error: 
Cannot run program "/home/maurizio/Development/esp/esp-idf/tools/cmake" (in directory "/home/maurizio/Development/eclipse-workspace/hello_world/build"): error=13, Permission denied
Building hello_world

I noticed that  "/home/maurizio/Development/esp/esp-idf/tools/cmake" is a directory name...

kolipakakondal, 2019-11-11:
Oh! Understood. But this is fixed with **idf-eclipse-plugin-v1.0.0-beta.2** release.  Are you on the latest changes, if not could you please get the latest changes using the update site url(we just released beta3 as well) and check once again?

Why this is not working for you?
In your PATH, there is no cmake path is defined(I wonder why?), so in the process of finding cmake executable it has found /home/maurizio/Development/esp/esp-idf/tools/cmake and assuming it's cmake executable which is wrong! Have you manually configured this?

To quickly fix this, you can remove /home/maurizio/Development/esp/esp-idf/tools your PATH a try, probably it might find CMake executable from your system path.  **As a clean approach, But I would suggest you do the following.**

1. Update to the latest plugin using the url
2. Remove all the environment variables. Preferences >Build > Environment
3. Run "Install Toos" again to reconfigure the paths. 
4. Build the project with by selecting the correct launch configuration

kolipakakondal, 2019-11-11:
For example, this is how my eclipse build environment PATH looks like and which is auto configured.

```
/Users/kondal/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin:
/Users/kondal/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:
/Users/kondal/.espressif/tools/cmake/3.13.4/CMake.app/Contents/bin:
/Users/kondal/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin:/
Users/kondal/.espressif/tools/ninja/1.9.0/:
/Users/kondal/.espressif/python_env/idf4.0_py3.7_env/bin:
/Users/kondal/esp/esp-idf/tools:/usr/bin:/bin:/usr/sbin:/sbin:/Users/kondal/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190708/openocd-esp32/bin
```
I don't want you to take the headache of configuring the PATH, let "install tools" action do the job. If you see any issues here, report us.

Maurizio1, 2019-11-11:
No, I didn’t manually configure the cmake path.

 

Also I noticed that in your configuration you have: /Users/kondal/.espressif/python_env/idf4.0_py3.7_env/bin:

while I have /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/bin:

(I have both 2.7 and 3.7 Python versions on my system)

 

Don’t know if that matters…

 

I will do what you suggested and let you know…

 

Thanks

Maurizio1, 2019-11-11:
I have gone through points from 1 to 4, as you suggested.
The only data in enviroment is now the IDF_PATH.
I could compile a program and includes are not a problem anymore! :-)

What happens now is that Eclipse takes forever to start, so does the launch of the JSON configuration server by opening sdkconfig.
After, let's say, five minutes, I get: "error retrieving output from the JSON configuration server" and the file wont open. I don't know if this depends on the plugin or not.

Also, here is a picture of the project's property window, is that complete or there's something missing in the menu? 
[image]

kolipakakondal, 2019-11-14:
After install tools, all the required paths need to be configured automatically. It seems to be there is an issue over there for you - can you look into install tools console and error log and what is error reported.

If you could fix that issue, sdkconfig should work

Maurizio1, 2019-11-16:
Installing tools...
This can take a while. Please be patient.
Executing /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/bin/python /home/maurizio/Development/esp/esp-idf/tools/idf_tools.py install all
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, ninja
Skipping xtensa-esp32-elf@esp32-2019r1-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190313 (already installed)
Skipping ninja@1.9.0 (already installed)

Executing /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/bin/python /home/maurizio/Development/esp/esp-idf/tools/idf_tools.py install-python-env
Requirement already satisfied: setuptools in /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/maurizio/Development/esp/esp-idf/requirements.txt (line 4)) (41.6.0)
Requirement already satisfied: click>=5.0 in /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/maurizio/Development/esp/esp-idf/requirements.txt (line 8)) (7.0)
Requirement already satisfied: pyserial>=3.0 in /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/maurizio/Development/esp/esp-idf/requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/maurizio/Development/esp/esp-idf/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/maurizio/Development/esp/esp-idf/requirements.txt (line 11)) (2.8)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/maurizio/Development/esp/esp-idf/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/maurizio/Development/esp/esp-idf/requirements.txt (line 13)) (0.25)
Requirement already satisfied: enum34; python_version < "3" in /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/maurizio/Development/esp/esp-idf/requirements.txt (line 11)) (1.1.6)
Requirement already satisfied: ipaddress; python_version < "3" in /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/maurizio/Development/esp/esp-idf/requirements.txt (line 11)) (1.0.23)
Requirement already satisfied: six>=1.4.1 in /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/maurizio/Development/esp/esp-idf/requirements.txt (line 11)) (1.13.0)
This is what I have in the install tools console (if I well understand your request):
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/maurizio/Development/esp/esp-idf/requirements.txt (line 11)) (1.13.2)
Requirement already satisfied: pycparser in /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /home/maurizio/Development/esp/esp-idf/requirements.txt (line 11)) (2.19)
Installing Python packages from /home/maurizio/Development/esp/esp-idf/requirements.txt

Executing /home/maurizio/.espressif/python_env/idf4.0_py2.7_env/bin/python /home/maurizio/Development/esp/esp-idf/tools/idf_tools.py export --format=key-value

The only difference I notice is that your installation has used Python 3.7 while mine version 2.7 (I even set up an alias to make Python 3.7 the default one)

As for the error log, which file you mean?

Build error is still the same: Errors occurred during the build.
Errors running builder 'CDT Core Builder' on project 'idf_default_template'.
Building idf_default_template
Cannot run program "/home/maurizio/Development/esp/esp-idf/tools/cmake" (in directory "/home/maurizio/Development/eclipse-workspace/idf_default_template/build"): error=13, Permission denied
Building idf_default_template
Cannot run program "/home/maurizio/Development/esp/esp-idf/tools/cmake" (in directory "/home/maurizio/Development/eclipse-workspace/idf_default_template/build"): error=13, Permission denied

If I look at launch configuration I see that is not configured:

[image]

Maurizio1, 2019-11-19:
So I finally could get it working.
The trick was to use ESP-IDF v4.1-dev-1086-g93a8603c5 and Ubuntu 18.04, other things left as before.
Includes and sdkconfig are not a problem anymore.

To compile and flash I need to do the following procedure:
create or import an IDF project, compile it, at that point I can see the sdkconfig and use it to edit the configuration, I cant add any Run configuration but if I launch the Run command then the program gets recompiled and flashed altogether. Please confirm if this is the expected behaviour.

Though, other weird things (for me) happen. I can no longer see some configuration items in Eclipse, particularly in the project's properties. I attach a few images to show what I mean.

[image]
[image]
[image]
[image]
[image]

Would you please confirm if this is what's expected?
Thanks for the precious help!

kolipakakondal, 2019-11-20:
> To compile and flash I need to do the following procedure:
> create or import an IDF project, compile it, at that point I can see the sdkconfig and use it to edit the configuration, I cant add any Run [...]

https://github.com/espressif/idf-eclipse-plugin#compiling-the-project
Please take a look at the respective buttons which are marked in the image - build and run/flash

Yes, clicking on the "run" button will flash the program into the device. It will compile only if it's not compiled earlier during the "build" step.

> Though, other weird things (for me) happen. I can no longer see some configuration items in Eclipse, particularly in the project's properties. I attach a few images to show what I mean (two for project's properties and three for Eclipse's ones.
> 
> [image]

Are you missing anything here?

> [image]

No needn't to configure anything here. Though it looks empty when you open it, it will be automatically configured internally when you flash it. Modify this configuration only if you wish to change the flash arguments. Please check the doc for this.

> [image]

It seems to be IDF_PYTHON_ENV_PATH is missing here?

> [image]

> [image]
> 
> Maybe I just don't remember well how it was in the previous environment...
> Would you please confirm if this is what's expected?
> Thanks for the precious help!

I don't think you need to configure anything in the properties. Let me know what you're looking for. GCC toolchain configuration is automatically taken care of.

Maurizio1, 2019-11-22:
Thanks.
I have manually corrected the missing IDF_PYTHON_ENV_PATH problem.
The only annoying thing is that I need **first** to build the application and only then the headers are solved and I can see and use sdkconfig in order to edit the configuration, then build again and flash. I can live with that but I signal it in case it's not the expected behavior.

kolipakakondal, 2019-11-22:
That's the expected behaviour as of now.  But we will be improving this in the coming releases.

kolipakakondal, 2019-11-22:
Can this be closed?

Maurizio1, 2019-11-22:
> That's the expected behaviour as of now. But we will be improving this in the coming releases.
That's OK.

> Can this be closed?
Sure! 
[image]

Thank you for all, especially for the patience.

closed 2019-11-22 by Maurizio1