espressif/idf-eclipse-plugin #26: Problem to load plugin

winkste, 2019-11-10:
My system environment is:
- OSX Mojave
- Eclipse 
       - Version: 2019-09 R (4.13.0)
       - Build id: 20190917-1200

I'm getting an error during the load process of the plugin. I did a manual download which generated other errors. 

[image]

winkste, 2019-11-10:
while trying to update the manual installed plugin, I got the following message:
[image]

kolipakakondal, 2019-11-11:
It seems to be an issue with your Java version. What's is the Java version that you're using?

It is a common error message reported by the Java Virtual Machine. This is caused when the Java environment does not have information about the HTTPS server to verify that it is a valid website.

Search google with " sun.security.validator.ValidatorException: PKIX path" to find possible answers.

winkste, 2019-11-12:
Hi.
Thanks for the fast response. I updated my JAVA version to:
[image]

That did not solve the problem. Than I tried some of the mentioned solutions found in google:
_Add following to the eclipse.ini file:
- Dorg.eclipse.ecf.provider.filetransfer.excludeContributors=org.eclipse.ecf.provider.filetransfer.httpclient4

None so far worked.

kolipakakondal, 2019-11-12:
Can you provide eclipse configuration details by following this?
`Eclipse > About Eclipse >Installation details >Configuration`

winkste, 2019-11-14:
[conf_details.txt](https://github.com/espressif/idf-eclipse-plugin/files/3844602/conf_details.txt)

kolipakakondal, 2019-11-14:
From the configuration details - I could see, eclipse is still pointing to /Library/Java/JavaVirtualMachines/jdk1.8.0_25.jdk/

That's an issue.  If you don't use jdk1.8.0_25  - straight away you can delete it and restart the eclipse. If you need old jdk for some reason - set new java path to the eclipse.ini as mentioned here https://wiki.eclipse.org/Eclipse.ini

winkste, 2019-11-14:
Interesting, although I installed the version specified above: 231. Running java -version in the Terminal shows:
[image]

and in the usual installation path of java it's showing:

[image]

So I re-installed JAVA 231 manually by downloading it from oracle:
[image]

Now it is showing up in the installation path of JAVA, ... look there the server update works!!!
[image]

Thanks Kondal for your support. One issue fixed... several left ;-)

winkste, 2019-11-15:
Problem solved.

closed 2019-11-15 by winkste