espressif/idf-eclipse-plugin #27: Problem with manual installation of package BETA 3

winkste, 2019-11-12:
Hi.

I tried to install the BETA3 package manually with the following error:

- Installation selection:
[image]
- error message:
[image]

kolipakakondal, 2019-11-12:
It seems to be you've selected the content.jar, can you select the com.espressif.idf.update-xxx.zip file directly without extracting it.

winkste, 2019-11-13:
Hi.
Manual installation now worked. Nevertheless, I still figure out other problems wich I will comment in other issue lists. Thanks for your fast response!

[image]

[image]

winkste, 2019-11-13:
see my comment above, thanks!

closed 2019-11-13 by winkste