espressif/idf-eclipse-plugin #28: Install Tools does not work

bbeliveau, 2019-11-13:
Hi Community.
Sorry in advance, I don't have a lot of experience in Eclipse and almost none with Espressif.

I'm using Eclipse 2018-12 (4.10.0) build 20181214-0600
I have installed succesfully ESP-IDF 3.3 and the tools with the Tools Installer and was able to open the menuconfig and build the Hello Worl example.

Then I tried to install the pluggin in Eclipse and found that IDF 4.0 (at least) was required. So I have installed 4.0 and the Tools with the Tool Installer again. And I was also  able to open the menuconfig and build the Hello Worl example in IDF 4.0.

Unfortunately when I try to install the pluggin in Eclipse, when it tries to install the ESP-IDF Tools (according to https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md, section "Installing ESP-IDF Tools"), it fails. And this is what the Eclipse Console shows:

> Executing C:\Users\bbeliveau\AppData\Local\Programs\Python\Python37-32\python.exe C:\Users\bbeliveau\Documents\Espressif\esp-idf_3_3\tools\idf_tools.py export --format=key-value
> C:\Users\bbeliveau\AppData\Local\Programs\Python\Python37-32\python.exe: can't open file 'C:\Users\bbeliveau\Documents\Espressif\esp-idf_3_3\tools\idf_tools.py': [Errno 2] No such file or directory
 
...where it looks for idf_tools.py from my old ESP-IDF 3.3 folder. 

So where I can change this to make sure it is looking for my 4.0 folder instead?

Thanks for your help!

kolipakakondal, 2019-11-14:
It can be changed using CDT build environment variables. Look for IDF_PATH and modify that to esp-idf 4.0 path.
https://github.com/espressif/idf-eclipse-plugin#configureEnvironmentVariables

bbeliveau, 2019-11-14:
Hi kolipakakondal. 
It helps and now it is working. Thank you!

closed 2019-11-14 by bbeliveau