espressif/idf-eclipse-plugin #29: eclipse IDE editor    unresolved includes

w4sepp, 2019-11-14:
building esp-idf project with pure make you can specify include directories with the item  
[image]

how can it be defined in the idf-eclipse plugin as there i cant find such entry

kolipakakondal, 2019-11-15:
Eclipse plugin is for ESP-IDF CMake based projects (4.x and above) only

w4sepp, 2019-11-18:
How eclipse will find the includes????

w4sepp, 2019-11-18:
Is there a way to define the include path for eclipse?

w4sepp, 2019-11-18:
Before I was using the python wrapper for compiling and I could define the include path for eclipse and therefore I could use all the editing features (intellisense ...).

With your eclipse plugin this is does not work any more for all the includes files eclipse does not find (and in a project thats a lot!!!). 

In the picture you see the example project (blink) - I selected the right ESP32 launnch target - and it compiled without errors. 

BUT !!!!   you see in the eclipse editor window some defines eclipse does not find (CONFIG_BLINK_GPIO, GPIO_MODE_OUTPUT,...) and therefore shows errtor. 

Maybe you have a trick which you can share how to solve this problem. 

I was checking other error-reports and from my point of view several refering to the same 
- Not able to access function definition #33
- Unresolved inclusion: <xxx.h> #31 
- After successful build still errors in the errors view and unresolved references #30

and your suggestion was always Make sure you select/configure "Launch Target" before triggering the build 
this does not help eclipse to find the right includes.
[image]

closed 2019-11-19 by w4sepp