espressif/idf-eclipse-plugin #30: After successful build still errors in the errors view and unresolved references

winkste, 2019-11-14:
Hi.

After I successfully build the hello world project, eclipse is still showing unresolved references and unknown variables. The error view also still mentions errors:

[image]

Same build:
[image]

kolipakakondal, 2019-11-15:
Make sure you select/configure "Launch Target" before triggering the build. Check here https://github.com/espressif/idf-eclipse-plugin#ConfigureLaunchTarget

This will take care of "unresolved references"

techny4287, 2019-11-26:
I don't know if this helps but I just had this issue as well, I can confirm that specifying a new launch target (even though i had one) resolved the issue. 
After taking a look at my project files, it seems like .cproject file for eclipse is empty or blank when this error appears.

sch0bert, 2019-12-25:
It didn't help me to make a new launch target, I still have the errors appearing. Regarding your comment @techny4287 about .cproject, I have it with information and doesn't matter regarding this issue. This is something else, else where.

kolipakakondal, 2020-02-12:
Hi All, IDF Eclipse Plugin v1.0.0-beta5 (Build: 1.0.0.202002120536) is published to the update site. Please get the latest https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

It has a fix for "unresolved includes" issue. Please check the release notes https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.0.0-beta5

closed 2020-02-12 by kolipakakondal

sch0bert, 2020-02-14:
Hello,

I updated the plug in to the latest and it still unrecognized most of the references:

[image]

Am I need to do something else?