espressif/idf-eclipse-plugin #31: Unresolved inclusion: <xxx.h>

PLAY233, 2019-11-15:
I installed “esp-idf-tools-setup-2.1.exe" "eclipse CDT" "IDF Plugins" according to the steps, and after completion, the Unresolved inclusion appeared: < XXX. H >
[image]

kolipakakondal, 2019-11-15:
Make sure you select/configure "Launch Target" before triggering the build. Check here https://github.com/espressif/idf-eclipse-plugin#ConfigureLaunchTarget

This will take care of "unresolved references"

**esp-idf-tools-setup-2.1.exe** is not required while working with idf-eclipse-plugin. Required tools can be installed using the "Install Tools" option from eclipse.
Check here https://github.com/espressif/idf-eclipse-plugin#installing-esp-idf-tools

PLAY233, 2019-11-15:
How do custom files import header files
[image]

kolipakakondal, 2019-11-26:
Check this https://github.com/espressif/idf-eclipse-plugin/issues/34 to add the external libs to the project

markku63, 2019-12-14:
> Make sure you select/configure "Launch Target" before triggering the build. Check here https://github.com/espressif/idf-eclipse-plugin#ConfigureLaunchTarget
> 
> This will take care of "unresolved references"

I have this same issue, configuring launch target has no effect at all to unresolved includes.

HansPeterHaastrup, 2019-12-24:
+1 for @markku63 's comment.
I, too, am unable to resolve includes. this occurs after setting the launch target.

I have two 'hello_world' projects and one of them is open in the editor. I can see that the error flag goes away on the project that is not open. If I open it it will reappear immediately.

Herschdorfer, 2019-12-28:
I can confirm the issue.

I have tried a fresh eclipse installation with the latest plugin and cannot get the includes to be resovled.

Plugin Version

```
Espressif IDF Plugins for Eclipse	1.0.0.201912090756	com.espressif.idf.feature.feature.group	ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD
```

CDT Version
```
Eclipse IDE for C/C++ Developers

Version: 2019-12 (4.14.0)
Build id: 20191212-1212
```

quartzpulse, 2020-01-07:
Thanks @kolipakakondal , This step resolved my inclusion issue.

su-Koch, 2020-01-13:
same issue here, tryed nearly everthing but no success

kolipakakondal, 2020-01-14:
> same issue here, tryed nearly everthing but no success
> @kolipakakondal this does not work for cmake projects, like the newer versioned projects of esp-idf are.

Is the resolved inclusion issue with the 3rd-party libs or even for the hello-world project?

markku63, 2020-01-14:
Happens even for hello-world or blink.

su-Koch, 2020-01-14:
I can confirm this. Happens for all Projects. Seams like the Eclipse indexer doesn't get the Paths from the CMake  *.txt files. If I add in absolute paths everthing works just fine. Any suggestions how to solve this? If needed I can add the Indexer logfile.

su-Koch, 2020-01-14:
What is different for me: the issue is related to the esp and freeRTOS includes. The std includes are found.

kambersimsek, 2020-01-21:
the problem still persists for beta 4 and beta 3.   
( windows 10- 64 bit
Eclipse IDE for C/C++ Developers
Version: 2019-12 (4.14.0)
Build id: 20191212-1212

kolipakakondal, 2020-02-11:
Here is the local build with the fix for "unresolved inclusion" issue in Windows OS, could you guys check and provide the feedback

Here is the build zip file to download
https://github.com/kolipakakondal/idf-eclipse-plugin-dev/blob/master/com.espressif.idf.update-1.0.0.202002111040.zip

Installation instructions:
https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugins-from-local-archive

kolipakakondal, 2020-02-12:
Hi All, IDF Eclipse Plugin v1.0.0-beta5 (Build: 1.0.0.202002120536) is published to the update site. Please get the latest https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

closed 2020-02-12 by kolipakakondal

mbratch, 2022-03-25:
This should not be closed. As of now, 3/25/2022 this is still an issue. I'm using Eclipse version 2021-12 (4.22.0) and ESP-IDF plugin version 2.4.2.202007271621. It behaves erratically. Sometimes _almost_ all of the errors will go away. And then sometimes it flags many of my standard and local includes. It seems to have a mind of its own. I've tried doing clean, fullclean, re-indexing, and every possible C/C++ Indexer option in the Eclipse menu. When it gets bad I need to go back over to use Visual Studio Code with it's ESP-IDF plug-in.

MNVasilev, 2023-02-08:
This should not be closed. As of now, 3/02/2023 this is still an issue.
[image]
[image]

edevald, 2023-05-13:
Same here, 05-13-2023.

[image]

edevald, 2023-05-13:
/

edevald, 2023-05-13:
The same is happening with me.

2023-05-13

[image]

kolipakakondal, 2023-05-15:
Hi @edevald Could you please share the IDE [Product Information](https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-view-product-information)? 

I tried to reproduce the issue with the below environment and couldn't see any errors after the build with a blink example.

> Operating System: mac os x
> Java Runtime Version: 17.0.5+9-LTS-191
> Eclipse Version: 4.25.0.v20220831-1800
> Eclipse CDT Version: 10.7.1.202208160035
> IDF Eclipse Plugin Version: 2.9.1.202304271158
> ESP-IDF v5.2-dev-412-gb3b0fe1898-dirty
> Python set for IDF_PYTHON_ENV: Python 3.8.9

[image]

kolipakakondal, 2023-05-15:
**Note: ESP-IDF 5.0 and higher + Espressif-IDE is having some unresolved symbol issues**

If you're working with ESP-IDF 5.0 and higher with Espressif-IDE, you might encounter unresolved symbol issues with certain projects, considering ESP-IDF 5.0 and higher is using some of the **C/C++ 20 language features** which are not yet supported by tine Eclipse CDT Indexer. These are only indexer problems it shouldn't cause any build failures.

To disable them temporarily,  Please refer to [this](https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-to-disable-the-unresolved-symbol-error-markers) doc.

mbratch, 2023-05-15:
@kolipakakondal thank you for the heads-up! Much appreciated. I am currently using ESP-IDF 5.0, but I have having Eclipse CDT indexer issues for quite some time, since I started using Eclipse with ESP-IDF. I don't recall the first version of ESP-IDF I used with Eclipse, perhaps 4.3.

edevald, 2023-05-15:
Sure, you'll find all info below.

I've made some other install attempts:

1: On a fresh VM with Windows 10.
[image]

2: On a Ubuntu VM machine.
[image]

The same thing happened in both environments: Unresolved inclusion.

I've noticed sometimes when I click "build" the Unresolved inclusions error goes away.

Sometimes deleting the build folder and rebuilding also works.

It's definitively a bug in both Eclipse and ESP-IDF, something with the indexer not indexing the header files.

kolipakakondal, 2023-05-17:
> Sure, you'll find all info below.
>
> I've made some other install attempts:
>
> 1: On a fresh VM with Windows 10. [image]
>
> 2: On a Ubuntu VM machine. [image]
>
> The same thing happened in both [...]

Ok, thanks for the update. Only after the project build headers will be resolved as the indexer look into the compiler_commands.json generated by CMake.

kolipakakondal, 2023-05-17:
> @kolipakakondal thank you for the heads-up! Much appreciated. I am currently using ESP-IDF 5.0, but I have having Eclipse CDT indexer issues for quite some time, since I started using Eclipse with ESP-IDF. I don't recall the first version of ESP-IDF I used with Eclipse, perhaps 4.3.

Most of the issues were reported earlier with ESP-IDF 4.x and IDE were fixed. We started surfacing new issues with 5.x

kolipakakondal, 2023-05-17:
Hi @AndriiFilippov I would like to request your help in summarizing the Indexer issues on various platforms. This will help us to prioritize our work internally and ensure that we are addressing the most important issues first.

You can follow the below format.

| OS Platform   | ESP-IDF Version | IDE Version | Project Name | Target | Status |
| ------------- | ------------- | ------------- | ------------- | ------------- | ------------- |
| macOS | 4.4.2 | 2.9.1| Blink Example | esp32 |