espressif/idf-eclipse-plugin #33: Not able to access function definition

skydot-user, 2019-11-16:
I have setup with the plugin. Everything compiles, I can check declaration of ESP functions but not able to go to function definitions. Is there any easy way to configure Eclipse to show function definition so that debugging is easy

kolipakakondal, 2019-11-26:
You should be able to navigate to function definitions once you link the esp-idf components to the project. We already have a PR for this - you can expect this in the next update.

However, you can also do the same with the following steps.

1. Right-click on the project
2. New > Folder
3. Click on the "<<Advanced"
4. Choose "Link to alternate location(Linked Folder)"
5. Click on Browse
6. Choose esp-idf components directory. example: /Users/kondal/esp/esp-idf/components
7. Finish the dialog
8. Wait for the "C/C++ Indexer" to finish the job - this will index all the components so that you should be able to navigate the functions.

bastianhjaeger, 2020-06-08:
Seems like an unpleasant workarround to me. 
I have a project with custom components, so including another folder called "components" does not work.

HrMitrev, 2020-08-25:
> You should be able to navigate to function definitions once you link the esp-idf components to the project. We already have a PR for this - you can expect this in the next update.
>
> However, you can [...]

Hi!
Before when I did that the indexer got a bit messed up and some variables/functions(particularly from the components) couldn't be resolved regardless of me trying to tweak the indexer settings.

Now unfortunately when I include the components folder it's automatically excluded from the build and I cannot access the settings to un-exclude it.

Any Ideas how to show the C/C++ Build settings?

[image]

HrMitrev, 2020-08-26:
Update: the problem was that the project was imported and the the original had the components included but with different path. The IDE didn't show the linked components, but at the same time didn't accept other folders with the same name. So I linked them with other name and it worked.

Now the biggest problem is [this](https://github.com/espressif/idf-eclipse-plugin/issues/113#issuecomment-678288515)

kolipakakondal, 2022-02-14:
Hello,

With our recent releases v2.4.0/v2.4.1, we have fixed header unresolved issues, indexer problems, and source code navigation problems. Now you should be able to navigate to the source code by pressing F3. 

Please get the latest changes using the update site. Please check our release page for more details.
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.0

Thank you

closed 2022-02-14 by kolipakakondal