espressif/idf-eclipse-plugin #34: Getting Started with MQTT

saahilshroff, 2019-11-18:
Hey!

I downloaded Mosquitto and successfully transmitted and received messages sent using command prompt. However, while trying to build a code on Eclipse throws me errors; primarily that the mosquitto.h is missing. Kindly help.

**ERROR SNIPPET:**
[image]

**CODE: (reference: Kolban's book on ESP32)**

#include "freertos/FreeRTOS.h"
#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "driver/gpio.h"
#include <stdio.h>
#include <string.h>
#include "mosquitto.h"

int app_main(int argc, char *argv[])
{
	 char *host="pc9100";
	 int port = 1883;
	 char *message = "hello world!";
	 char *topic = "greeting";

	 mosquitto_lib_init();
	 struct mosquitto *mosq = mosquitto_new(
			 NULL,  // Generate an id
			 true,  // Create a clean session
			 NULL);

	 int rc = mosquitto_connect(
			 	 mosq,   // Client handle
				 host,   // Host of the broker
				 port,   // Port of the broker
				 false);

	 if (rc != MOSQ_ERR_SUCCESS)
	 {
		 printf("Error with connect: %d\n", rc);
		 return(rc);
	 }

	 mosquitto_publish(
			 mosq,
			 NULL,
			 topic,
			 strlen(message)+1;
	 	 	 (const void*)message,
	 	 	0,
			false);

	 mosquitto_destroy(mosq);
	 mosquitto_clean_up;
}

kolipakakondal, 2019-11-22:
How did you add a mosquitto library? AFAIK, you can use **add_library** to the external libraries.

Check this:
https://github.com/espressif/esp-idf/issues/3613
https://cmake.org/cmake/help/latest/command/add_library.html#imported-libraries

kolipakakondal, 2019-11-25:
Here is the example:  https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/import_lib
Basically, this part: https://github.com/espressif/esp-idf/blob/master/examples/build_system/cmake/import_lib/main/CMakeLists.txt#L19-L23

kolipakakondal, 2020-02-12:
Closing this issue as there is no update for quite some time. 

IDF Eclipse Plugin v1.0.0-beta5 (Build: 1.0.0.202002120536) is published to the update site. Please get the latest https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

closed 2020-02-12 by kolipakakondal