espressif/idf-eclipse-plugin #35: Configuring Launch target - Instructions to review?

bbeliveau, 2019-11-19:
Hi the community!

I'm using Eclipse 2018-12 CDT. 
I'm trying to run the HelloWorld Exemple. So far I was able to do all steps in https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md#ConfigureLaunchTarget until the Configuration Launch Target step. The instrutions talk about a third dropdown which is this one I guess:

[image]

Unfortunately I don't have (or I don't see) this dropdown. Here is what I have:

[image]

I'm able to get this, which is, I guess, equivalent:

[image]
[image]

But then I'm expecting to set the Serial Port as shown in the instructions:

[image]

... but this is what I have instead:

[image]

So, is there a problem with the instructions or am I doing something wrong?

Thanks in advance.

kolipakakondal, 2019-11-19:
I'm able to see this issue when I play around with the launch configuration or when I create a new ESP Application configuration- Until we fix this, can you please do the following:
1. Create a new ESP-IDF Project in eclipse
2. This will automatically create the default launch configuration with the project name(it's selected by default). We needn't change anything here.
3.  Now you should be able to see the third drop-down(Launch Target) which you're asking for - there you should create a new launch target as mentioned in the doc.

bbeliveau, 2019-11-20:
Hi Kolipakakondal,

What you said works fine. Thanks.

But when I try to execute the HelloWorld on my ESP-WROVER-KIT 4.1 using the built-in JTAG, and following all the steps concerning debugging with JTAG in Eclipse, it fails because xtensa-esp32-elf-gdb --version returns nothing. So I get this in Eclipse:

[image]

And if I ask for the verion in the DOS shell I get this (nothing):

[image]

Any idea what the problem is?
Thanks.

kolipakakondal, 2019-11-20:
It seems to be xtensa-esp32-elf esp32-2019r1-8.2.0 is having this issue - let us check, can you try with this https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip and this seems to work fine.

While debugging choose the newly downloaded xtensa-esp32-elf-gdb file

bbeliveau, 2019-11-20:
I'm not sure how to choose the newly downloaded xtensa-esp32-elf-gdb file. I'm not enough familiar with all of this. So I tried this:

I deleted the original content of the folder and replaced by what is in the zip file. Then I try and still get the same error. Then I called export.bat to verify if the batch file was seeing and shown this:

[image]

Can you help on this please?
Thanks

bbeliveau, 2019-11-22:
Make some progress but now I have this:
[image]

It says (if I translate): No Connection have been established because the target computer has expressly refused it.

I did have my ESP-Wrover_Kit 4.1 connected to my computer.

Any idea?

kolipakakondal, 2019-11-23:
Have you started the GDB server from the command line?

Check this https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/index.html#jtag-debugging-configuring-esp32-target

bbeliveau, 2019-11-23:
That was it, it works, I can now debug. 
Thanks you so much Kondal for your help, very appreciated!
And nice video!

Thanks again.

szekelyisz, 2020-10-06:
I have a similar issue. I also thought that the Launch Target drop-down disappeared, but then I had to realize that you need to select an entry with a different type. Here's how it looks for me:

[image]

You can see the Launch Target drop-down but if you select '8k_firmware Configuration', then it disappears. '8k_firmware Configuration' is I think a Debug Configuration that I created manually under "ESP-IDF GDB OpenOCD Debugging":

[image]

I have no idea why it shows up in the Launch Bar.

I can see on  @bbeliveau's screenshots that the configuration selected there also had a different icon (that's like a crosshair, mine is an Espressif logo). The other looks like a "C program" type of icon for both of us, and that's the right one. Quite confusing.

BTW I can successfully start a debug session on my Ethernet Kit by selecting the "wrong" "8k_firmware Configuration" entry and clicking the Debug button on the Launch Bar after seeing `No esp launch target found. Please create/select the correct 'Launch Target'` on the build console in red...

And I am able to come up with a constellation like this:

[image]

Which is weird: how do I "Run" a configuration which is a _Debug_ configuration?

kolipakakondal, 2020-10-06:
@szekelyisz 

That's an Eclipse CDT standard naming convention for defining **Run** and **Debug** configuration.

8k_firmware  >> Application RUN configuration
8k_firmware Configuration >>> Application DEBUG configuration

The first drop-down defines the application launch mode - It can be either **Run** or **Debug**

If you want to **compile** an application - please select **Run** from the drop-down, this automatically selects the required configuration from the second drop-down which is `8k_firmware` in your case

If you want to **Debug** the application, select **Debug** from the first drop-down, this should ideally select the required debug configuration for the selected project. However, I could see some issues here. I would request to select it manually if you see some issue here, which is `8k_firmware Configuration` in your case. We will fix this soon.

We will also see how we can re-work on the naming convention for a run and debug to make it more clear.

steph0xr, 2020-12-05:
@kolipakakondal 

I verify the same issue on current master branch. It's very unconfortable and error prone, after modifying code while debugging, that code will not automatically build before launch debugging (the message "No esp launch target found. Please create/select the correct 'Launch Target'" is very fast and fade away as soon as debugger messages are shown). there is possibly some workaround to fix the build process while in debugging mode or we have to wait for new plugin release?

kolipakakondal, 2020-12-07:
Hi @steph0xr 

Sorry for the inconvenience, we have fixed this with #155

We will update here once it's released

steph0xr, 2020-12-18:
Hi @kolipakakondal ,
can you tell me when your fix on this issue has been planned to be released?
Development it's very slow changing view for build while debugging. Moreover after some swap between view it's not possible to build anymore, even in build view, and eclipse must be restarted each time.

thanks

forcecodema, 2021-01-02:
I have just installed idf-eclipse plugin on eclipse 2020-12 (windows 10). I can't see Build button widget. When I try to build project based on hello_world tmplate I see: No esp launch target found. Please create/select the correct 'Launch Target'

Please help

kolipakakondal, 2021-01-04:
Hi @forcecodema Eclipse 2020-12 is not yet supported by the plugin, could you switch to Eclipse 2020-09 and see that helps

JKSAR, 2021-02-04:
Hi,

I have installed idf-eclipse plugin on Eclipse 2020-09 (windows 10), but just like I @forcecodema I can't see Build button widget when creating an Espressif IDF Project.

[image]

Any ideia?

Regards!

kolipakakondal, 2021-02-04:
> 09

Hi, How did you install the Eclipse CDT Package? Have you installed this https://www.eclipse.org/downloads/packages/release/2020-06/r/eclipse-ide-cc-developers

Looks like some of the CDT components are not installed.

JKSAR, 2021-02-04:
> > 09
> 
> Hi, How did you install the Eclipse CDT Package? Have you installed this https://www.eclipse.org/downloads/packages/release/2020-06/r/eclipse-ide-cc-developers
> 
> Looks like some of the CDT components are not installed.

Hello @kolipakakondal !

I did the installation of Eclipse through this link that you passed and now the Menu and Widget are appearing correctly!

Thank you!

sigmaaa, 2021-08-17:
The problem related to disappearing the target's drop-down menu is fixed. You can find related changes in the [2.2.0 release](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.2.0)

closed 2021-08-17 by sigmaaa

Abdull, 2024-08-26:
Was this issue actually fully fixed, or is it back since at least *idf-eclipse-plugin v3.0.0*?

I have added `ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD	ESP-IDF Eclipse Plugin	3.0.0.202406051940	com.espressif.idf.feature` to my [Flathub-/flatpak-based Eclipse installation](https://flathub.org/apps/org.eclipse.Java). And I have added all kinds of *C/C++ Developer Tools* plugins. Yet my eclipse toolbar/coolbar looks like this:

[image]

... while the [documentation shows](https://github.com/espressif/idf-eclipse-plugin/tree/v3.0.0?tab=readme-ov-file#configuring-build-target) there should be a dropdown for choosing the target like this:

[image]

sigmaaa, 2024-08-27:
Hi @Abdull,

We don't see this problem in v3.0.0. Previously, we encountered this issue when users updated our plugin with older Eclipse CDT, leading to missing Batik dependencies. However, we’ve since added Batik to our plugin dependencies, ensuring they are installed during plugin installation.

Could you please check if the dependencies from our update site are installed?
[image]

sigmaaa, 2024-08-27:
Hi @Abdull,

also. please check if the launch bar is enabled in the Eclipse preferences:
[image]

Abdull, 2024-08-27:
Hi @sigmaaa, thank you for your help.

It turned out my Eclipse environment didn't have such Launch Bar entry in the preferences:
[image]

A [StackOverflow answer](https://stackoverflow.com/a/46952856/923560) points out the missing feature items/plugin: *LaunchBar* (*org.eclipse.launchbar.***) from the *CDT* software site ( https://download.eclipse.org/tools/cdt/releases/latest )

[image]

[image]

After having installed this and having restarted Eclipse, my toolbar/coolbar now includes the previously missing dropdowns:

[image]

As [several ESP users have this issue](https://esp32.com/viewtopic.php?t=19574) **maybe the *idf-eclipse-plugin* documentation should include a list of required/dependent-upon Eclipse feature items. And maybe these dependencies can also be expressed in *idf-eclipse-plugin* so that they are installed automatically?**