espressif/idf-eclipse-plugin #36: eclipse plugin was autoconfigured wrong

preal97, 2019-11-23:
Hi there!

I am running Ubuntu 19.04 with Eclipse 2019-09.
I installed the latest ESP-IDF (by git clone, as described in the Get Started) and the latest Eclipse Plugin from the URL.
I installed the Tools with the "Install Tools"-command in Eclipse.

When trying to build a application, it said in the console:
---
Errors occurred during the build.
Errors running builder 'CDT Core Builder' on project 'hello_world'.
Building hello_world
Cannot run program "/home/alexanderp/esp/esp-idf/tools/cmake" (in directory "/home/alexanderp/eclipse-workspace/hello_world/build"): error=13, Permission denied
Building hello_world
Cannot run program "/home/alexanderp/esp/esp-idf/tools/cmake" (in directory "/home/alexanderp/eclipse-workspace/hello_world/build"): error=13, Permission denied
---

After examining my PATH variable, i found out, that the entry for the "cmake" - Tool was not automatically made during the installation of the tool. 
PATH before:

/home/alexanderp/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin:
/home/alexanderp/.espressif/tools/xtensa-esp32s2-elf/esp-2019r2-8.2.0/xtensa-esp32s2-elf/bin:
/home/alexanderp/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:
/home/alexanderp/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190708/openocd-esp32/bin:/home/alexanderp/.espressif/tools/ninja/1.9.0/:
/home/alexanderp/.espressif/python_env/idf4.1_py2.7_env/bin:
/home/alexanderp/esp/esp-idf/tools:
/home/alexanderp/.local/bin:
/usr/local/sbin:
/usr/local/bin:
/usr/sbin:
/usr/bin:
/sbin:
/bin:
/usr/games:
/usr/local/games:
/snap/bin:

So after making the entry manually, 

--> /home/alexanderp/.espressif/tools/cmake/3.13.4/bin: (manually added to PATH)

i was able to build and flash a application.

kolipakakondal, 2019-11-25:
Thanks for reporting, we will verify.

kolipakakondal, 2019-11-26:
I've verified with Ubuntu and macOS X and couldn't reproduce this issue where cmake is missing in the PATH.  Do you have any specific case for this?

Maurizio1, 2019-12-04:
Kondal, preal97, I have finally found what's the problem and the temporary solution here: https://esp32.com/viewtopic.php?t=8148
Please look at the answer given by ESP_Angus: "What happens is this: Eclipse CDT searches the PATH (first Eclipse project path then environment variable PATH) for "cmake" to execute. Unfortunately it does not check if the "cmake" that it finds is an executable file.

If "$IDF_PATH/tools" is part of the search path, then Eclipse CDT tries to run the directory "$IDF_PATH/tools/cmake"...

We're going to send a patch to Eclipse to fix this, so that only executable files named "cmake" are executed.

In the meantime, if you edit the search path (either PATH environment variable or the Eclipse project search path) so that the directory which contains the cmake executable comes before "$IDF_PATH/tools/cmake", then it should start working."

I just put ~/esp/esp-idf/tools at the end of the PATH variable and now everything it's working fine even with latest Ubuntu version (19.10) and Eclipse (2019-09)!

kolipakakondal, 2019-12-04:
Submitted a patch to CDT https://git.eclipse.org/r/#/c/153770/

kolipakakondal, 2019-12-24:
Fix is part of the idf-eclipse-plugin-beta.4 update. Please get the latest changes.

closed 2019-12-24 by kolipakakondal