espressif/idf-eclipse-plugin #37: ESP-IDF Directory selection dialog doesn't appear #3

ralphs99, 2019-12-02:
Similar to: 'ESP-IDF Directory selection dialog doesn't appear #2'

"I followed step by step the instruction. Installed the plugin using URL method. Then I go to: Help > Espressif IDF Tools Manager > Install Tools and after pushing install nothing happens. Directory selection dialog doesn't appear."

After installing with v1.0.0-beta.2

Eclipse:
Version: 2019-09 R (4.13.0)
Build id: 20190917-1200
Python: 3.80
Java SE Development Kit 13

Any thoughts about the cause of the trouble?
Thanks in advance, Ralph

kolipakakondal, 2019-12-03:
That might be because of IDF_PATH directory would have already configured in the CDT preferences.
To verify, check this https://github.com/espressif/idf-eclipse-plugin#configureEnvironmentVariables

Look for IDF_PATH in the preferences. If it's configured wrongly, you can modify it in the preferences directly.

ralphs99, 2019-12-04:
Hello Kolipakakondal, Thanks for your reply.

I've had a look at the environment variables as indicated in the linked document and the Windows path looks OK:
[image]

The I've populated the Eclipse environment fields per the instructions:
[image]

There are are some small path differences as shown by the asterisked entries for Cmake and Python:
C:\Users\r\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin:
C:\Users\r\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin:
C:\Users\r\.espressif\tools\cmake\3.13.4\bin: ****
C:\Users\r\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin:
C:\Users\r\.espressif\tools\ninja\1.9.0:
C:\Users\r\.espressif\python_env\idf4.0_py3.7_env: ****
C:\Users\r\Desktop\esp-idf\tools:
$echo %PATH%

The console log is empty but the error log contains many errors. I've included a small section of the error log here. Let me know if you need more:
!ENTRY com.espressif.idf.core 1 0 2019-12-04 14:22:07.963
!MESSAGE Command id:com.espressif.idf.ui.command.install

!ENTRY org.eclipse.e4.ui.workbench 4 0 2019-12-04 14:22:07.966
!MESSAGE Command 'com.espressif.idf.ui.command.install' failed
!STACK 0
org.eclipse.core.commands.ExecutionException: Error executing 'com.espressif.idf.ui.command.install': java.lang.NullPointerException
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:170)
	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:498)
	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:487)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:213)
	at org.eclipse.e4.ui.workbench.renderers.swt.HandledContributionItem.executeItem(HandledContributionItem.java:438)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.handleWidgetSelection(AbstractContributionItem.java:449)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.lambda$2(AbstractContributionItem.java:475)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4173)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1057)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:3986)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3585)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1160)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1049)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:633)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:557)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:150)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:150)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:400)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:660)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:597)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1468)
Caused by: org.eclipse.e4.core.di.InjectionException: java.lang.NullPointerException
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:68)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:318)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:252)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:173)
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:156)
	... 32 more
Caused by: java.lang.NullPointerException
	at com.espressif.idf.core.util.PyWinRegistryReader.getPythonVersions(PyWinRegistryReader.java:47)
	at com.espressif.idf.ui.update.AbstractToolsHandler.execute(AbstractToolsHandler.java:68)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:283)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:95)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
	... 36 more

kolipakakondal, 2019-12-04:
Since IDF_PATH already configured, directory selection won't appear while installing tools. However, I could see all the required environment variables are already configured and that would be sufficient to move on to the next step.

Are you still facing the Tools installation error? That's not possible if the installation had failed.

ralphs99, 2019-12-04:
Unfortunately, the ESP IDF tools manager is still not working.

I'm not sure what you mean by "if the installation had failed". Eclipse appears to have installed correctly and I was able to install the IDF plugins. Clicking on the IDF tools manager opens a further choice of two selection 'Install Tools' & 'List Installed Tools', but clicking on either selection does nothing. 

Anything else I can try?

ralphs99, 2019-12-04:
In case it helps

Installation History:
[image]

Installed Software:
[image]

Installed Plugins:
[image]

gedolesku, 2019-12-05:
> Unfortunately, the ESP IDF tools manager is still not working.
>
> I'm not sure what you mean by "if the installation had failed". Eclipse appears to have installed correctly and I was able to install [...]

I have the same problem. It appeared when I tried to switch to ESP-IDF Version 4.0. In v3.3everything worked.

gedolesku, 2019-12-05:
> Unfortunately, the ESP IDF tools manager is still not working.
>
> I'm not sure what you mean by "if the installation had failed". Eclipse appears to have installed correctly and I was able to install [...]

And I found what was causing it. I could not run the Help > Espressiff Tools Manager > Install Tools command from the eclipse. I think this was because I used python from Anaconda. To solve this, I installed only Python 3.7 at different location (without anaconda). Then I installed everything again with "esp-idf-tools-setup-2.1.exe", but selected this new python version. After that, I could run Help > Espressiff Tools Manager > Install Tools command from the eclipse. Everything works now.

ralphs99, 2019-12-05:
Hi Gedolesku, thanks for your reply

I haven't installed Python from Anaconda, but I'll try your suggestion of Python 3.7 at a non-default location plus reinstalling Eclipse and let you know what happens.

ralphs99, 2019-12-06:
No luck with Gedolesku's suggetsion

Other things I've tries from: ESP-IDF Directory selection dialog doesn't appear #2 
I've looked at the Python entries in the registry per Serenagrl's sugggestion:
[image]
[image]

And edited the Java related line in the eclipse.ini per Ede2016's suggestion:
-startup
plugins/org.eclipse.equinox.launcher_1.5.500.v20190715-1310.jar
--launcher.library
C:\Users\r\.p2\pool\plugins\org.eclipse.equinox.launcher.win32.win32.x86_64_1.1.1100.v20190907-0426
-product
org.eclipse.epp.package.cpp.product
-showsplash
org.eclipse.epp.package.common
--launcher.defaultAction
openFile
--launcher.appendVmargs
-vm
C:\Program Files\Java\jre1.8.0_231\bin
-Dosgi.instance.area.default=@user.home/eclipse-workspace
-XX:+UseG1GC
-XX:+UseStringDeduplication
--add-modules=ALL-SYSTEM
-Dosgi.requiredJavaVersion=1.8
-Dosgi.dataAreaRequiresExplicitInit=true
-Xms256m
-Xmx1024m
--add-modules=ALL-SYSTEM
-Declipse.p2.max.threads=10
-Doomph.update.url=http://download.eclipse.org/oomph/updates/milestone/latest
-Doomph.redirection.index.redirection=index:/->http://git.eclipse.org/c/oomph/org.eclipse.oomph.git/plain/setups/

All to no avail, so I'm now running out of ideas!

kolipakakondal, 2019-12-17:
This is a weird issue and Windows platform-specific. The problem here is - eclipse is unable to load the Windows registry library - "winreg".

https://github.com/eclipse-cdt/cdt/blob/master/core/org.eclipse.cdt.core.native/src/org/eclipse/cdt/utils/WindowsRegistry.java#L38

To investigate further, can you provide eclipse configuration information as a file.
> About Eclipse >Installation Details >Configuration

kolipakakondal, 2021-02-16:
Closing this issue for now, please report us back if you see the same issue with our recent release.

closed 2021-02-16 by kolipakakondal