espressif/idf-eclipse-plugin #39: Another "Unable to open sdkconfig"

executer-uno, 2019-12-15:
When double clicking on the sdkconfig, it will prompt a dialog stating "Starting JSON Configuration Server" and the it will failed with a message on a new tab:

"Error retrieving output from the json configuration server, please check the console"

I have been check issue https://github.com/espressif/idf-eclipse-plugin/issues/4 but no result. I have slightly anotger messages in Console.
So I updated back to latest "Espressif IDF Plugins for Eclipse" v1.0.0.201911081309 after trying local build from #4 issue.

Im trying with hello world example project. I manage to build it and flash to ESP32, checked response from COM port. But Can not open sdkconfig tool:
[image]

ErrorLog looks clear, I only see problem in Console:
```
  File "C:\esp\esp-idf-v4.1-dev\tools\kconfig_new\confgen.py", line 94, in replace
    with open(sdkconfig_in, 'r') as f_in, open(sdkconfig_out, 'w') as f_out:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\E_CAD\\TEMP\\tmpoq0ztips'
```
[ErrorLog.log](https://github.com/espressif/idf-eclipse-plugin/files/3964920/ErrorLog.log)
[Console.txt](https://github.com/espressif/idf-eclipse-plugin/files/3964921/Console.txt)

It is not default TEMP location, was trying different paths, not helps.
Folder properties in explorer looks ok:
[image]

PS: I experienced a python issue during tutorial installation on clear machine
https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md
Python from
C:\Users\E_CAD\.espressif\python_env
don't see pip installed, so everything I was done - just remove
C:\Users\E_CAD\.espressif\python_env
directory and set IDF_PYTHON_ENV_PATH to system preinstalled python path
C:\Program Files\Python38
After that Eclipse ask to manually do
`C:\\Program Files\\Python38\\python.exe -m pip install -r C:\\esp\\esp-idf-v4.1-dev\\requirements.txt`
And after that compillation and flashing goes fine.

My system PATH variable from CMD:
```
C:\Users\E_CAD>PATH
PATH=C:\Program Files\Python38;C:\Program Files\Python38\Scripts;C:\Windows\syst
em32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v
1.0\;C:\Program Files\Git\cmd;C:\Program Files\Java\jdk-13.0.1\bin
```
IDF_PATH:
`C:\esp\esp-idf-v4.1-dev`
IDF_PYTHON_ENV_PATH:
`C:\Program Files\Python38`
OPENOCD_SCRIPTS:
`C:\Users\E_CAD\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190708\openocd-esp32\share\openocd\scripts`
PATH (Eclipse enviroment):
`C:\Users\E_CAD\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin;C:\Users\E_CAD\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin;C:\Users\E_CAD\.espressif\tools\cmake\3.13.4\bin;C:\Users\E_CAD\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190708\openocd-esp32\bin;C:\Users\E_CAD\.espressif\tools\mconf\v4.6.0.0-idf-20190628\;C:\Users\E_CAD\.espressif\tools\ninja\1.9.0\;C:\Users\E_CAD\.espressif\tools\idf-exe\1.0.1\;C:\Users\E_CAD\.espressif\tools\ccache\3.7\;C:\esp\esp-idf-v4.1-dev\tools;C:/Program Files/Java/jdk-13.0.1/bin/server;C:/Program Files/Java/jdk-13.0.1/bin;C:\Program Files\Python38;C:\Program Files\Python38\Scripts;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\Git\cmd;C:\Program Files\Java\jdk-13.0.1\bin;C:\Users\E_CAD\eclipse\cpp-latest-released\eclipse`

I also done everything similar on ubuntu - and there sdkconfig opens fine.

**Environment Information:** 
[Configuration.txt](https://github.com/espressif/idf-eclipse-plugin/files/3964926/Configuration.txt)

**ESP-IDF Tools Information:**
[ESP-IDF Tools.txt](https://github.com/espressif/idf-eclipse-plugin/files/3964928/ESP-IDF.Tools.txt)

kolipakakondal, 2019-12-17:
I was having a chat with other team members on this issue, their guess is - either antivirus or something has opened this file "C:\\Users\\E_CAD\\TEMP\\tmpoq0ztips". Can you investigate on that part and see?

Can you also verify does this file exist now?

executer-uno, 2019-12-18:
Hi. Thanks for answer.
It is a clear Win7 SP1 install. I even try again with firewall switched off.
File stated in error message not even appears in TEMP folder.

If I try to doubleclick on sdconfig object in project tree when SDK Configuration tab already previously opened - then nothing happens but focus changed to the tab. No new messages appears in console.

Can I do some other tests for clarification? I have no idea.

PS: In meantime I moved to another config Eclipse + Arduino Plugin so its not urgent, but I still have VM snapshot to make furter tests.

kolipakakondal, 2020-03-31:
Please check with the latest release v1.0.1

dasarne, 2020-05-12:
I have the same issue with V 1.0.1.202003301016 using Arch-Linux.

kolipakakondal, 2021-02-16:
Please verify the latest build and report back if you see same issue

closed 2021-02-16 by kolipakakondal