espressif/idf-eclipse-plugin #40: macOS Catalina Eclipse, PATH variable is messed up by ESP plugin

duaneellissd, 2019-12-17:
NOTE: Github is requires I put down the "Eclipse Environment" in an attachment.

**Describe the bug**
Execute BUILD under Eclipse fails with ```Some Python dependencies must be installed...```

This is caused by something unique in the ESP build system or plugin modifying the PATH variable when executing CMAKE.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the "env.sh" script to set tools variables. (I do not have these in my .profile), I run the script when I need these variables set.

2. On MACOS, to launch Eclipse from the command line, type: 

```/Applications/Eclipse.app/Contents/MacOS/eclipse```

3. Import any generic simple project and attempt to build.

The build fails, with ```Some Python dependencies must be installed``

4. Exit Eclipse, and build from the command line

for example, type: 

```cmake -G Ninja -DCMAKE_EXPORT_COMPILE_COMMANDS=ON /Users/mynamehere/ESP32/eclipse-project1```

RESULT: it works

5) Upon further inspection (I modified the CMAKE file to print the Shell Environment variables) and determined that either (A) eclipse, or (B) the EXPRESSIF plugin, or (C) the CDT build system forcibly places/inserts:  /usr/bin at the front of the PATH variable before executing the build. 

This causes the following problem:

When I run the ```env.sh``` script the EXPRESSIF version of PYTHON is first in my PATH, thus things work correctly from the command line.

When run from inside Eclipse - Because ```/usr/bin``` is inserted by something (A, B, or C above) the system provided python is found instead, which does not have the required features installed.

**IN CONTRAST** 

If I launch other tools, such as the editor Emacs, ie: ```/Applications/Emacs/Content/MacOS/emacs``` this same way, the shell variables are NOT modified.

It is only eclipse that inserts /usr/bin in my PATH.

**Environment Information:** 

(See attached 

[eclipse-env.txt](https://github.com/espressif/idf-eclipse-plugin/files/3971655/eclipse-env.txt)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools

See attached:  eclipse-env.txt

**NOTE** See right here - where it executes /usr/bin/python not the IDF python!

Executing /usr/bin/python /Users/duane/ESP32/esp-idf/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51.20170517 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20190708 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.9.0 (recommended, installed)

kolipakakondal, 2019-12-17:
To avoid this issue, while using eclipse,  please run "Help > Espressif Tools Manager > Install Tools", this will take care of installing tools(if not already) and configuring the required environment variables to run the builds.

https://github.com/espressif/idf-eclipse-plugin#InstallTools

duaneellissd, 2019-12-17:
I did that - problem not solved
also restarted eclipse after

** LOG OUTPUT from 'install tools' **
```
Installing tools...
This can take a while. Please be patient.
Executing /usr/bin/python /Users/duane/ESP32/esp-idf/tools/idf_tools.py install all
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, cmake, openocd-esp32, ninja
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2019r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190708 (already installed)
Skipping ninja@1.9.0 (already installed)

Executing /usr/bin/python /Users/duane/ESP32/esp-idf/tools/idf_tools.py install-python-env
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools>=21 in /Users/duane/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /Users/duane/ESP32/esp-idf/requirements.txt (line 4)) (42.0.1)
Requirement already satisfied: click>=5.0 in /Users/duane/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /Users/duane/ESP32/esp-idf/requirements.txt (line 8)) (7.0)
Requirement already satisfied: pyserial>=3.0 in /Users/duane/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /Users/duane/ESP32/esp-idf/requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in /Users/duane/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /Users/duane/ESP32/esp-idf/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /Users/duane/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /Users/duane/ESP32/esp-idf/requirements.txt (line 11)) (2.8)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /Users/duane/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /Users/duane/ESP32/esp-idf/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /Users/duane/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /Users/duane/ESP32/esp-idf/requirements.txt (line 13)) (0.25)
Requirement already satisfied: enum34; python_version < "3" in /Users/duane/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /Users/duane/ESP32/esp-idf/requirements.txt (line 11)) (1.1.6)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /Users/duane/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /Users/duane/ESP32/esp-idf/requirements.txt (line 11)) (1.13.2)
Requirement already satisfied: six>=1.4.1 in /Users/duane/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /Users/duane/ESP32/esp-idf/requirements.txt (line 11)) (1.13.0)
Requirement already satisfied: ipaddress; python_version < "3" in /Users/duane/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /Users/duane/ESP32/esp-idf/requirements.txt (line 11)) (1.0.23)
Requirement already satisfied: pycparser in /Users/duane/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /Users/duane/ESP32/esp-idf/requirements.txt (line 11)) (2.19)
Installing Python packages from /Users/duane/ESP32/esp-idf/requirements.txt

Executing /usr/bin/python /Users/duane/ESP32/esp-idf/tools/idf_tools.py export --format=key-value
```

** My Modified CMakefile.txt **

Specifically note the 4 "execute_process()" calls at the top
You will notice that even though I have installed the tools Eclipse is installing /usr/bin at the front of the path, I think that is the root of the problem.

```
# The following lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

# determine which python is in use
execute_process(COMMAND python -c "import sys; print(sys.executable)")
execute_process(COMMAND python2 -c "import sys; print(sys.executable)")
execute_process(COMMAND python3 -c "import sys; print(sys.executable)")
execute_process(COMMAND python -c "import os; print(os.environ[\"PATH\"])")

execute_process(COMMAND bash -c "set")

get_cmake_property(_variableNames VARIABLES)
list (SORT _variableNames)
foreach (_variableName ${_variableNames})
    message(STATUS "${_variableName}=${${_variableName}}")
endforeach()

set(SUPPORTED_TARGETS esp32)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(spp_client_demo)
```

*** Output from above ***
```
Building in: /Users/duane/ESP32/eclipse-project1/build
cmake -G Ninja -DCMAKE_EXPORT_COMPILE_COMMANDS=ON /Users/duane/ESP32/eclipse-project1
/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
[... 228 lines cut ...]
-- spp_client_demo_SOURCE_DIR=/Users/duane/ESP32/eclipse-project1
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
The following Python requirements are not satisfied:
click>=5.0
pyserial>=3.0
pyparsing>=2.0.3,<2.4.0
[... cut ...]
Call Stack (most recent call first):
  /Users/duane/ESP32/esp-idf/tools/cmake/build.cmake:396 (__build_check_python)
  /Users/duane/ESP32/esp-idf/tools/cmake/project.cmake:357 (idf_build_process)
  CMakeLists.txt:22 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/duane/ESP32/eclipse-project1/CMakeFiles/CMakeOutput.log".
cmake --build . -- -v
ninja: error: loading 'build.ninja': No such file or directory
Build complete (0 errors, 0 warnings): /Users/duane/ESP32/eclipse-project1/build
```

StavrosD, 2019-12-29:
I had similar problems.
Catalina uses zsh, not bash. You have to copy the settings from  bash_profile to zprofile
Open a terminal window and go to the esp-idf folder.
./install.sh
. ./export.sh

If zprofile does not exist, just copy bash profile.
cp ~/.bash_profile ~/.zprofile

igrr, 2020-03-17:
Indeed there seems to be an issue here with the PATH variable, for some reason `/usr/bin` is prepended to it:

```
/usr/bin:/Users/duane/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin:/Users/duane/.espressif/tools/xtensa-esp32s2-elf/esp-2019r2-8.2.0/xtensa-esp32s2-elf/bin:/Users/duane/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:/Users/duane/.espressif/tools/cmake/3.13.4/CMake.app/Contents/bin:/Users/duane/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190708/openocd-esp32/bin:/Users/duane/.espressif/python_env/idf4.1_py2.7_env/bin:/Users/duane/ESP32/esp-idf/tools:/opt/local/bin:/opt/local/sbin:/usr/local/bin:/bin:/usr/sbin:/sbin:/usr/local/share/dotnet:/opt/X11/bin:~/.dotnet/tools:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/Users/duane/.espressif/tools/ninja/1.9.0/
```

I've seen this happening on macOS occasionally, but so far haven't been able to narrow down what causes this.

kolipakakondal, 2020-03-23:
> Indeed there seems to be an issue here with the PATH variable, for some reason `/usr/bin` is prepended to it:
>
> ``` [...]

Thanks for the pointer @igrr I'll investigate further on PATH issue

RJPoelstra, 2021-11-12:
I'm experiencing the same problems.
I'm running macOS 12.0.1 and Eclipse 2021-06.

Is there any news on this issue? Maybe a workaround?

kolipakakondal, 2021-11-12:
Hi @RJPoelstra This will be fixed with https://github.com/espressif/idf-eclipse-plugin/pull/397

**Workaround**:
When you run `ESP-IDF Install Tools > Install Tools` from Eclipse, it will output the `PATH` value. Please take this PATH value and replace it with the  `Preferences > C/C++ > Build > Environment > PATH` value

I hope that helps.

RJPoelstra, 2021-11-12:
Hi,

Thanks for the response.
The workaround made it worse, as the $PATH from the output doesn't get expanded, so basic tools are no longer found.

I'll wait for the pull request to get merged.

Thanks.

kolipakakondal, 2022-04-04:
Hello, This is fixed with earlier releases. Please verify the latest builds and let us know if you see problems.

closed 2022-04-04 by kolipakakondal