espressif/idf-eclipse-plugin #41: Help > Espressif IDF Tools Manager > Install Tools - Installation running more than 24 Hours - Eclipse /Windows10/Local

ShanmugaCA, 2019-12-22:
Hello,
While installing the Help > Espressif IDF Tools Manager > Install Tools in the Eclipse 
(Eclipse IDE for C/C++ Developers Version: 2019-12 (4.14.0), the installation is keep running for more than 24 hours.

Steps to reproduce:
----------------------
- I followed the instruction in the https://github.com/espressif/idf-eclipse-plugin
- Downloaded the local folder
- Provided the ESP-IDF Directory path to get started with the installation and.... after this the message in the right bottom of the Eclipse keep running xxx of 256 MB and i waited for 24 hours !

[image]

- Got the local from :
https://dl.espressif.com/dl/idf-eclipse-plugin/updates/com.espressif.idf.update-v999-24-g120495da.zip

Any help?

Thanks in advance
Shanmuganathan

kolipakakondal, 2019-12-24:
That's the heap status. You can enable/disable this https://dzone.com/articles/show-heap-status-eclipse

When you run **Help > Espressif IDF Tools Manager > Install Tools**,  install status will be shown on the console. Check this https://github.com/espressif/idf-eclipse-plugin#espressif-idf-tools-console

If you see any error on the console, let us know

ShanmugaCA, 2019-12-25:
Hello,
thanks for responding!

* Regarding the heap, the said heap value changes continuously, assuming there is some update happening in the back.

* There is no update in the console

* I was trying to  install the tools thru the : Eclipse -> Help->Install New software, there are two options possible. 
       1. From the provided web link and unfortunately, that link is invalid ( this is the link provided - https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/)
       2. The second option is obviously, download and install from local (did that with the Zip file ( this is available  here- https://dl.espressif.com/dl/idf-eclipse-plugin/updates/com.espressif.idf.update-v999-38-g7360fdc2.zip) 

Now the issue is in the above "2. ". The Eclipse keep running (i am assuming that it is keep running based on the change in Heap status :) )  
 - I did Help -> Install new software-> Add -> Local (gave some name and the location pointing to the Zip file downloaded in "2.".

Not sure where I am missing 

-Shanmuganathan

kolipakakondal, 2020-01-06:
> Regarding the heap, the said heap value changes continuously, assuming there is some update happening in the back.

No, that's not the way to look at it. When you are running an eclipse, there are some background processes that will be running continuously hence heap will be changing as well.

> 1. From the provided web link and unfortunately, that link is invalid ( this is the link provided - https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/)

Though you would see 403 Forbidden(that's not an error) page if access directly but it should work if you are updating through eclipse

Probably, error log should provide some insights into the issue, can you provide that.
https://github.com/espressif/idf-eclipse-plugin#error-log

ShanmugaCA, 2020-01-20:
Hi!

_>> Probably, error log should provide some insights into the issue, can you provide that._

This is the error log:

+++++++++++++++++++++++
Unable to read repository at https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/content.xml.
Unable to read repository at https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/content.xml.
PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
+++++++++++++++++++++++

[image]

kolipakakondal, 2020-01-21:
Looks like Java version is the problem. Can you make sure Java 8 and above installed in the system?

Even if you're working with Java 8, please get the latest version of Java 8. That should solve the problem!

Similar thread https://esp32.com/viewtopic.php?f=13&t=12327&start=10

ShanmugaCA, 2020-01-21:
Hi! 
The Java version is much latest than Version 8. I have the Java version 13.0.1 in the machine.

Am i missing something else ?

[image]

kolipakakondal, 2020-01-21:
Okay, interesting! let's check what's the java version used by eclipse. To do that,

Help > About Eclipse >Installation Details > Configuration > Look for "-vm" argument

It shows something like this 

```
-vm
/Library/Java/JavaVirtualMachines/jdk-13.0.2.jdk/Contents/Home/bin/../lib/server/libjvm.dylib
```

ShanmugaCA, 2020-01-21:
thanks for your quick replies, it 
-vm
C:\Program Files\Java\jre1.8.0_231\bin\server\jvm.dll

[image]

kolipakakondal, 2020-01-21:
Can you try these?
1. Get the latest Java 8 Update 241 
2. Try this solution https://stackoverflow.com/a/36427118/1812235

It seems to be certain configuration parameters(org unit, etc) are missing for certificates, maybe that's causing the issue. At least I don't see that issue on macOS, but let me also try on Windows.

ShanmugaCA, 2020-01-24:
Thanks again, but no use. It is not working

- Java is 241 now, see the snippet
[image]

- error log from Eclipse is below:
[image]

kolipakakondal, 2020-01-28:
> 2. Try this solution https://stackoverflow.com/a/36427118/1812235

Can you also try this?

kolipakakondal, 2021-02-16:
Please verify with the latest build, report back if you have the same issue even after installing Java 11 and higher.

closed 2021-02-16 by kolipakakondal