espressif/idf-eclipse-plugin #42: "Git Executable Location" is required for install tools

sch0bert, 2019-12-25:
**Description**

I'm trying to setup my environment in Eclipse to run the Hello World project by following the steps from:

https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md

But when I get to:

https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md#installing-esp-idf-tools

The steps are not the same as the ones show in the latest IDF-Eclipse-Plugin. In the tutorial you can see that is only required to add the path to the IDF and now you need also to add the path for "Git Executable Location" which is weird because there is no longer any file for git update in the latest IDF (3.3.1 and 4.0-beta2). I believe it used to be "esp-idf.git". So in order to continue, I selected any file from the .git folder from the IDF. I don't think this is right or at least there should be something documented in the tutorial about it because in case you don't fill it, it will not download any esp-idf-tools.

**Screenshots**

[image]

StavrosD, 2019-12-29:
Install git or Xcode.
Xcode installs Apple git.
I had Xcode installed, the Git executable location is: /usr/bin/git

sch0bert, 2019-12-29:
Well I forgot to mentioned that I'm using Windows

StavrosD, 2019-12-29:
Install git from the official website
https://git-scm.com/download/win

kolipakakondal, 2020-01-06:
Git is required for "install tools" action as it's mentioned in the pre-requisites. https://github.com/espressif/idf-eclipse-plugin#installing-prerequisites

Documentation is updated now, please check this https://github.com/espressif/idf-eclipse-plugin#InstallTools

Thanks for reporting.

closed 2020-01-06 by kolipakakondal