espressif/idf-eclipse-plugin #43: still unresolved includes using latest idf-eclipse-plugin

HarryDuerr, 2019-12-27:
after building and flashing successfully the blink template I see still unresolved inclusions in my source file.
eclipse project has been created via [https://github.com/espressif/idf-eclipse-plugin#create-a-new-project-using-esp-idf-templates](url)

I use "latest" espressif plugin (see attached file)

[image]

I expect inclusions to be resolved after building for target
 
**Environment Information:** 
[eclipse info.log](https://github.com/espressif/idf-eclipse-plugin/files/4005753/eclipse.info.log)

**ESP-IDF Tools Information:**
Executing C:\Users\hpd\AppData\Local\Programs\Python\Python37\python.exe C:\Users\hpd\Documents\HPDPool\Privat\HpdDev2020\ESP32\ESPV4\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51.20170517 (recommended, installed)
* cmake: CMake build system
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20190313 (recommended, installed)
* mconf: menuconfig tool
  - v4.6.0.0-idf-20190628 (recommended, installed)
* ninja: Ninja build system
  - 1.9.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)

HarryDuerr, 2019-12-27:
OK! I found a work around.
I tried again with eclipse 2019-09 and IDF-Eclipse Plugin Beta 3.  Now includes are recognized as expected. I leave it for the team to figure out which one of both are causing the issue.

Herschdorfer, 2019-12-28:
I tried that and can confirm that 19-09 ans beta 3 let the warnings disappear.

krzychb, 2019-12-30:
This pesky issue finds it's way back over and over again, and that is really annoying.

To verify it I have just made fresh installation of the plugin Beta 4 with Eclipse 2019-12 on Windows 10 and Linux Ubuntu 18.04. The issue shows up for me on Windows only. I made the requisite [Configuring Launch target](https://github.com/espressif/idf-eclipse-plugin#ConfigureLaunchTarget).

I believe that https://github.com/espressif/idf-eclipse-plugin/issues/29 and https://github.com/espressif/idf-eclipse-plugin/issues/31 are reported for Windows as well. 

Do other of you see the same? Or maybe some of you do not see it on Windows at all (for the latest Beta 4 with Eclipse 2019-12)? Hopefully we can narrow down the scope of troubleshooting.

su-Koch, 2020-01-15:
Can confirm this for windows and 2019-12 and current idf plugin. Is there a way to test it with beta 3? I don't know how to get it. Are there any suggested solutions?

infoschwab, 2020-01-23:
I can conference this for Windows and Eclipse 2019-19 with esp-idf 4. No Problem on Ubuntu in Virtual Box with Same install Procedere. I already made a reinstall. 
I copied the folder ...\esp-idf\components into the Projektfolder in the Workspace and the probleme is gone. This is not the solution, but it shows that the problem is, that the Indexes dies not find the path to this path. I realy new to Eclipse but i canˋt find a way to tell the indexer the Library path.
How can i do this?

kolipakakondal, 2020-01-24:
Anyone up for the short call(15mins)? We can use the  https://zoom.us/ for screen sharing.  Let me know the timezone if anyone is interested to join the call.

infoschwab, 2020-01-24:
Yes, i like to Support you, but My Englisch is Not Good. My timezone is gmt+1 Germany Near Stuttgart.

su-Koch, 2020-01-24:
@infoschwab this is so funny because I'm located near stuttgart too. I got it working with vscode. Here I'm pretty shure we're on the right track. It has to do with the indexer not finding the right path. So I expect exactly the same problem you do.

infoschwab, 2020-01-24:
Yes, i do think, there must be a way to tell the indexer the path for the Librarys. I found a lot of tutorials to set the path, but in My Eclipse i dont find the menueentry. There Must be a way to do this but i am Not firm with Eclipse. Perhaps i change to vscode.

su-Koch, 2020-01-24:
I am doing basic tests with both eclipse and vscode at the moment. Both of them doesn't work properly in my opinion. There are still a lot of issues.

HarryHpd, 2020-01-24:
I stated in a previous post, that meanwhile I am quite happy with this toolchain.

Yes, there are some annoying flaws, but since I am used to the toolchain and the way how to handle it, the "unresolved inclusion" issue became relatively rare and can mostly be solved by reindexing. 

Looking back to my initial problems I would like to say, they were derived form jumping between projects, not understanding how to handle run/Debug configurations etc. or changing between run and Debug configurations. Unfortunately I can still not find a reproducable way to cause the "Unresolved Inclusion" issue. 

I am currently suffering more from "Permission Error" but for this issue I found as well a workaround to live with it.

BTW, I live near Mannheim  :-)

kolipakakondal, 2020-01-27:
That's cool that you guys are from Germany :-) I was in Stuttgart in Oct for EclipseCon2019 and we presented this esp-idf eclipse plugin. Check this https://youtu.be/DqnZy4d03_s . It might be interesting to see an end-to-end flow

For guys, who could help me in understanding the issue, please use my skype name: **live:kondal.kolipaka**. I'll be available from CET 10AM to 12PM

krzychb, 2020-01-27:
@kolipakakondal indeed https://youtu.be/DqnZy4d03_s from EclipseCon2019 attracted me to check the plugin!

Regarding the issue with unresolved includes on Windows, what diagnostic information or configuration details do you need? I would prepare it upfront.

kolipakakondal, 2020-01-27:
@krzychb Probably a couple of things:

- If possible provide sample project which was already built and having errors
- Provide the CDT Build log:  Preferences > Build > Logging
- Provide error log https://github.com/espressif/idf-eclipse-plugin#error-log
- Eclipse installation details https://help.eclipse.org/2019-12/index.jsp?topic=%2Forg.eclipse.platform.doc.user%2Ftasks%2Ftasks-130.htm
- Comile_commands.json file from the project build directory

**Check these:**

- CDT Build Console - are you seeing any error during the build?
- Project > C/C++ Index > Rebuild - Does it help?
- Have you selected the esp launch target before the build?

krzychb, 2020-01-27:
@kolipakakondal please see below requested diagnostic information:

### Logs
* I did the tests using `/esp-idf/examples/get-started/blink`
* [global-build.log](https://github.com/espressif/idf-eclipse-plugin/files/4116020/global-build.log)
* [error-log.txt](https://github.com/espressif/idf-eclipse-plugin/files/4116032/error-log.txt)
* [installation-details.zip](https://github.com/espressif/idf-eclipse-plugin/files/4116087/installation-details.zip) (I made screenshots but some windows do not fit the screen to provide all the details. Is there a better way export this information instead of making several screenshots?)
* [compile_commands.zip](https://github.com/espressif/idf-eclipse-plugin/files/4116108/compile_commands.zip)

### Checked those:
* CDT Build Console - no build errors
* Index Rebuild - Indexed 'blink' (9 sources, 35 headers) in 0.42 sec: 168 declarations; 2,354 references; 3 unresolved inclusions; 10 syntax errors; 33 unresolved names (1.3%)
* ESP launch target selected 
[image]
* Checked. Project cleaned and rebuild. Still see unresolved inclusions:
[image]

krzychb, 2020-01-27:
@kolipakakondal attached is another copy of `compile_commands.json` file discussed during our call.

[compile_command_json.txt](https://github.com/espressif/idf-eclipse-plugin/files/4116528/compile_command_json.txt)

kolipakakondal, 2020-01-28:
Thanks @krzychb for providing the detailed info and joining the call. It seems to be headers indexing is not happening in Windows OS and I'm able to reproduce the issue. I'll update here once the fix is ready!

HarryHpd, 2020-01-29:
@Kondal  

as already said I had no "unresolved inclusion issues" in my windows setup for a long time. 

But recently, after changing from Run to Debug configuration I got again "unresolved inclusion" for all Includes in component directory.

I saw something like "invalid Build configuration" in console too.

Reindexing and rebuild after clean did not help. After restarting eclipse and disconnecting my WROVER Kit I could build and the issue disappeared.

Arag0to, 2020-02-02:
@kolipakakondal 

You migth want to get rid of the "Windows" label since I have the same problem on Ubuntu 18.04 but only a bit more complicated.

On the main page, "main.c", no problem but when I create a new file or import a class. Then I get the unresolvel inclusion issues even if the compiler report 0 errors. There is also that problem with one of my cpp lib that I don't understand.

Here is a link to a screenshot of my Eclipse showing the problem.
[https://servimg.com/view/20121761/8](url)

My Unbuntu is install on a VMWare virtual machine so if you want I can send it to you. It takes around 15-20 go.

If you want me to create a new issue, tell me. I will do my best. 2 weeks ago I new nothing about all this.

krzychb, 2020-02-02:
@Arag0to from the screenshot (of your Eclipse showing the problem) it looks like the launch target is not selected. Could you try selecting the target, cleaning and rebuilding the project?

Arag0to, 2020-02-02:
Thank you @krzychb, happy to know it wasn't a big problem. But i still got the cpp library not recognize in the main.c. I would like to know why. Tryed to include <fstream> with the same problem, If you know the answer please tell me but I don't think this problem belongs to this issue.
[image]

I would also like to understand why the target matter ? From what I understand, the target specifies on which port flashing the Esp32 but it worked and I have 0 port connected on my machine right now.

JAICHANGPARK, 2020-02-06:
Hi there 
I have some same problem about this issue in windows OS

[image]

any updates?

kolipakakondal, 2020-02-10:
Could you guys check this build and provide the feedback. This should fix an issue related to "unresolved inclusion" errors in Windows OS

Here is the zip file to download
https://github.com/kolipakakondal/idf-eclipse-plugin-dev/blob/master/com.espressif.idf.update-1.0.0.202002101000.zip

Installation instructions:
https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugins-from-local-archive

krzychb, 2020-02-11:
@kolipakakondal the new build works great!

[image]

Thank you for resolving this issue 👍

su-Koch, 2020-02-11:
I can also confirm that the issue seams to be solved. Thanks a lot. Is there a function to flash without building ?

kolipakakondal, 2020-02-12:
Hi All, IDF Eclipse Plugin v1.0.0-beta5 (Build: 1.0.0.202002120536) is published to the update site. Please get the latest https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

closed 2020-02-12 by kolipakakondal

HarryHpd, 2020-02-12:
﻿..Yes, I am working with your fix since some time now and it appears to work as expected.

@Kondal . Thanks for your help.

﻿

HarryHpd, 2020-02-13:
@Kondal. Unfortunately "unresolved inludes" issue came back today. 

I can not give you more details than the fact itself .

I upgraded meanwile to the latest Plugin offered by eclipse update check.

Reindexing does not help. deleting build directory sometimes helps, but the error comes back after a short while.

But I can remember one thing. Last time the problem appeared just after fireing the debugging launch configuration for my wrover kit.

niruhegde, 2020-02-21:
@kolipakakondal  

I have just installed plugin. Still I am getting same issue.
Eclipse 2019-12
[error.log](https://github.com/espressif/idf-eclipse-plugin/files/4237355/error.log)

And when I clean the project I will get the below error :
ninja clean
ninja: error: loading 'build.ninja': No such file or directory
Build complete

I have also tried to download and updating archive. But as I have already installed latest plugin, archive is not installed.

Thanks
-Niru

su-Koch, 2020-02-21:
hei niruhegde, 
did you download esp-idf via git clone or via zip? I had a similar error when downloading zip and dependencies got lost but solved it with git clone in windows cmd line.

niruhegde, 2020-02-21:
Hi su-Koch,

I have cloned repo. My esp-idf version is ESP-IDF v4.2-dev-414-g132cc67c0-dirty.
I am using Ubuntu 19.10.

Thanks
-Niru

eliaxyz, 2020-02-21:
The problem still remains in Mac OS.

After copying example project "blink", firstly no errors are shown and the build also works without errors.

But after restarting eclipse, the editor shows the errors as in the screenshot of JAICHANGPARK commented 16 days ago.

And while trying to build the project, there happens also a Java-Error:

> Errors occurred during the build.
> Errors running builder 'CDT Core Builder' on project 'test7'.
> java.lang.NullPointerException

mvgorban, 2020-02-26:
Hi! The same issue. Windows10/Version: 2019-12 (4.14.0)/Build id: 20191212-1212/ Espressif IDF Plugins for Eclipse 1.0.0.202002201116.

shivam5594, 2020-05-19:
If warnings doesn't resolve even after building the code, open terminal in that project and run idf.py fullclean. Somehow clean in eclipse is not cleaning, it builds the code again.

Just-Eugene, 2020-05-31:
Guys, I have the same annoying issue.
I'd **marked** these **options**:
[image]
and **got**:
[image]
But it **doesn't solve** the issue cause I can reach only headers but not sources.
Earlier when I was using _ESP IDF v3.3_ with _make_ build system without use the plugin I just added (**linked**) needed sources, e.g. _IDF_ parent directory:
[image]
But this submenu can't be accessible when the plugin is used.
Maybe did someone find any solution? Or is the only way to use my projects as CDT general ones  without the Espressif plugin?

kolipakakondal, 2020-06-01:
Hi @Just-Eugene - Paths and Symbols got removed with the latest Eclipse CDT.

You can follow the below instructions to link esp-idf to the project so that function definitions will be linked.
https://github.com/espressif/idf-eclipse-plugin/issues/33#issuecomment-558574707

I hope that helps.

Just-Eugene, 2020-06-01:
> Hi @Just-Eugene - Paths and Symbols got removed with the latest Eclipse CDT.
> 
> You can follow the below instructions to link esp-idf to the project so that function definitions will be linked.
> [#33 (comment)](https://github.com/espressif/idf-eclipse-plugin/issues/33#issuecomment-558574707)
> 
> I hope that helps.

Thank you!

fweiss, 2020-07-22:
I had this problem as well with fresh install Eclipse 2020-06 (4.16..0) and ESP_IDF Eclipse Plugin 1.1.1, after creating a new ESP_IDF project with the default template.

Resolved after Project > Build Project

There were also symbol errors in the editor, such as "NULL". Resolved this by:

Project > Properties > C++ General > C++ Indexer
Check the option Enable project-specific settings
Check the option Index source file not included in the build