espressif/idf-eclipse-plugin #44: Eclipse F3 navigation not as expected

HarryDuerr, 2020-01-02:
If I press  (F3) while having cursor pointed to a function call like

` esp_mqtt_client_publish(mqtt_client, CONFIG_EXAMPLE_PUBLISH_TOPIC, buf , strlen(buf), 1, 0);`

I expect the type definition to be shown and after further (F3) , the implementation of this function to be shown.

But after showing Type definition as expected, the second (F3) will not show me the function implementation.

Meanwhile i learned, that {STRG}{TAB} at least will bring up the source file where a function is implemented. But this works only, if the function's source file has the same name as the header file and is as well located in the same directory.

I am using Eclipse 19-09 with ESP-IDF-ECLIPSE-PLUGIN  (Beta 4). 

Executing C:\Users\hpd\AppData\Local\Programs\Python\Python37\python.exe C:\Users\hpd\Documents\HPDPool\Privat\HpdDev2020\ESP32\ESPV4\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51.20170517 (recommended, installed)
* cmake: CMake build system
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20190313 (recommended, installed)
* mconf: menuconfig tool
  - v4.6.0.0-idf-20190628 (recommended, installed)
* ninja: Ninja build system
  - 1.9.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)

kolipakakondal, 2020-01-06:
This can be addressed by linking the esp-idf "components" folder to the project. Please follow the instructions from here https://github.com/espressif/idf-eclipse-plugin/issues/33#issuecomment-558574707

HarryHpd, 2020-01-06:
Yes, this did the trick! This is a really huge break through in usability. Many thanks for this hint.

Let me add for subsequent readers, that indexing takes a really long time and that I had some flaws until the project had been completely rebuild.

kolipakakondal, 2020-01-07:
Happy that it's working for you!  yes, indexing takes more than 5mins since esp-idf components have a lot of source files. But that's a one-time activity for the project.

If you see any issues after linking, let us know - we can improve further on this.

HarryHpd, 2020-01-07:
I am currently indeed chasing the reason for a related issue. I get more and more "unresolved Symbol" warnings in various eclipse ESP-IDF projects (all derived from Examples).  Sometimes I can get rid of them by "index rebuild", but mostly they persist without any obvious reason.

kolipakakondal, 2020-01-07:
Are the errors is from esp-idf "components" or from the project source code? I've seen certain components are not getting indexed properly causing the errors on the project.

HarryHpd, 2020-01-07:
the errors are in the "components" directory which I added as you recommended AND in my project sources.

The following image shows one components source file deliberately chosen.

[image]

HarryHpd, 2020-01-16:
I owe you an update with regards to the "unresolved items issue".
I am heavily working with your plugin and I can say, the more you are used to the toolchain the less this issue appears. 
But Reproducing the issue for 100% is still not possible.

But I have following hints:
1. It appears to happen sometimes if you change from a run configuration to a debug configuration
2. sometimes index rebuild helps
3. Sometimes index is not rebuilding anymore. 
4 sometimes complete rebuild after clear helps

kolipakakondal, 2020-01-17:
Thanks @HarryHpd  for the update. I'm trying various scenarios to reproduce the issue. I hope to provide you the fix soon.

Feel free to create different tickets, if you see any other issues with the plugins. That really helps us to improve the plugin quality.

HarryHpd, 2020-01-17:
... I know, derived from my description it will be extremely difficult to find the root of this issue.

This certainly because you will not make the silly mistakes which beginners usually make. 

What about finding someone with little experience in eclipse playing with your examples and watch him?

kolipakakondal, 2020-01-24:
I'm able to see the errors in the project "components" directory when we try to navigate to the function definition from the project source code.  Since we added the esp-idf "components" to the project - we are able to see error markers on the project level also.

Let us work on this! You can expect the fix in the coming releases.

alirana01, 2021-11-05:
The issue is addressed in upcoming release.

kolipakakondal, 2022-02-14:
Hello,

With our recent releases v2.4.0/v2.4.1, we have fixed header unresolved issues, indexer problems, and source code navigation problems. Now you should be able to navigate to the source code by pressing F3.

Please get the latest changes using the update site. Please check our release page for more details.
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.0

Thank you

closed 2022-02-14 by kolipakakondal