espressif/idf-eclipse-plugin #45: Eclipse does not mark the line when breakpoint is hit

axbusch, 2020-01-03:
I use Eclipse 2019-09 and configured everything as described.
I can debug my esp32, i can step trough my code etc

But the actual line in the eclipse editor is not highlighted which is a little bit anoying as i can't really see where i am in my code

This is the openocd output after some breakpoint hits

```
** Programming Started **
auto erase enabled
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
wrote 155648 bytes from file D:/esp/projects/testdebug/build/blink.bin in 2.159124s (70.399 KiB/s)
** Programming Finished **
** Verify Started **
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
read 151936 bytes from file D:/esp/projects/testdebug/build/blink.bin and flash bank 0 at offset 0x00010000 in 0.808046s (183.622 KiB/s)
contents match
** Verified OK **
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
Info : esp32: Core 0 was reset (pwrstat=0x1F, after clear 0x0F).
Info : esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
Info : Detected debug stubs @ 3ffb2a90 on core0 of target 'esp32'
Info : Target halted. PRO_CPU: PC=0x400D3B90 (active)    APP_CPU: PC=0x400E2D12
Info : Target halted. PRO_CPU: PC=0x400D3BE6 (active)    APP_CPU: PC=0x400E2D12
Info : Target halted. PRO_CPU: PC=0x400D3BE6 (active)    APP_CPU: PC=0x400E2D12
```

List of installed tools and plugin

Espressif IDF Plugins for Eclipse	1.0.0.201912090756	com.espressif.idf.feature.feature.group	ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD

```
Executing C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe D:\esp\idf\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51.20170517 (recommended, installed)
* cmake: CMake build system
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20190313 (recommended, installed)
* mconf: menuconfig tool
  - v4.6.0.0-idf-20190628 (recommended, installed)
* ninja: Ninja build system
  - 1.9.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
```
  

Do you have any idea what might be wrong? I can provide further details if needed

kolipakakondal, 2020-01-07:
Let's first try with these common eclipse solutions.

1. Check your preferences here: Window > Preferences > General > Editors > All Text Editors > Annotations > Debug Call Stack and Debug Current Instruction Pointer
2. Restart eclipse and debug!

kolipakakondal, 2021-02-16:
Closing this issue as there is no response for quite some time, please report us back if you still have an issue with our recent release.

closed 2021-02-16 by kolipakakondal