espressif/idf-eclipse-plugin #47: Incomplete missing project files

MarekBujko, 2020-01-19:
Hello all,
I clean install this plugin and all requirements (Eclipse IDE version: 2019-12).
I start with New -> Espressif IDF Project.
I choose for example Hello World.

Problem is that projects in explorer not have sdkconfig and here is only what you see in the picture.
[image]

**Environment Information
[eclipseEnvironmentInformation.txt](https://github.com/espressif/idf-eclipse-plugin/files/4083203/eclipseEnvironmentInformation.txt)

MarekBujko, 2020-01-19:
List tools:
Executing C:\Users\marek\AppData\Local\Programs\Python\Python38-32\python.exe C:\Users\marek\Desktop\workspace\esp-idf-v4.0-beta2\esp-idf-v4.0-beta2\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51.20170517 (recommended, installed)
* cmake: CMake build system
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20190313 (recommended, installed)
* mconf: menuconfig tool
  - v4.6.0.0-idf-20190628 (recommended, installed)
* ninja: Ninja build system
  - 1.9.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)

MarekBujko, 2020-01-19:
I had to compile the project first.

closed 2020-01-19 by MarekBujko

kolipakakondal, 2020-01-20:
> I had to compile the project first.

Yes, you need to compile the project first to get the sdkconfig file.