espressif/idf-eclipse-plugin #48: Unresolved inclusion: #include "freertos/FreeRTOS.h"

psommerfeld, 2020-01-25:
Re-installed eclipse, ESP-IDF & tools, twice to make sure I didn't miss a step.
Tried to build "hello_world"
Got unresolved inclusions on:

dpmcgabriel, 2020-01-27:
Same problem here

kolipakakondal, 2020-01-28:
Thanks for reporting. It seems this issue appears only on Windows OS.  On macOS and Ubuntu it's working. 

This is a duplicate of https://github.com/espressif/idf-eclipse-plugin/issues/43

We will update it soon.

ludo-d, 2020-02-05:
same issue here.
indeed using windows environment.
in Ubuntu it is ok.

samnangor, 2020-02-06:
Same problem in window 7
could not find environment in project properties; yet it does exist in Preferences.
Build and flash are working fine. Even though most include header files show Unresolved inclusion.

kolipakakondal, 2020-02-10:
Could you guys check this build and provide the feedback. **### This build works only if your windows user name doesn't have space in it!!!** since that's where all the tools are getting installed.  CMake is generating short paths if the path has space in it and that's causing an issue in Eclipse CDT. Until we provide a fix for this, please check this build.

Here is the build zip file to download 
https://github.com/kolipakakondal/idf-eclipse-plugin-dev/blob/master/com.espressif.idf.update-1.0.0.202002100648.zip

Installation instructions:
https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugins-from-local-archive

axbusch, 2020-02-10:
For me this seems to work. No more include warnings

but now this 

[image]

[Log](https://gist.github.com/axbusch/5e585dab9f8eb766ddd2181ad0497145)
-

JAICHANGPARK, 2020-02-11:
@kolipakakondal Thanks for guide 
But I had still problem about 'unresolved inclusion'

[image]

[image]

[image]

Do I need any more settings ?

JAICHANGPARK, 2020-02-11:
Solved .. 

[image]

@kolipakakondal Thank you

samnangor, 2020-02-11:
It is working fine with my win 10 and hello_world.
I will have to try with my win 7 to see if it is ok.

JAICHANGPARK, 2020-02-11:
after restarted I had this error.  

[image]

> For me this seems to work. No more include warnings
> 
> but now this
> 
> [image]
> 
> ## Here is a logfile
> [Log](https://gist.github.com/axbusch/5e585dab9f8eb766ddd2181ad0497145)

kolipakakondal, 2020-02-11:
> after restarted I had this error.
> 
Sure, I'll check that

kolipakakondal, 2020-02-11:
@JAICHANGPARK @axbusch NPE issue got fixed, could you please verify this build https://github.com/kolipakakondal/idf-eclipse-plugin-dev/blob/master/com.espressif.idf.update-1.0.0.202002111040.zip

JAICHANGPARK, 2020-02-12:
@kolipakakondal It's works fine! Awesome

kolipakakondal, 2020-02-12:
Hi All, IDF Eclipse Plugin v1.0.0-beta5 (Build: 1.0.0.202002120536) is published to the update site. Please get the latest https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

closed 2020-02-12 by kolipakakondal

BorisDess, 2020-07-06:
Installed eclipse today on a laptop, fresh install of everything.
Encountered the same problem.
Meh.
Tried VS ; worked.

atom327i, 2020-10-05:
Hello Guys!
I am trying to resolve the similar issue.
I have installed the Eclipse 2020.06CDT version with latest ESP-IDF. (from Friday)
As I create new ESP project based on any example, works fine.
But as I add any new component to be used, which obviously needs include files, includes will not be resolved. (I need to add the folders manually to Cmake.txt)
I copy the components to IDF Component folder

I have 20 years experience with SW engineeing, but I just starting with ESP32, so srry if I ask some trivial thing!
If somebody can have some hints i would appreciate it :)
Thanks!