espressif/idf-eclipse-plugin #49: Type 'esp_chip_info_t' could not be resolved in new idf project created using helloworld tmplate project

Fick707, 2020-01-29:
**Describe the bug**
Type 'esp_chip_info_t' could not be resolved 
osx
eclipse 2019-12 cdt

**To Reproduce**
Steps to reproduce the behavior:
1. create a new idf project using hello_world template project;
2. add some include directories;
3. build success,but Type 'esp_chip_info_t' could not be resolved in hello_world_main.c file

**Screenshots**
[image]

**Environment Information:** 
[eclipse.info.txt](https://github.com/espressif/idf-eclipse-plugin/files/4128732/eclipse.info.txt)

**ESP-IDF Tools Information:**
Executing /usr/bin/python /Users/fick/esp/esp-idf/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20191114 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.9.0 (recommended, installed)

Fick707, 2020-01-29:
when I double click the sdkconfig file in my helloesp32 project, sdkconfig editor shows a errror "Error retrieving output from the JSON configuration server,please check the console",and the console print below:

Executing action: confserver
Build directory '/Users/fick/eclipse-workspace/helloesp32/build' configured for project '/Users/fick/esp/esp-idf/examples/get-started/hello_world' not '/Users/fick/eclipse-workspace/helloesp32'. Run '/Users/fick/esp/esp-idf/tools/idf.py fullclean' to start again.

something wrong?

Fick707, 2020-01-30:
I miss "/usr/local/bin" in PATH environment variable .(Go into project properties,C/C++ Build->Environment->PATH)

closed 2020-01-30 by Fick707

kolipakakondal, 2020-01-31:
> usr/local/bin

Wasn't this added automatically when you run "install tools" from the eclipse menu?

pavan-vibhute, 2020-09-20:
I am getting this same error. What is the resolution ? I am running on Windows 10. 
I am using esp-idf-v4.1

lakshminarayanaks79, 2020-11-20:
/usr/local/bin" what exactly you mean from this because all the bin path are added and i am still getting error. so help me solve this