espressif/idf-eclipse-plugin #51: Please switch to EPL 2.0

ruspl-afed, 2020-01-31:
The EPL-1.0 has been deprecated

https://www.eclipse.org/legal/epl-2.0/faq.php#h.60mjudroo8e5

kolipakakondal, 2020-01-31:
> The EPL-1.0 has been deprecated
> 
> https://www.eclipse.org/legal/epl-2.0/faq.php#h.60mjudroo8e5

Thanks for reporting. We will switch to EPL 2.0 soon. Due to some dependency plugins, we are currently maintaining EPL 1.0

ruspl-afed, 2020-01-31:
Actually the upstream EPL-1.0 dependencies should not force you to use EPL-1.0 as well
You have a lot of issues regarding EPL in the sources, but let's start from top to bottom. By the "top" I mean repository LICENSE file.

ruspl-afed, 2020-01-31:
The situation is much worse, until you have GPL component inside the usage of EPL os not possible, 
https://www.eclipse.org/legal/eplfaq.php#USEINANOTHER

kolipakakondal, 2020-03-11:
With v1.0.0-rc2 we have updated to EPL 2.0. Hence closing the issue.

closed 2020-03-11 by kolipakakondal