espressif/idf-eclipse-plugin #52: Can't compile copy of project

Arag0to, 2020-02-11:
When trying to compile a C project, I can get the message :
`Build not configured correctly`
     from the CDT Build Console
In my C++ project, it compile with 0 error and warning but when looking, I see it fail at step 4/9 and when looking in the Problems windows, there is two error, One for not recognizing 'vTaskDelay' and the other one for 'portTick_PERIOD_MS' for the same problem and then if I compile again I can get the same error as for the C project.
When compiling and flashing in command line I don't get any error.
My target is configured.
[image]

Steps to reproduce the behavior:
C project:
1.Create a new project espressif template like exemple : peripherals/spi_slave/sender
2.Compile the project (Should compile properly)
3.Copy the project and paste it
4.Recompile the project copy

C++ project:
1.Create a new project espressif without template
2.Rename 'main.c' file with 'main.cpp'
3.In file 'CMakeList.txt' of 'main/' change line with 'main.c' for 'main.cpp'
4.Add code : 'extern "C" { void app_main(void); } in 'main.cpp
5.Continue with step 2 of C project

**Expected behavior**
When compile the git clone or the copy of a project, this project should compile correctly and I should be able to flash the programm.

**Environment Information:** 
[Environment Information.log](https://github.com/espressif/idf-eclipse-plugin/files/4188691/Environment.Information.log)

**ESP-IDF Tools Information:**
```
Executing /home/isecurity/.espressif/python_env/idf4.0_py2.7_env/bin/python /home/isecurity/esp/esp-idf/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51.20170517 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20190313 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.9.0 (recommended, installed)
```

Arag0to, 2020-02-14:
Little update.

The real problem here is changing the name of a project. If you rename a project, it won't compile even if you rename, compile, rename to original and compile, It won't work.

So my issue with github is that my project slug is not the same as the name of the project so when I clone the project onto another computer, it rename the project.
When I name the project the same as the slug and clone the project, it compile but I don't know why only on the computer where I cloned the project I get error on the vTaskDelay and the portTICK saying it's not regonize. 
The two computer have the same setting because it's a copy of the same Ubuntu vm.

noweare1, 2020-04-13:
I had the same problem. If you  use the "rename" in eclipse your project is hosed. It will not compile. The configuration is hidden and it seems once anything is changed now it doesn't correspond with the cofiguration. I think the configuration is in .cproject and .project files and there is a unique id number that respresents the configuration in eclipse.  Not very robust and not fixable.

isoftproject, 2020-09-22:
I solved this issue setting the User Defined Toochains following the steps (Configuring Core Build Toolchains) in:https://github.com/espressif/idf-eclipse-plugin#cmakeproject

expresspotato, 2021-04-10:
Removing these two lines in the .cproject fixed it for me (diff'd from the samples):
<storageModule moduleId="org.eclipse.cdt.make.core.buildtargets"/>
<storageModule moduleId="org.eclipse.cdt.internal.ui.text.commentOwnerProjectMappings"/>

sigmaaa, 2021-08-17:
This problem is fixed. The fix for it is available in the [latest version 2.2.0 ](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.2.0).

closed 2021-08-17 by sigmaaa