espressif/idf-eclipse-plugin #53: List of installed tools fails

axbusch, 2020-02-12:
Hi

i use the newest plugin
 Espressif IDF Plugins for Eclipse	1.0.0.202002120536	

The list of the installed tools does not work (the wrong window does not appear anymore)
[Log](https://gist.github.com/axbusch/53e783f2fa1b758c91633507ad35a626)

Regards

kolipakakondal, 2020-02-13:
> org.eclipse.core.commands.ExecutionException: Paths can't be empty. Please check IDF_PATH and Python

As per the error log, IDF_PATH and python are not found. I think you're trying to run the list tools before you're running install tools(this also take care of configuring the paths)

axbusch, 2020-02-13:
I will test again later and provide further logs. I think I saw the listed tools in the logs afterwards but they were not shown in eclipse

kolipakakondal, 2020-02-13:
BTW, the window dialog is removed. It just fetches the list of installed tools based on the existing configuration we have made.

axbusch, 2020-02-14:
`
!ENTRY com.espressif.idf.core 1 0 2020-02-14 22:57:06.963
!MESSAGE IDF_PATH :D:\esp\idf

!ENTRY com.espressif.idf.core 1 0 2020-02-14 22:57:06.963
!MESSAGE SOFTWARE\Python\PythonCore\2.7\InstallPath

!ENTRY com.espressif.idf.core 1 0 2020-02-14 22:57:06.963
!MESSAGE 

!ENTRY com.espressif.idf.core 1 0 2020-02-14 22:57:06.963
!MESSAGE SOFTWARE\Python\PythonCore\3.8-32\InstallPath

!ENTRY com.espressif.idf.core 1 0 2020-02-14 22:57:06.973
!MESSAGE C:\Python38_32\python.exe

!ENTRY com.espressif.idf.core 1 0 2020-02-14 22:57:06.973
!MESSAGE SOFTWARE\Python\PythonCore\2.7\InstallPath

!ENTRY com.espressif.idf.core 1 0 2020-02-14 22:57:06.973
!MESSAGE 

!ENTRY com.espressif.idf.core 1 0 2020-02-14 22:57:06.973
!MESSAGE SOFTWARE\Python\PythonCore\3.7\InstallPath

!ENTRY com.espressif.idf.core 1 0 2020-02-14 22:57:06.973
!MESSAGE C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe

!ENTRY com.espressif.idf.core 1 0 2020-02-14 22:57:06.983
!MESSAGE Python Path :null

!ENTRY org.eclipse.e4.ui.workbench 4 0 2020-02-14 22:57:06.983
!MESSAGE Command 'com.espressif.idf.ui.command.list' failed
!STACK 0
org.eclipse.core.commands.ExecutionException: Paths can't be empty. Please check IDF_PATH and Python
	at com.espressif.idf.ui.update.ListInstalledToolsHandler.execute(ListInstalledToolsHandler.java:38)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:283)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:97)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:318)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:252)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:173)
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:156)
	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:488)
	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:487)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:213)
	at org.eclipse.e4.ui.workbench.renderers.swt.HandledContributionItem.executeItem(HandledContributionItem.java:438)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.handleWidgetSelection(AbstractContributionItem.java:449)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.lambda$2(AbstractContributionItem.java:475)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4175)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1057)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:3988)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3587)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1160)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1049)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:660)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:559)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:154)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:150)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:657)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:594)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1465)
`

axbusch, 2020-02-14:
ok the tools are listed. Dont know if it i sa good decision to show nothing after selecting the menu

hmm it seems like my python installation is somehow broken. had problems with python before(was using the python from visual studio)

kolipakakondal, 2020-02-24:
Okay, thanks for verifying.  closing the issue.

closed 2020-02-24 by kolipakakondal