espressif/idf-eclipse-plugin #54: NPE after Build

axbusch, 2020-02-12:
Hi

i use the newest plugin
 Espressif IDF Plugins for Eclipse	1.0.0.202002120536	

After building my project i get an NPE

[Log](https://gist.github.com/axbusch/46bee0969d78305fb01813f8ecbe1cf5)

Regards

axbusch, 2020-02-12:
btw the other bug with the missing includes like freertos.h is fixed :) thx :+1:

pllanosclaria, 2020-02-13:
I am having the same issue, after upgrading to the newest plugin I started getting NPE. Uninstalled everything and installed from scratch didn't solve the issue.

**Eclipse** 
Version: 2019-12 (4.14.0). 
Build id: 20191212-1212

**Espressif IDF Plugin:** 
ID: 202002120536.com.espressif.idf. 
Version: 1.0.0.47I-cKb_nh7737337573C9E

[build.log](https://github.com/espressif/idf-eclipse-plugin/files/4197780/build.log)

Kind regards

pllanosclaria, 2020-02-13:
btw, On Linux (Ubuntu 18.04.1 x64) indexer isn't resolving the includes:
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"

Rolled back to IDF Eclipse Plugin v1.0.0-beta.4 (Build: 1.0.0.201912090756) and both issues were fixed.

Regards,

xXxlxsXx, 2020-02-19:
I almost have the same error message as your last one, maybe the same error? For me it occurs when I create an idf project then build (builds fine in this step) and then restart eclipse and build again (now this error appears).
[https://gist.github.com/xXxlxsXx/6f75def4353116833997bace7bba05ef](error)

pllanosclaria, 2020-02-19:
> I almost have the same error message as your last one, maybe the same error? For me it occurs when I create an idf project then build (builds fine in this step) and then restart eclipse and build [...]

It seems to be the same error, I'm also able to compile the first time (after creating a project) but if you restart the IDE you will get this error. 

If you create another project, copy all the files from the original project and build, it will build normally (I've done this to test if some file was causing the issue, e.g. project corruption file).

xXxlxsXx, 2020-02-19:
> If you create another project, copy all the files from the original project and build, it will build normally.

Indeed it works after that (but indexing still doesn't work and some types appear unresolved).

salocinx, 2020-02-22:
I can confirm the NPE issue discussed here with `IDF Eclipse Plugin v1.0.0-beta5`. It's a "project killer" ;-) I now rolled back to `IDF Eclipse Plugin v1.0.0-beta.4` and everything is working fine, including debugging with my Wrover Development Board. I don't post the error log, since it's the same as already reported. I am using Ubuntu 18.04; Eclipse CDT 2019.12; ESP32 Wrover DevKit and ESP-IDF from the current master branch.

By the way, thanks for your efforts! Great work and simplifies ESP32 development!

veerendra-nath, 2020-02-22:
how can i rollback to old version of plugin ?

pllanosclaria, 2020-02-22:
> how can i rollback to old version of plugin ?

1. Open Eclipse IDE and Uninstall the plugin
2. Restart Eclipse IDE
3. Download the previous version using this [link](https://github.com/espressif/idf-eclipse-plugin/archive/v1.0.0-beta.4.zip)
4. Run Eclipse again, goto Help->Install New Software"
5. Press the "Add" button, a window will open with the name of "Add Repository"
6. Introduce a name (whatever you want, e.g. IDF Eclipse Plugin v4)
7. Press the "Archive" button and select the file that you downloaded at step #3
8. Proceed with the installation as any other plugin.
9. Restart Eclipse

Note: At this point, your project could be messed up by the new version of the plugin, if so you will need to create a new project and copy all the files from the original project.

danak6jq, 2020-02-23:
I am also seeing this exact NPE

kolipakakondal, 2020-02-24:
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.0.0-rc1

Please check with a new build and let us know. This build also provides a support for esp32s2 chip along with other issues. Please check the release notes.

pllanosclaria, 2020-02-24:
> https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.0.0-rc1
> 
> Please check with a new build and let us know. This build also provides a support for esp32s2 chip along with other issues. Please check the release notes.

Confirmed, NPE issue and indexer not resolving some includes are both fixed in the new Release Candidate `https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.0.0-rc1`.

### Testing Platform:

**Eclipse**
Version: 2019-12 (4.14.0).
Build id: 20191212-1212

**OS**
Ubuntu 18.04.1 x64

Thank you @kolipakakondal!

salocinx, 2020-02-24:
Now everything working so far (creating IDF projects from example, indexing, compiling, flashing). I am using Ubuntu 18.04.04 LTS / Bionic, Eclipse CDT 2019.12 and ESP-WROVER-KIT development board. Thank you!

kolipakakondal, 2020-02-24:
> Now everything working so far (creating IDF projects from examples, indexing, compiling, flashing) with `IDF Eclipse Plugin v1.0.0-rc1 (Build: 1.0.0.202002201116)`.
> 
> I am using Ubuntu 18.04.04 LTS / Bionic, Eclipse CDT 2019.12 and ESP-WROVER-KIT development board. Thank you!

Awesome. I'm happy for you!

danak6jq, 2020-02-24:
I also confirm this 1.0.0-rc1 is working well on my Windows 10 installation (indexing, compiling, flashing). Thank you!

veerendra-nath, 2020-02-25:
Problem solved , Thank you

kolipakakondal, 2020-02-25:
Closing the issue as issue got fixed. If you guys see any other problem please report through a new ticket

closed 2020-02-25 by kolipakakondal