espressif/idf-eclipse-plugin #55: Not using virtualenv

veerendra-nath, 2020-02-23:
i installed and set up plugin , 
i created the first project , the build using the python virtual environment at this time 
i created 2nd project this time its not using virtual environment, build is not working because pyparsing in system python module is newer than the requirements , i am using arch linux

kolipakakondal, 2020-02-24:
Hi, Thanks for reporting the issue. Build will use the python path from the CDT build preferences. Can 
you verify the python path environment variable - IDF_PYTHON_ENV_PATH
(Preferences > C/C++ > Build >Environment)

Is the same python used during compiling the project?

veerendra-nath, 2020-02-25:
Hi, thanks fro responding ,
now the error not occurred(IN old(beta) and new (rc)version of plugin ) , IDF_PYTHON_ENV_PATH variable is set to idf virtual env folder ,
will report if  it occurred again

closed 2020-02-25 by kolipakakondal