espressif/idf-eclipse-plugin #57: Unresolved inclusion

jenya7, 2020-02-24:
Some of header files compiled OK. 
Like 
#include "driver/i2c.h"
#include "sdkconfig.h"

But when I include in the project BLE header files
#include "bta_api.h"
#include "esp_gap_ble_api.h"
#include "esp_gatts_api.h"
#include "esp_bt_defs.h"
#include "esp_bt_main.h"
I get - Unresolved inclusion error.

kolipakakondal, 2020-02-25:
Thanks for reporting. Could you name the project template which you've used from the examples?

jenya7, 2020-02-25:
It's a gatt_server example. I suppose the path to all unresolved inclusions could be added in CMakeLists.txt. Couldn't it?

kolipakakondal, 2020-03-31:
Please check this thread https://github.com/espressif/idf-eclipse-plugin/issues/34#issuecomment-557410081

kolipakakondal, 2021-02-16:
Closing this for now, please report us back if you see the same issue. Recently we have released 2.0.0-beta as well, this is fixes for unresolved headers and inclusion issues, please get the latest changes using https://dl.espressif.com/dl/idf-eclipse-plugin/updates/beta/ update site

closed 2021-02-16 by kolipakakondal