espressif/idf-eclipse-plugin #58: A target flash problem

jenya7, 2020-02-26:
I open a new Espressif IDF project as a template. It compiles OK. Then I connect my ESP32 board and see it in Device Manager. When I press Run button to flash the board I get an error

> Errors occurred during the build.
Errors running builder 'CDT Core Builder' on project 'door_beacon'.
java.lang.NullPointerException 

How do I configure the project to flash on my dev board?

kolipakakondal, 2020-03-04:
Thanks for reporting the issue. Could you please share with me an error log?
https://github.com/espressif/idf-eclipse-plugin#error-log

kolipakakondal, 2020-03-11:
Can you please try with the latest build?

jenya7, 2020-03-11:
Thank you. I'll try.

kolipakakondal, 2020-03-31:
Closing the issue as it's not producible with the latest release. We also have a latest release v1.0.1, please update to the latest and let us know if you see any issue

closed 2020-03-31 by kolipakakondal