espressif/idf-eclipse-plugin #60: Unresolved includes in latest idf-eclipse-plugin in Debug Launch mode

mvgorban, 2020-02-27:
Hi! 
I use  Windows10-1809/CDT Version: 2019-12 (4.14.0)/Build id: 20191212-1212/ Espressif IDF Plugins for Eclipse 1.0.0.202002201116 (updated to latest BETA 5 com.espressif.idf.update-v999-71-g9550085e that should fix the includes issue, as discussed before in https://github.com/espressif/idf-eclipse-plugin/issues/43#).

1) Using last plugin version In RUN launch mode all is OK **after building**, So, Indexer knows about includes only after build. OK.
2) In Debug Launch mode the including issue is permanent. No ways here to say Indexer about required path to IDF components... "Source" tab in GDB Hardware Debugging configuration cannot to help unfortunatly... or can?
So, how I could to fix this issue in Debug mode? Is it possible? Thanks!
[image]

[image]

[image]

mvgorban, 2020-02-27:
I have checked Indexer log file, no any component pathes (-I option) in Debug mode (in Run mode all is OK, components PATH added **after build only**).
As I understand, in Run mode, if target esp32 is selected, the plugin provides info for Indexer properly, but only after building. But in Debug mode Indexer is not notified. Why? What did I wrong? May be its happened because there is not any Target is selected in Debug mode? But I cannot to select esp32 target even  it created before,  this new target is just not present in selection...  

[dbg_DEBUG_MODE_AFTER_BUILD.log](https://github.com/espressif/idf-eclipse-plugin/files/4262901/dbg_DEBUG_MODE_AFTER_BUILD.log)
[dbg_DEBUG_MODE_BEFORE_BUILD.log](https://github.com/espressif/idf-eclipse-plugin/files/4262902/dbg_DEBUG_MODE_BEFORE_BUILD.log)
[dbg_RUN_MODE_AFTER_BUILD.log](https://github.com/espressif/idf-eclipse-plugin/files/4262903/dbg_RUN_MODE_AFTER_BUILD.log)
[dbg_RUN_MODE_BEFORE_BUILD.log](https://github.com/espressif/idf-eclipse-plugin/files/4262904/dbg_RUN_MODE_BEFORE_BUILD.log)

kolipakakondal, 2020-03-09:
I'm able to reproduce the issue in the debug mode. As a workaround, when you move to 'C/C++' perspective run mode, do the re-build of the project using 'Project >C/C++ Index >Rebuild' that should resolve the issue.

We will update you once the issue is resolved. Thanks for reporting.

kolipakakondal, 2020-03-09:
When you move back to the run mode,  just clicking on the build icon also resolves the issue.

kolipakakondal, 2021-02-16:
We have released 2.0.0-beta, this has a fix for unresolved includes issue, please get the latest changes using https://dl.espressif.com/dl/idf-eclipse-plugin/updates/beta/ update site

sigmaaa, 2021-08-17:
Indexer works better in the latest release. You also could face a problem with the indexer after changing modes from debug to launch, now it's fixed. Now it's fixed. Please, check the [2.2.0 plugin version](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.2.0) and feel free to raise an issue if you will face some problems with it

closed 2021-08-17 by sigmaaa