espressif/idf-eclipse-plugin #61: java.lang.NullPointerException Esp-idf only

techny4287, 2020-03-16:
**Describe the bug**
Every few times I recompile the code for a custom project in eclipse I get the following error:
Errors occurred during the build.
Errors running builder 'CDT Core Builder' on project 'PROJECT_NAME'.
java.lang.NullPointerException

**To Reproduce**
Steps to reproduce the behavior:
1. Install tools and configuration as listed in ESP-idf read the docs latest
2. create a custom project -> add code
3. Build muiltiple times, no exact number to reproduce
4. See error

What "can kindof fix the error temporarily"

**Expected behavior**
What "can fix the error temporarily"
Step 1) close project
Step 2) delete project (just from eclipse)
Step 3) Open projects from file system,
Step 4) select the directory

What DOESN'T work -> 
1 ) Rebuilding indexer
2) select/ deselect enable indexer
3) select/deselect Allow heuristic resolution of includes
4) select/deselect Index source and dheader files open in editor
5) changing any sort environment variable to allow for includes to be resolved.

The only thing that fixes this error are the steps above and they do not last long, it does not appear to be a X# of times or period, but does happen quite often ( > 2 per hour). 

**Screenshots**
Attached
[image]

**Environment Information:** 
Attached
[config.txt](https://github.com/espressif/idf-eclipse-plugin/files/4338965/config.txt)

**ESP-IDF Tools Information:**
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20191114 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.9.0 (recommended, installed)
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

techny4287, 2020-03-16:
I forgot to add that this was a minor issue when using the GCC build system in esp-idf but has been increasing in frequency since migrating to cmake.

Alvin1Zhang, 2020-03-17:
@techny4287 Thanks for reporting. @kolipakakondal Would you please help take a look? Thanks.

kolipakakondal, 2020-03-17:
Hi @techny4287 

Looks like you're working with the build number 1.0.0.202002120536 (beta 5), could you please update to the latest build which is https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.0.0-rc2

The issue got fixed with the latest build. 

Best Regards,
Kondal

kolipakakondal, 2020-03-31:
Closing this issue as it's fixed with the recent release. Please update and let us know if you see any issue.

closed 2020-03-31 by kolipakakondal

akotowski, 2022-06-03:
I see this issue using the "hello_world" example in 
Espressif IDE plugin for CDT:
Version: 2.4.2
Build id: 20220308-1406
and 
ESP-IDF v4.4.1
Installed per instructions and created new Espressif IDE project using "hello_world" template.  Then attempted to build project.

This does not happen with "blink" example. or when pasting the "hello_world" main.c into the "sample_project" main.c

kolipakakondal, 2022-06-05:
Hi @akotowski Could you share the error log to analyze further?
https://github.com/espressif/idf-eclipse-plugin#error-log

akotowski, 2022-06-05:
My choices are "Show View - Problems" or "Show View - Problem Details."  Both are empty.
The error occurs instantly when I click the build button.